dsmrreader/dsmr-reader #1022: Datalogger blijft hangen wanneer MQTT aan staat

meijjaa, 2020-06-15:
Wat gebruik je?
- DSMR-reader versie:   v3.2 - v3.10
- Type RaspberryPi or server:  PC, Ubuntu 20.04 x64
- Standaardinstallatie of Docker:   Standaard
- DSMR-protocol slimme meter v4/v5:   v5

---
Hi Dennis,

Sinds half maart loopt de datalogger niet meer door. Er komen geen nieuwe telegrammen binnen en Postgres draait continue 3 threads op 14% CPU load. Server draait nu ~2.5 jaar met een DB van 44 GB ondertussen.
DSMR geupgrade naar v3.10 en Postgres naar 12, retentie aangezet (half jaar) maar hielp allemaal niet. Als ik MQTT uit zet werkt het ineens wel weer, geen hoge CPU load meer en telegrammen komen netjes binnen. 
Ik gebruik MQTT om actuele energieverbruik in Home Assistant te tonen, zou fijn zijn als dat weer kan werken. Zou je hier naar kunnen kijken?

In ieder geval bedankt!

dennissiemensma, 2020-06-15:
Bedankt voor je melding. Er zijn een tijd terug aardig wat verbeteringen gedaan m.b.t. MQTT, maar ik vermoed dat dat al in de versie zat die je draaide voor de update.

Desnietemin is het goed om eerst te kijken naar de logging. Als het MQTT-proces de load veroorzaakt, verwacht ik ook meldingen in de bijbehorende log: https://dsmr-reader.readthedocs.io/nl/v3/troubleshooting.html#supervisor

Als je niets ziet, dan kun je DEBUG level aanzetten, de processen herstarten en het dan nogmaals proberen: https://dsmr-reader.readthedocs.io/nl/v3/troubleshooting.html#logging
Vervolgens is het de vraag is het de vraag of het in het MQTT-proces zit of in het backend-proces, waarbij laatstgenoemde wat trager kan zijn bij een grote inhaalslag voor dataopschoning.

Verder kun je nog kijken of ern een grote achterstand aan MQTT-berichten is. In een eerdere release zou dit niet meer moeten voorkomen, maar wellicht ging het bij jou nog niet goed. Hier zou hooguit een paar honder of minder uit moeten komen:
```
sudo su - postgres
psql dsmrreader

select count(id) from dsmr_mqtt_message;
```

spdfrk, 2020-06-15:
Ik denk dat je het probleem gevonden hebt, select count(id) from dsmr_mqtt_message geeft 333812805 rows :). 
Zou je kunnen uitleggen waarom deze MQTT berichten worden opgeslagen? Kan ik deze tabel straffeloos legen?

Log ziet er zo uit:
> Jun 13 23:42:08 pc11 python3[28181]: DEBUG:commands:Received telegram:
> --- knip telegram ---
> Jun 13 23:48:31 pc11 python3[28181]: [2020-06-13 23:48:31,048] WARNING  MQTT: Ignoring dsmr/reading/timestamp to max messages in queue reached
> Jun 13 23:48:31 pc11 python3[28181]: WARNING:commands:MQTT: Ignoring dsmr/reading/timestamp to max messages in queue reached
> Jun 13 23:54:57 pc11 python3[28181]: [2020-06-13 23:54:57,255] WARNING  MQTT: Ignoring dsmr/reading/electricity_delivered_1 to max messages in queue reached
> Jun 13 23:54:57 pc11 python3[28181]: WARNING:commands:MQTT: Ignoring dsmr/reading/electricity_delivered_1 to max messages in queue reached
> Jun 14 00:01:22 pc11 python3[28181]: [2020-06-14 00:01:22,384] WARNING  MQTT: Ignoring dsmr/reading/electricity_returned_1 to max messages in queue reached
> Jun 14 00:01:22 pc11 python3[28181]: WARNING:commands:MQTT: Ignoring dsmr/reading/electricity_returned_1 to max messages in queue reached
> Jun 14 00:07:43 pc11 python3[28181]: [2020-06-14 00:07:43,369] WARNING  MQTT: Ignoring dsmr/reading/electricity_delivered_2 to max messages in queue reached
> Jun 14 00:07:43 pc11 python3[28181]: WARNING:commands:MQTT: Ignoring dsmr/reading/electricity_delivered_2 to max messages in queue reached
> Jun 14 00:14:06 pc11 python3[28181]: [2020-06-14 00:14:06,330] WARNING  MQTT: Ignoring dsmr/reading/electricity_returned_2 to max messages in queue reached
> Jun 14 00:14:06 pc11 python3[28181]: WARNING:commands:MQTT: Ignoring dsmr/reading/electricity_returned_2 to max messages in queue reached
> Jun 14 00:20:56 pc11 python3[28181]: [2020-06-14 00:20:56,581] WARNING  MQTT: Ignoring dsmr/reading/electricity_currently_delivered to max messages in queue reached
> Jun 14 00:20:56 pc11 python3[28181]: WARNING:commands:MQTT: Ignoring dsmr/reading/electricity_currently_delivered to max messages in queue reached
> Jun 14 00:27:32 pc11 python3[28181]: [2020-06-14 00:27:32,528] WARNING  MQTT: Ignoring dsmr/reading/electricity_currently_returned to max messages in queue reached
> Jun 14 00:27:32 pc11 python3[28181]: WARNING:commands:MQTT: Ignoring dsmr/reading/electricity_currently_returned to max messages in queue reached
> Jun 14 00:33:56 pc11 python3[28181]: [2020-06-14 00:33:56,126] WARNING  MQTT: Ignoring dsmr/reading/phase_currently_delivered_l1 to max messages in queue reached
> Jun 14 00:33:56 pc11 python3[28181]: WARNING:commands:MQTT: Ignoring dsmr/reading/phase_currently_delivered_l1 to max messages in queue reached
> Jun 14 00:40:28 pc11 python3[28181]: [2020-06-14 00:40:28,202] WARNING  MQTT: Ignoring dsmr/reading/phase_currently_delivered_l2 to max messages in queue reached
> Jun 14 00:40:28 pc11 python3[28181]: WARNING:commands:MQTT: Ignoring dsmr/reading/phase_currently_delivered_l2 to max messages in queue reached
> Jun 14 00:46:53 pc11 python3[28181]: [2020-06-14 00:46:53,678] WARNING  MQTT: Ignoring dsmr/reading/phase_currently_delivered_l3 to max messages in queue reached
> Jun 14 00:46:53 pc11 python3[28181]: WARNING:commands:MQTT: Ignoring dsmr/reading/phase_currently_delivered_l3 to max messages in queue reached
> Jun 14 00:53:20 pc11 python3[28181]: [2020-06-14 00:53:20,882] WARNING  MQTT: Ignoring dsmr/reading/extra_device_timestamp to max messages in queue reached
> Jun 14 00:53:20 pc11 python3[28181]: WARNING:commands:MQTT: Ignoring dsmr/reading/extra_device_timestamp to max messages in queue reached
> Jun 14 00:59:49 pc11 python3[28181]: [2020-06-14 00:59:49,146] WARNING  MQTT: Ignoring dsmr/reading/extra_device_delivered to max messages in queue reached
> Jun 14 00:59:49 pc11 python3[28181]: WARNING:commands:MQTT: Ignoring dsmr/reading/extra_device_delivered to max messages in queue reached
> Jun 14 01:06:09 pc11 python3[28181]: [2020-06-14 01:06:09,616] WARNING  MQTT: Ignoring dsmr/reading/id to max messages in queue reached
> Jun 14 01:06:09 pc11 python3[28181]: WARNING:commands:MQTT: Ignoring dsmr/reading/id to max messages in queue reached
> Jun 14 01:06:09 pc11 python3[28181]: [2020-06-14 01:06:09,719] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 10.0s
> Jun 14 01:06:09 pc11 python3[28181]: DEBUG:commands:dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 10.0s
> Jun 14 01:06:19 pc11 python3[28181]: [2020-06-14 01:06:19,728] DEBUG    Reconnecting to database to refresh connection...
> Jun 14 01:06:19 pc11 python3[28181]: DEBUG:commands:Reconnecting to database to refresh connection...
> Jun 14 01:06:20 pc11 python3[28181]: [2020-06-14 01:06:20,024] DEBUG    Received telegram:
> --- knip telegram ---

dennissiemensma, 2020-06-15:
Je kunt die tabel rustig legen. 

De techniek erachter is dat MQTT berichten daar worden opgeslagen en dat het ``dsmr_mqtt`` proces ze vervolgens verstuurt. Hierdoor kan het ``dsmr_mqtt`` proces een persistente verbinding behouden, zonder voor elk bericht een aparte verbinding op te zeten.

Een oudere versie van DSMR-reader hield er niet rekening mee dat iemand ook (onbewust) MQTT aan heeft staan, maar ``dsmr_mqtt`` niet. Dan krijg je dus deze situatie.

Nieuwere versies hanteren daarom een max, maar ik had achteraf wellicht een automatische opschoning moeten inbouwen. Als je de tabel leegt dan komen er nog max 200 ofzo in. 

Schakel dus vooral ook het MQTT-proces in: https://dsmr-reader.readthedocs.io/en/latest/mqtt.html#recent-versions-v3-0-0-or-higher

dennissiemensma, 2020-06-15:
Je kunt na afloop ook nog die tabel even laten 'vacuumen'. Ik weet niet of het in nieuwere versies van Postgres nog steeds nodig is, maar het kan sowieso nooit kwaad:

```
sudo su - postgres
vacuumdb -f -v -d dsmrreader -t dsmr_mqtt_message
```

spdfrk, 2020-06-15:
Dank je! Na een truncate is mijn DB van 44 GB naar 1.9 GB gegaan dus blijkbaar geen vacuum meer nodig tegenwoordig :). Alsnog even gedraaid.

Blijkt dat ik even de memo had gemist over dsmr_mqtt! Het ging alweer bijna mis (mqtt msgs begonnen weer op te stapelen)  maar nu dus ook meteen duidelijk waar het probleem ontstaan is :D Ik draai de services allemaal vanuit systemd, dsmr_mqtt toegevoegd en zo te zien zijn alle problemen opgelost. Mijn home automation setup had duidelijk last van gebrekkig onderhoud van mijn kant, inderdaad geen bug dus.

meijjaa, 2020-06-15:
Dank je! Na een truncate is mijn DB van 44 GB naar 1.9 GB gegaan dus blijkbaar geen vacuum meer nodig tegenwoordig :). Alsnog even gedraaid.

Blijkt dat ik even de memo had gemist over dsmr_mqtt! Het ging alweer bijna mis (mqtt msgs begonnen weer op te stapelen) maar nu dus ook meteen duidelijk waar het probleem ontstaan is :D Ik draai de services allemaal vanuit systemd, dsmr_mqtt toegevoegd en zo te zien zijn alle problemen opgelost. Mijn home automation setup had duidelijk last van gebrekkig onderhoud van mijn kant, inderdaad geen bug dus.

closed 2020-06-15 by meijjaa