dsmrreader/dsmr-reader #1026: Dag en uur statistieken worden niet meer berekend na meterwissel

n9iels, 2020-06-18:
Mogge!

Ik heb gisteren helaas een nieuwe meter gekregen van Stedin (het oude ding deed alles prima maar stuurde enkel geen gegevens meer op naar de beheerder...). Echter is deze nieuwe meter niet echt "nieuw", als in: de meterstanden zijn niet 0.

Deze ochtend was ik van plan om de dag/uur statistieken te repareren, echter blijken deze sinds de meterwissel niet meer gegenereerd te worden. Ik krijg wel netjes alle telegrammen binnen (zie onderstaand voor meest recente) Na wat zoekwerk kwam ik uit op #742 en volgens mij heb ik hetzelfde probleem. De laatste telegram geeft een foutieve datum aan voor de gasmeter, namelijk `
Jan. 1, 1970, 1 a.m.`

Het niet goed kunnen uitlezen van gas is voor mij geen echt groot punt, echter zou ik de electriciteit wel graag weer kunnen uitlezen. Heb je suggesties hoe ik op kan lossen?
---

Wat gebruik je?
- DSMR-reader versie:  3.10
- Type RaspberryPi or server: Ubuntu 18.04.4
- Standaardinstallatie of Docker: Docker, telegrammen komen via een remote datalogger binnen
- DSMR-protocol slimme meter v4/v5:  5

**Laatste telegram**
```
/ISK5\2M550T-1011

1-3:0.2.8(50)
0-0:1.0.0(200618082014S)
0-0:96.1.1(4530303334303036373933393631373136)
1-0:1.8.1(003670.567*kWh)
1-0:1.8.2(003607.663*kWh)
1-0:2.8.1(000000.106*kWh)
1-0:2.8.2(000016.246*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.089*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00008)
0-0:96.7.9(00002)
1-0:99.97.0()
1-0:32.32.0(00001)
1-0:52.32.0(00000)
1-0:72.32.0(00000)
1-0:32.36.0(00001)
1-0:52.36.0(00001)
1-0:72.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(233.6*V)
1-0:52.7.0(231.9*V)
1-0:72.7.0(234.1*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.000*kW)
1-0:41.7.0(00.059*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.193*kW)
1-0:42.7.0(00.017*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0()
0-1:24.2.1(700101010000W)(00000000)
0-2:24.1.0(003)
0-2:96.1.0()
0-2:24.2.1(700101010000W)(00000000)
!C693
```

dennissiemensma, 2020-06-18:
Bedankt voor je melding. Het lijkt in dit geval ook deels op #818. 

Ik heb daar in 26b8f37bec97cd09efb7d80cf978992a611ed28a een fix gemaakt die ervoor zorgt dat dit probleem zichzelf 24 uur na gebrek aan verdere gasmetingen moet oplossen. Alleen weet ik dus niet of dat helemaal hetzelfde voor jou is.

Denk je dat er inmiddels al 24 uur voorbij zijn?

n9iels, 2020-06-18:
Ja we zijn inmiddels 24 uur verder, laatste gasmeting was om `June 17, 2020, 9:55 a.m.`

Probleem hier is wellicht dat ik wel gasmetingen door krijg maar de datum compleet fout is? Krijg het idee dat ie daarom eeuwig wacht op "de volgende dag"

Overigens zie ik wel dat de gasmeter gekoppeld is met de slimme meter, alleen geven ze nog geen data door. Vermoedelijk moet de netbeheerder hier nog wat stappen nemen.

dennissiemensma, 2020-06-18:
Wellicht kun je dan nog even checken wat de melding nu in de logs is. Of het is iets anders, of die fix waar ik naar verwees werkt niet (goed).

Je kunt ook nog even kijken naar de meest recente gasmetingen op basis van ID ipv tijd. Ik weet niet uit mn hoofd of hij die gasmetingen simpelweg overslaat als er teveel tijd tussen 'nu' en de betreffende datum uit de meter zit.

```
sudo su - postgres
psql dsmrreader

select read_at from dsmr_consumption_gasconsumption order by id desc limit 10;
```

n9iels, 2020-06-18:
De logs geeft nog steeds dezelfde melding (`Stats: Waiting for first gas reading on the next day...`). De laatste 10 recente metingen die zijn opgeslagen zijn van 24 uur terug, dit is de output:

```
 2020-06-17 09:55:04+02
 2020-06-17 09:50:09+02
 2020-06-17 09:45:02+02
 2020-06-17 09:40:05+02
 2020-06-17 09:35:06+02
 2020-06-17 09:30:09+02
 2020-06-17 09:25:02+02
 2020-06-17 09:20:01+02
 2020-06-17 09:15:04+02
 2020-06-17 09:10:04+02
```

dennissiemensma, 2020-06-18:
Ik zou die afwijkende records voor nu laten voor wat ze zijn, want ze zijn vermoedelijk al verwerkt in de `dsmr_consumption_*` tabel(len).

Wat betreft de dagstatistieken, ik denk dat de eerdere fix niet helemaal goed werkt. Je kunt nog proberen om handmatig de dag van gisteren in te vullen in: `/admin/dsmr_stats/daystatistics/add/`. Mogelijk gaat die daarna wel verder. Ter referentie kun je een dag uit `/admin/dsmr_stats/daystatistics/` bekijken.

Als alternatief kun je wachten op een nieuwe fix, maar dat kan zo een paar dagen of een week duren.

n9iels, 2020-06-19:
Ik heb dit nu tijdelijk opgelost door in de `dsmr_consumption_gasconsumption` tabel voor elke dag 1 record toe te voegen.

```sql
INSERT INTO dsmr_consumption_gasconsumption (read_at, delivered, currently_delivered) VALUES ('2020-06-19 00:00:01+02', 730.459, 0.000);
```
Dit zorgt ervoor dat de  uur en dag statistieken in ieder geval weer berekend worden omdat de vorige dag is afgesloten :)

dennissiemensma, 2020-06-19:
Je zou kunnen kijken of dat nu nog steeds nodig is voor komende dagen. Ik denk dat die fix niet goed werkt wanneer het midden op de dag gebeurde zonder dagafsluiting.

dennissiemensma, 2020-06-24:
Heb je inmiddels kunnen proberen of het nog steeds werkt als je niet meer handmatig die gasmeting aanmaakt?

n9iels, 2020-06-25:
Helaas niet. De volgende dag deed de gasmeter het weer volledig en kreeg ik weer de juiste standen door. Sinds dien geen problemen meer gehad :)

dennissiemensma, 2020-06-25:
Oke bedankt voor je terugkoppeling! In dat geval laat ik het verder voor wat het is. : ]

closed 2020-06-25 by dennissiemensma