dsmrreader/dsmr-reader #1027: Updaten vanaf v3.9.1 lukt niet meer.

quadrophenianl, 2020-06-19:
Goedemorgen Dennis,

Al enkele maanden maak ik met veel plezier gebruik van deze applicatie, waarvoor dank! Nu is het echter zo dat het updaten van de applicatie sinds v3.9.1 niet meer soepel verloopt. Voorheen verliepen alle updates perfect met hulp van het _./deploy_ command, maar telkens als ik nu dit commando uitvoer loopt de hele DSMR Reader in de soep.

Momenteel ben ik op het werk, dus kan even niet concreet alle foutmeldingen weergeven, maar wellicht is dit een bekend probleem en kan het met deze summiere info opgelost worden. Mijn excuus in ieder geval voor de onvolledige info, als je meer nodig blijkt te hebben zal ik dat dit weekend uiteraard aanleveren.

Telkens tijdens het updaten krijg ik sinds kort de melding dat een PIP file niet gevonden is. Hij maakt vervolgens wel de update af, maar vervolgens is de webapplicatie offline. Ook bij handmatig starten van de services wil de applicatie niet meer starten. Gelukkig heb ik vrijwel altijd een up-to-date SD clone ter beschikking welke ik dan verwissel om weer gebruik te kunnen maken van de werkende v3.9.1 versie.

Wellicht kun je me met bovenstaande (onvolledige) info al op weg helpen?

Alvast een fijn weekend gewenst!

----

**- DSMR-reader versie:**   v3.9.1
**- Type RaspberryPi or server**:   RPi 4
**- Standaardinstallatie of Docker:**   Standaardinstallatie

dennissiemensma, 2020-06-19:
Bedankt voor je melding. Ik denk dat je toch de exacte fout even moet opzoeken van het weekend. Het wordt anders erg lastig debuggen.

quadrophenianl, 2020-06-22:
Hoi Dennis,

Hierbij de complete log van de installatie, met als gevolg dat de webinterface een **502 Bad Gateway** error geeft:

` --- You are currently running version: 
3.9.1

 --- Checking Python version.
Running Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0]

 --- Pulling remote repository for new commits...
remote: Enumerating objects: 630, done.
remote: Counting objects: 100% (630/630), done.
remote: Compressing objects: 100% (161/161), done.
remote: Total 684 (delta 474), reused 578 (delta 425), pack-reused 54
Receiving objects: 100% (684/684), 376.42 KiB | 1.09 MiB/s, done.
Resolving deltas: 100% (481/481), completed with 193 local objects.
From https://github.com/dennissiemensma/dsmr-reader
   743091f9..a1869c3c  v3          -> origin/v3
   743091f9..a1869c3c  development -> origin/development
 * [new branch]        v4          -> origin/v4
 * [new branch]        wip/v4.0    -> origin/wip/v4.0
 * [new tag]           v3.10.1     -> v3.10.1
 * [new tag]           v3.11.0     -> v3.11.0
 * [new tag]           v3.10.0     -> v3.10.0

 --- Merging/updating checkout.
Updating 743091f9..a1869c3c
Fast-forward
 .github/ISSUE_TEMPLATE/bug.md                      |   17 +-
 .github/ISSUE_TEMPLATE/feature.md                  |    1 -
 .github/ISSUE_TEMPLATE/help.md                     |   17 +-
 .github/PULL_REQUEST_TEMPLATE.md                   |    2 +-
 docs/changelog.rst                                 |   31 +
 docs/credits.rst                                   |    3 +-
 docs/data_integrity.rst                            |    2 +-
 docs/faq.rst                                       |    7 +
 docs/home_assistant.rst                            |    7 +-
 docs/installation/datalogger.rst                   |   30 +-
 docs/installation/docker.rst                       |   38 +-
 docs/intro.rst                                     |   20 +-
 docs/locale/nl/LC_MESSAGES/admin/consumption.po    |    3 -
 docs/locale/nl/LC_MESSAGES/application.po          |   12 -
 docs/locale/nl/LC_MESSAGES/changelog.po            | 1174 ++++++++++----------
 docs/locale/nl/LC_MESSAGES/contributing.po         |   10 -
 docs/locale/nl/LC_MESSAGES/credits.mo              |  Bin 5348 -> 5523 bytes
 docs/locale/nl/LC_MESSAGES/credits.po              |  330 +-----
 docs/locale/nl/LC_MESSAGES/data_integrity.mo       |  Bin 4498 -> 4497 bytes
 docs/locale/nl/LC_MESSAGES/data_integrity.po       |    4 +-
 docs/locale/nl/LC_MESSAGES/development.po          |    3 -
 docs/locale/nl/LC_MESSAGES/faq.mo                  |  Bin 5193 -> 6194 bytes
 docs/locale/nl/LC_MESSAGES/faq.po                  |  310 +-----
 docs/locale/nl/LC_MESSAGES/faq/database.po         |  274 -----
 docs/locale/nl/LC_MESSAGES/faq/uninstall.po        |  507 ---------
 docs/locale/nl/LC_MESSAGES/faq/update.po           |  362 ------
 docs/locale/nl/LC_MESSAGES/home_assistant.mo       |  Bin 1001 -> 1334 bytes
 docs/locale/nl/LC_MESSAGES/home_assistant.po       |   39 +-
 docs/locale/nl/LC_MESSAGES/installation.mo         |  Bin 2783 -> 2787 bytes
 docs/locale/nl/LC_MESSAGES/installation.po         |   18 +-
 .../nl/LC_MESSAGES/installation/datalogger.mo      |  Bin 3390 -> 4167 bytes
 .../nl/LC_MESSAGES/installation/datalogger.po      |   89 +-
 docs/locale/nl/LC_MESSAGES/installation/docker.mo  |  Bin 2147 -> 1802 bytes
 docs/locale/nl/LC_MESSAGES/installation/docker.po  |   86 +-
 .../nl/LC_MESSAGES/installation/explained.po       |  100 --
 docs/locale/nl/LC_MESSAGES/intro.mo                |  Bin 2013 -> 2078 bytes
 docs/locale/nl/LC_MESSAGES/intro.po                |   71 +-
 docs/locale/nl/LC_MESSAGES/mqtt.mo                 |  Bin 2581 -> 2594 bytes
 docs/locale/nl/LC_MESSAGES/mqtt.po                 |  505 +--------
 docs/locale/nl/LC_MESSAGES/requirements.po         |  107 --
[... 103 lines cut ...]
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 357, in execute
    django.setup()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/__init__.py", line 24, in setup
    apps.populate(settings.INSTALLED_APPS)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/apps/registry.py", line 91, in populate
    app_config = AppConfig.create(entry)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/apps/config.py", line 90, in create
    module = import_module(entry)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'rangefilter'
[!] Halted. Post-deployment script exited with non-zero code`

Als ik vervolgens het command **./reload.sh** uitvoer, krijg ik de volgende uitdraai:

` * Reloading process: dsmr_webinterface (Gunicorn)   [??] PID file does not exist
 * Reloading process: dsmr_backend                   [??] PID file does not exist
 * Reloading process: dsmr_datalogger                [??] PID file does not exist
 * Reloading process: dsmr_mqtt                   kill: (13601): Proces bestaat niet
   [OK]
`

quadrophenianl, 2020-06-22:
> test

dennissiemensma, 2020-06-22:
Bedankt voor de aanvulling. De eerste foutmelding die ik lees lijkt vrij helder:
```
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Toegang geweigerd: '/home/dsmr/.cache/pip/wheels/68/06/08/c46a34165480208978c36648857044e56c0b4b1de60c210b2f'
Consider using the `--user` option or check the permissions.
```

Je kunt dit proberen:
```
sudo chown -R dsmr.dsmr /home/dsmr
```

En daarna opnieuw:
```
sudo su - dsmr
./deploy.sh
```

quadrophenianl, 2020-06-23:
Hoi Dennis,

Bedankt voor je hulp! Helaas geeft hij na uitvoering van jouw werkwijze dezelfde foutmelding. Wel duurde het deployen een stuk langer dan normaal. Hierbij de nieuwe log, ik hoor graag wat ik nog zou kunnen proberen...

Oh ja, hoe zet ik onderstaande code in zo'n grijs vakje? Dat oogt wel zo netjes en opgeruimd ;)

 --- You are currently running version: 
3.9.1

 --- Checking Python version.
Running Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0]

 --- Pulling remote repository for new commits...
remote: Enumerating objects: 630, done.
remote: Counting objects: 100% (630/630), done.
remote: Compressing objects: 100% (161/161), done.
remote: Total 684 (delta 474), reused 578 (delta 425), pack-reused 54
Receiving objects: 100% (684/684), 376.42 KiB | 1.75 MiB/s, done.
Resolving deltas: 100% (481/481), completed with 193 local objects.
From https://github.com/dennissiemensma/dsmr-reader
   743091f9..a1869c3c  v3          -> origin/v3
   743091f9..a1869c3c  development -> origin/development
 * [new branch]        v4          -> origin/v4
 * [new branch]        wip/v4.0    -> origin/wip/v4.0
 * [new tag]           v3.10.1     -> v3.10.1
 * [new tag]           v3.11.0     -> v3.11.0
 * [new tag]           v3.10.0     -> v3.10.0

 --- Merging/updating checkout.
Updating 743091f9..a1869c3c
Fast-forward
 .github/ISSUE_TEMPLATE/bug.md                      |   17 +-
 .github/ISSUE_TEMPLATE/feature.md                  |    1 -
 .github/ISSUE_TEMPLATE/help.md                     |   17 +-
 .github/PULL_REQUEST_TEMPLATE.md                   |    2 +-
 docs/changelog.rst                                 |   31 +
 docs/credits.rst                                   |    3 +-
 docs/data_integrity.rst                            |    2 +-
 docs/faq.rst                                       |    7 +
 docs/home_assistant.rst                            |    7 +-
 docs/installation/datalogger.rst                   |   30 +-
 docs/installation/docker.rst                       |   38 +-
 docs/intro.rst                                     |   20 +-
 docs/locale/nl/LC_MESSAGES/admin/consumption.po    |    3 -
 docs/locale/nl/LC_MESSAGES/application.po          |   12 -
 docs/locale/nl/LC_MESSAGES/changelog.po            | 1174 ++++++++++----------
 docs/locale/nl/LC_MESSAGES/contributing.po         |   10 -
 docs/locale/nl/LC_MESSAGES/credits.mo              |  Bin 5348 -> 5523 bytes
 docs/locale/nl/LC_MESSAGES/credits.po              |  330 +-----
 docs/locale/nl/LC_MESSAGES/data_integrity.mo       |  Bin 4498 -> 4497 bytes
 docs/locale/nl/LC_MESSAGES/data_integrity.po       |    4 +-
 docs/locale/nl/LC_MESSAGES/development.po          |    3 -
 docs/locale/nl/LC_MESSAGES/faq.mo                  |  Bin 5193 -> 6194 bytes
 docs/locale/nl/LC_MESSAGES/faq.po                  |  310 +-----
 docs/locale/nl/LC_MESSAGES/faq/database.po         |  274 -----
 docs/locale/nl/LC_MESSAGES/faq/uninstall.po        |  507 ---------
 docs/locale/nl/LC_MESSAGES/faq/update.po           |  362 ------
 docs/locale/nl/LC_MESSAGES/home_assistant.mo       |  Bin 1001 -> 1334 bytes
 docs/locale/nl/LC_MESSAGES/home_assistant.po       |   39 +-
 docs/locale/nl/LC_MESSAGES/installation.mo         |  Bin 2783 -> 2787 bytes
 docs/locale/nl/LC_MESSAGES/installation.po         |   18 +-
 .../nl/LC_MESSAGES/installation/datalogger.mo      |  Bin 3390 -> 4167 bytes
 .../nl/LC_MESSAGES/installation/datalogger.po      |   89 +-
 docs/locale/nl/LC_MESSAGES/installation/docker.mo  |  Bin 2147 -> 1802 bytes
 docs/locale/nl/LC_MESSAGES/installation/docker.po  |   86 +-
 .../nl/LC_MESSAGES/installation/explained.po       |  100 --
 docs/locale/nl/LC_MESSAGES/intro.mo                |  Bin 2013 -> 2078 bytes
 docs/locale/nl/LC_MESSAGES/intro.po                |   71 +-
 docs/locale/nl/LC_MESSAGES/mqtt.mo                 |  Bin 2581 -> 2594 bytes
[... 97 lines cut ...]
 * Reloading process: dsmr_backend                   [OK]
 * Reloading process: dsmr_datalogger                [OK]
 * Reloading process: dsmr_mqtt                      [??] PID file does not exist

 --- Clearing cache...
Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 357, in execute
    django.setup()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/__init__.py", line 24, in setup
    apps.populate(settings.INSTALLED_APPS)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/apps/registry.py", line 91, in populate
    app_config = AppConfig.create(entry)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/apps/config.py", line 90, in create
    module = import_module(entry)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'rangefilter'
[!] Halted. Post-deployment script exited with non-zero code

quadrophenianl, 2020-06-23:
Update:

Aangezien er foutmeldingen bleven komen met betrekking tot permissies, heb ik zelf wat permissies aangepast en vervolgens door herhaaldelijke foutmeldingen ook handmatig de 'niet toegankelijke' mappen aangemaakt. Dit leidde vervolgens tot een installatie zonder rode foutmeldingen, echter worden er aan het eind van de rit alsnog twee PID files niet gevonden, met als gevolg dat de applicatie nog altijd niet draait. Hierbij de log van de meer 'succesvolle' deploy:

--- Checking Python version.
Running Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0]

 --- Pulling remote repository for new commits...

 --- Merging/updating checkout.
Already up to date.

 >>> Running post-deployment script. <<<

 --- Checking whether VirtualEnv is activated.

 --- Checking Python version.
Running Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0]

 --- Checking & synchronizing base requirements for changes.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: coreapi==2.3.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 1)) (2.3.3)
Requirement already up-to-date: crcmod==1.7 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 2)) (1.7)
Collecting django==3.0.7
  Downloading https://files.pythonhosted.org/packages/5c/63/6d7efecbf3f06db8c6577950a24a191e55cadf7cda4d7fe6976206c886dd/Django-3.0.7-py3-none-any.whl (7.5MB)
     |████████████████████████████████| 7.5MB 1.8MB/s 
Collecting django-admin-rangefilter==0.6.1
  Downloading https://files.pythonhosted.org/packages/3d/4e/28dbf28aa6412bbf0e0af62e11f07c454c809a4f1163731dd2876cad5151/django_admin_rangefilter-0.6.1-py2.py3-none-any.whl
Collecting django-colorfield==0.3.0
  Downloading https://www.piwheels.org/simple/django-colorfield/django_colorfield-0.3.0-py3-none-any.whl
Collecting django-debug-toolbar==2.2
  Downloading https://files.pythonhosted.org/packages/2c/73/45de34aa56facb61bca5458efda75bcd09730a3beea747a7f51ace945375/django_debug_toolbar-2.2-py3-none-any.whl (198kB)
     |████████████████████████████████| 204kB 12.9MB/s 
Collecting django-filter==2.3.0
  Downloading https://files.pythonhosted.org/packages/35/94/e516d7f41adecb9fd280c89feceb98fa58f6966a583035c8016ed455c4e5/django_filter-2.3.0-py3-none-any.whl (73kB)
     |████████████████████████████████| 81kB 7.4MB/s 
Requirement already up-to-date: djangorestframework==3.11 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 8)) (3.11.0)
Requirement already up-to-date: django-solo==1.1.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 9)) (1.1.3)
Collecting dropbox==10.2.0
  Downloading https://files.pythonhosted.org/packages/a0/86/5ea3466089883b0c3caff0b314ffc1b98c96f417e26903292b4a1e40342c/dropbox-10.2.0-py3-none-any.whl (668kB)
     |████████████████████████████████| 675kB 10.8MB/s 
Collecting gunicorn==20.0.4
  Downloading https://files.pythonhosted.org/packages/69/ca/926f7cd3a2014b16870086b2d0fdc84a9e49473c68a8dff8b57f7c156f43/gunicorn-20.0.4-py2.py3-none-any.whl (77kB)
     |████████████████████████████████| 81kB 10.3MB/s 
Requirement already up-to-date: paho-mqtt==1.5.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 12)) (1.5.0)
Requirement already up-to-date: pyserial==3.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 13)) (3.4)
Collecting pyserial-asyncio==0.4
  Downloading https://files.pythonhosted.org/packages/d5/5e/518ace81b212db9a6458bf86ae9626c850bdc06ab133edd11f9b2a29ec47/pyserial_asyncio-0.4-py3-none-any.whl
Requirement already up-to-date: python-dateutil==2.8.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 15)) (2.8.1)
Collecting pytz==2020.1
  Downloading https://files.pythonhosted.org/packages/4f/a4/879454d49688e2fad93e59d7d4efda580b783c745fd2ec2a3adf87b0808d/pytz-2020.1-py2.py3-none-any.whl (510kB)
     |████████████████████████████████| 512kB 8.1MB/s 
Collecting requests==2.23.0
  Downloading https://files.pythonhosted.org/packages/1a/70/1935c770cb3be6e3a8b78ced23d7e0f3b187f5cbfab4749523ed65d7c9b1/requests-2.23.0-py2.py3-none-any.whl (58kB)
     |████████████████████████████████| 61kB 9.5MB/s 
Collecting urllib3==1.25.9
  Downloading https://files.pythonhosted.org/packages/e1/e5/df302e8017440f111c11cc41a6b432838672f5a70aa29227bf58149dc72f/urllib3-1.25.9-py2.py3-none-any.whl (126kB)
     |████████████████████████████████| 133kB 11.6MB/s 
Requirement already satisfied, skipping upgrade: coreschema in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (0.0.4)
Requirement already satisfied, skipping upgrade: uritemplate in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (3.0.1)
Requirement already satisfied, skipping upgrade: itypes in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (1.1.0)
[... cut ...]
  No migrations to apply.

 --- Checking & synchronizing static file changes.

137 static files copied to '/var/www/dsmrreader/static', 420 unmodified.

 --- Reloading running apps...

 * Reloading process: dsmr_webinterface (Gunicorn)   [??] PID file does not exist
 * Reloading process: dsmr_backend                   [OK]
 * Reloading process: dsmr_datalogger                [OK]
 * Reloading process: dsmr_mqtt                      [??] PID file does not exist

 --- Clearing cache...

 --- Deployed version: 
3.11

quadrophenianl, 2020-06-23:
Hoppakee, ik ben weer draaiende... Voor degene die in de toekomst tegen dezelfde problemen aanlopen, dit is in een notendop wat ik gedaan heb.

- De DSMR en onderliggende submappen volledige **'open'** gezet voor iedereen;
- Mocht het vervolgens niet lukken om de deployment tot een goed einde te brengen, probeer dan de mappen welke zichtbaar zijn in de foutmelding handmatig aan te maken. Wellicht eerst de naam aanpassen van de bestaande mappen, want bij mij waren ze 'corrupt';
- ./deploy.sh nogmaals doorlopen:
- Geen idee of deze stappen zinvol waren, maar ik heb ze doorlopen: https://dsmr-reader.readthedocs.io/nl/v3/mqtt.html;
- Tot slot dit commando uitgevoerd: **sudo supervisorctl restart all**.

Vervolgens startte de applicatie weer netjes op. Wel lijkt het me handig om uiteindelijk de permissies weer in te perken na succesvolle installatie.

Wellicht heeft iemand iets aan bovenstaande stappen!

Dennis, nogmaals dank en zet het goede werk voort!

dennissiemensma, 2020-06-23:
Bedankt voor je update. Het deployscript herstart de processen alleen als ze al draaien, ik zal dat duidelijker maken in de melding.

Je kunt trouwens alle permissies standaard laten zoals ze bij de installatie waren. DSMR-reader draait standaard onder een eigen user in een eigen homedirectory. Het heeft geen toegevoegde waarde om die rechten in te perken. Vandaar de `chown` in mijn eerdere comment.

Je kunt codeblokken maken met drie backticks (`): https://guides.github.com/features/mastering-markdown/

closed 2020-06-23 by dennissiemensma