dsmrreader/dsmr-reader #1028: Database vol na stroomstoring

bartje006, 2020-06-19:
Hoi,

Ik ben een simpele gebruiker van DSMR reader op een Raspberry Pi. Werkte perfect, tot woensdag. Toen viel de stroom uit. Daarna werkte de Pi nog wel, maar DSMR niet meer. 

Toen kreeg ik dezelfde foutmelding als hier: https://github.com/dennissiemensma/dsmr-reader/issues/476. Dus toen heb ik het commando uitgevoerd om de postgresql db te herstarten. Maar toen kreeg ik 502 bad gateway. 

Pihole werkt nog wel, maar mijn SD kaart is helemaal vol. Resultaat van:  sudo du -xh / | grep -P "G\t":

`11G     /home/dsmr/dsmr-reader/backups
11G     /home/dsmr/dsmr-reader
11G     /home/dsmr
11G     /home
11G     /var/lib/postgresql/11/main/base/16385
11G     /var/lib/postgresql/11/main/base
11G     /var/lib/postgresql/11/main
11G     /var/lib/postgresql/11
11G     /var/lib/postgresql
11G     /var/lib
12G     /var
2.9G    /usr/lib
1.2G    /usr/share
4.4G    /usr
28G     /
`

Enig idee hoe ik dat kan oplossen? 

Bedankt!

bartje006, 2020-06-19:
Inmiddels kan ik DSMR weer bereiken, maar misschien kan ik nog wat weggooien?
Er staan in de backups folder 7 .gz bestanden (ieder 500 MB) voor iedere dag van de week. Maar van vrijdag staat er ook nog een sql van 6.7 GB. Kan die bv weg?

dennissiemensma, 2020-06-19:
Bedankt voor je melding. Je kunt rustig oude backups weggooien en ook die halve van vandaag. Bewaar er wel even een ergens voor de zekerheid.

Voor het structureer oplossen kun je data automatisch laten opschonen na een X-periode: https://dsmr-reader.readthedocs.io/nl/v3/admin/datalogger.html#data-cleanup-retention

Verder, mocht je DSMR v5 hebben, kun je in de admin bij de datalogger config de 'datalogger sleep' verhogen tot zo'n 5 a 10 seconden. Dat vermindert de hoeveelheid data ook erg.

closed 2020-06-24 by dennissiemensma