dsmrreader/dsmr-reader #1029: Totaal verbuik kWh in archief weergave

killabeenl, 2020-06-19:
Graag zou ik in de archief weergave een totaal willen zien van het aantal kWh.
Nu zie je bijvoorbeeld in het overzicht van 2020:

Verbruik:        Teruggeleverd:          Kosten:
1000               1000                         0

Zou daar een kolom bij kunnen komen zodat je het aantal verbruik kWh minus teruggeleverd kWh doet?
Op die manier kan je zien hoeveel de balans is over het jaar in het aantal kWh's. Dit zodat je weet hoeveel buffer je hebt voor de rest van het jaar.

2xdehelft, 2020-06-19:
Dit zou inderdaad wel heel handig zijn, ik probeer ook altijd aan het eind van een contractjaar op 0KW uit te komen. (electrisch bijverwarmen)

dennissiemensma, 2020-06-19:
Bedankt voor je verzoek. Het archief is bedoeld voor kalenderjaren en dat gaat dan denk ik niet altijd werken voor alle energiecontracten, die verschillende looptijden kunnen hebben.

Je kunt de gegevens die je zoekt vermoedelijk het beste bekijken bij het menu met alle Energiecontracten. Daar staan die gegevens volgens mij ook en kun je vrij makkelijk zien wat de tussenstand is.

dennissiemensma, 2020-08-03:
Ik heb hier naar gekeken. Ik begrijp de wens, maar zowel voor kalenderjaren via het Archief als de totalen op contract-basis is dit al goed in te zien. 
Het is mij duidelijk dat een automatische teller handig is, maar ik zie persoonlijk niet in wat dit toevoegt omdat je dit net zo goed in 1 oogopslag kan zien in beide pagina's door te kijken naar de jaar/contract-totalen daar. Ze staan al letterlijk naast elkaar en zijn ook nog eens opgeteld, ongeacht tarief.

### Contractjaar
[image]

### Kalenderjaar in Archief
[image]

closed 2020-08-03 by dennissiemensma