dsmrreader/dsmr-reader #1030: Monitoring DSMR-reader en datalogger

mrwolf78, 2020-06-19:
Ik heb de DSMR datalogger op een Raspberry Pi Model B rev2 draaien, en DSMR-reader op een Raspberry Pi 2 Model B in een Docker-container.

Ik ben zoek naar een manier om beiden te monitoren, en bij eventuele fouten een mail te sturen of een pushbericht. Zijn er mensen die dit ingericht hebben, en zo ja hoe? Ik zit te denken aan een mechanisme dat als er langer dan x minuten geen nieuwe data is gelezen, er een actie wordt uitgevoerd, maar wellicht heeft iemand al iets veel slimmers bedacht.

dennissiemensma, 2020-06-19:
Bedankt voor je vraag. Als je de mogelijkheid hebt om de ingebouwde notificaties via Pushover of Prowl te gebruiken, dan wordt er ook een notificatie verstuurd wanneer de datalogger er X aantal uren uit ligt. Daarbij ben je wel afhankelijk van dat DSMR-reader zelf goed draait, dus dat kan niet altijd gewenst zijn.

mrwolf78, 2020-06-24:
Ik heb het nu via Telegram werkend (dat was nog even gedoe), dus dat is in ieder geval een start, dank je

closed 2020-06-24 by dennissiemensma