dsmrreader/dsmr-reader #1031: dsmr_webinterface FATAL Exited too quickly (process log may have details)

bvansluis, 2020-06-21:
I had a short power outage last night and now the web interface isn't working any more. When I run the command sudo supervisorctl status I get the error: 
dsmr_webinterface FATAL Exited too quickly (process log may have details)

What can I do to resolve this error? I have tried the following:
- Restart the services
- Reboot the Raspberry
- Upgrade DSMR to the latest version

I'm not able to access the log files, because I always get an access denied error. With the DSMR user as well as with my own admin user.

dennissiemensma, 2020-06-22:
Thanks for reporting. You should be able to access the logfiles since you've also installed DSMR-reader and since you're able to run `sudo supervisorctl status`.

Try `sudo -s` when logged in as `pi` user on your Pi to become super user. Or simply add `sudo` to the command you're executing to view the logfiles.

bvansluis, 2020-06-24:
Not sure what I'm doing wrong or what is wrong with my instance, but I keep getting the access denied error:
[image]
I use this command to open the log files. Is that correct? 

`/var/log/supervisor/dsmr_webinterface.log`

dennissiemensma, 2020-06-24:
You should prefix the commands with `sudo` (when you're the `pi` user) to indicated superuser privileges.

Also, you will need to specify the command for viewing the logfiles. You can use any command line text editor (e.g.: `view`) or use `tail` or `cat`. See https://www.faqforge.com/linux/distributions/debian/linux-how-to-view-log-files-on-the-shell/

bvansluis, 2020-06-24:
Great, got it! I get this error from the log file: 
Error: Already running on PID 476 (or pid file '/var/tmp/gunicorn--dsmr_webinterface.pid' is stale)

dennissiemensma, 2020-06-24:
You can check whether the process is still up with:

```
sudo ps faux | grep 476 | grep -v grep
```

If it states something similar (make sure it's not another proces!):
```
dsmr     476  0.0  0.8  24532 16580 ?        S    Jun22   0:23  \_ /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi
```

Kill it:
```
kill -9 476
```

If there was no output or it looks like a non-DSMR-reader process, let it be and do no kill it.

Finally, remove the PID file
```
sudo rm /var/tmp/gunicorn--dsmr_webinterface.pid
```

And try again.

bvansluis, 2020-06-24:
I get this result after running `sudo ps faux | grep 476 | grep -v grep`

```
root       465  0.1  0.9  22476  9440 ?        Ss   Jun21   4:33 /usr/bin/python2 /usr/bin/supervisord -n -c /etc/supervisor/supervisord.conf
www-data   476  0.0  0.1  50324  1328 ?        S    Jun21   0:00  \_ nginx: worker process
postgres  7996  1.6  2.0 205664 19476 ?        Ss   12:03   0:01  \_ postgres: 11/main: dsmrreader dsmrreader ::1(48544) idle
```
So I guess I don't kill the process but just remove the PID file and restart the services?

dennissiemensma, 2020-06-24:
Yeah try the latter.

bvansluis, 2020-06-24:
Yes! Back online. Thanks a lot Dennis for all the support!

dennissiemensma, 2020-06-24:
Fijn om te horen!

closed 2020-06-24 by dennissiemensma

dennissiemensma, 2026-02-09:
Ik denk dat DSMR-reader v5 te oud is voor de native installatie op Python 3.13. Je kunt *naast* je huidige herinstallatie proberen of je de container-variant werkend krijgt, dat scheelt je uiteindelijk ook een hoop gedoe als het eenmaal draait. Gebruik dan wel :latest ( = nog v5) ipv :development (= v6). https://dsmr-reader.readthedocs.io/en/development/how-to/upgrade/to-v6/

dennissiemensma, 2026-02-14:
Inmiddels is v6 ook beschikbaar als versie in de container. Als je `:6` gebruikt ipv `:development` dan zit je later goed.

bennybubble, 2026-02-14:
Zal ik doen, bedankt voor de tip!