dsmrreader/dsmr-reader #1033: Datalogger start niet

TheMClaeys, 2020-06-22:
Wat gebruik je?
- DSMR-reader versie:   v3.11
- Type RaspberryPi or server:   Debian 10 VM op ESXi
- Standaardinstallatie of Docker:   Standaardinstallatie
- DSMR-protocol slimme meter v4/v5:   v4? (Fluvius België)

---
Ik heb mijn installatie gemigreerd naar een VM op ESXi. Mijn Raspberry PI had problemen dat hij na enkele dagen telkens niet meer bereikbaar was tot een powercycle waardoor veel te vaak logging verloren ging.
De installatie ging vlot, de database backup terugzetten ook. Nergens foutmeldingen gekregen of gezien.
Het probleem is echter dat er geen live data binnenkomt.

Manueel lukt het:
```
(dsmrreader) dsmr@dsmr:~/dsmr-reader$ cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q
Connected.

/FLU5\253770234_A

0-0:96.1.4(50213)
0-0:96.1.1(3153XXXXXXXXXXXXXXXXXXXXX532)
0-0:1.0.0(200622080217S)
1-0:1.8.1(000638.465*kWh)
1-0:1.8.2(001130.177*kWh)
1-0:2.8.1(001973.435*kWh)
1-0:2.8.2(000685.623*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.079*kW)
1-0:2.7.0(00.000*kW)
1-0:32.7.0(229.9*V)
1-0:31.7.0(001*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(3746XXXXXXXXXXXXXXXXXXXXX539)
0-1:24.4.0(1)
0-1:24.2.3(200622080149S)(00522.673*m3)
!F6AE
/FLU5\253770234cu: Got hangup signal

Disconnected.
```

De datalogger service switched tussen STARTING en RUNNING (en ik denk de mqtt ook omdat de uptime daar ook vrij kort is telkens)
```
root@dsmr:~# sudo supervisorctl status
dsmr_backend                     RUNNING   pid 28789, uptime 3 days, 10:06:06
dsmr_datalogger                  RUNNING   pid 29224, uptime 0:00:01
dsmr_mqtt                        RUNNING   pid 29197, uptime 0:00:09
dsmr_webinterface                RUNNING   pid 3803, uptime 6 days, 13:33:25
root@dsmr:~# sudo supervisorctl status
dsmr_backend                     RUNNING   pid 28789, uptime 3 days, 10:06:27
dsmr_datalogger                  RUNNING   pid 29309, uptime 0:00:01
dsmr_mqtt                        RUNNING   pid 29197, uptime 0:00:30
dsmr_webinterface                RUNNING   pid 3803, uptime 6 days, 13:33:46
root@dsmr:~# sudo supervisorctl status
dsmr_backend                     RUNNING   pid 28789, uptime 3 days, 10:06:33
dsmr_datalogger                  STARTING
dsmr_mqtt                        RUNNING   pid 29197, uptime 0:00:36
dsmr_webinterface                RUNNING   pid 3803, uptime 6 days, 13:33:52
root@dsmr:~# sudo supervisorctl status
dsmr_backend                     RUNNING   pid 28789, uptime 3 days, 10:06:37
dsmr_datalogger                  RUNNING   pid 29351, uptime 0:00:01
dsmr_mqtt                        RUNNING   pid 29197, uptime 0:00:40
dsmr_webinterface                RUNNING   pid 3803, uptime 6 days, 13:33:56
root@dsmr:~# sudo supervisorctl status
dsmr_backend                     RUNNING   pid 28789, uptime 3 days, 10:06:39
dsmr_datalogger                  RUNNING   pid 29361, uptime 0:00:01
dsmr_mqtt                        RUNNING   pid 29197, uptime 0:00:42
dsmr_webinterface                RUNNING   pid 3803, uptime 6 days, 13:33:58
root@dsmr:~# sudo supervisorctl status
dsmr_backend                     RUNNING   pid 28789, uptime 3 days, 10:06:41
dsmr_datalogger                  STARTING
dsmr_mqtt                        RUNNING   pid 29197, uptime 0:00:44
dsmr_webinterface                RUNNING   pid 3803, uptime 6 days, 13:34:00
```

Ik heb logging op debug gezet, echter blijven de logs tonen dat het logging niveau onvoldoende is:
```
root@dsmr:~# tail /var/log/supervisor/dsmr_datalogger.log
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html#logging
```
[image]

Wat ik wel gemerkt heb als ik dmesg typ, dat ik de melding 
`ttyUSB ttyUSB0: 1 input overrun(s)`
Dit ben ik momenteel nog aan het uitzoeken wat hier het probleem is. Gezien het manueel werkt ben ik niet zeker of het gerelateerd is.

Iemand hier ervaring mee? Ik zag vorige week een issue hier op Github met datalogger die niet wou starten als mqtt aanstond, maar initieel stond mqtt niet aan en werkte het ook niet met hetzelfde probleem.

dennissiemensma, 2020-06-22:
Bedankt voor je melding. Die meldingen over logging kunnen nog van eerder zijn, maar desnietemin vreemd dat je helemaal niets ziet.

Zie je bij live grafieken ook daadwerkelijk nieuwe data verschijnen? Gezien de X-as geen datums toont, maar alleen tijden, kan het toevallig overeen komen met twee maanden terug. Ook omdat er linksbovenin staat dat de laatste data van twee maanden terug is.

Wat je sowieso even kunt checken is `/admin/dsmr_datalogger/meterstatistics/1/change/` en dan onderaan de pagina. Daar staat het laatst gelezen telegram. Bij mij is de regel met `0-0:1.0.0` nu `0-0:1.0.0(200622195414S)` (2020-06-22 19:45:14, zomertijd). Je kunt kijken of de datum bij jou ook begint met `(200622...)`.

----

Verder kun je eens proberen of je handmatig kunt uitlezen via DSMR-reader zelf:
```
sudo su - dsmr

./manage.py dsmr_datalogger --run-once
```

Dat commando kan 10 seconden duren, afhankelijk van hoe snel het volgende telegram binnenkomt. Het kan zijn dat je een melding krijgt als: 
```
serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```
In dat geval kun je het beste even in Supervisor de datalogger stoppen (anders zitten ze elkaar in de weg) met `sudo supervisorctl stop dsmr_datalogger`. Vergeet niet hem erna weer aan te zetten.

----

En tot slot kun je nog even kijken of je in de datalogger configuratie in de admin-interface wel de juiste versie hebt geselecteerd, gezien je aangeeft dat je een niet-NL meter hebt maar Fluvius. Misschien moet je dit zelfs als eerste even checken.

TheMClaeys, 2020-06-22:
Bedankt voor het antwoord!

De graffiek loopt gewoon verder, of het live data is weet ik niet. Op dit moment lijkt het wel in de buurt te komen, zowel op mijn zonnepanelen als men invoer heb ik ook een Z-wave meetklem en als je ze naast elkaar legt:
[image]
Niet letten op het lege Dagelijks verbruik Elektriciteit, dat is een plot van wat ik met MQTT zou binnenkrijgen, vandaar geen history op de moment.
Maar de overeenkomsten kunnen dus puur toeval zijn, de graffiek verschuift wel, maar misschien plot hij gewoon oude data van x-tijd geleden.

Latest Telegram is leeg bij mij op die pagina. Het ID is 1 en de timestamp is 13 juni 2020 1:23 p.m, maar de rest van de pagina is dus leeg.

Het manueel uitlezen lukt deels:
```
(dsmrreader) dsmr@dsmr:~/dsmr-reader$ ./manage.py dsmr_datalogger --run-once
INFO:dsmrreader:Received telegram (base64 encoded): b'L0ZMVTVcMjUzNzcwMjM0X0ENCg0KMC0wOjk2LjEuNCg1MDIxMykNCjAtMDo5Ni4xLjEoMzE1MzQxNDczMTMxMzAzMDMwMzUzOTMwMzUzMikNCjAtMDoxLjAuMCgyMDA2MjIyMDQ5NTBTKQ0KMS0wOjEuOC4xKDAwMDY0MC45NzAqa1doKQ0KMS0wOjEuOC4yKDAwMTEzMC4xNzcqa1doKQ0KMS0wOjIuOC4xKDAwMjAwMS45OTcqa1doKQ0KMS0wOjIuOC4yKDAwMDY4NS42MjMqa1doKQ0KMC0wOjk2LjE0LjAoMDAwMSkNCjEtMDoxLjcuMCgwMC40OTEqa1cpDQoxLTA6Mi43LjAoMDAuMDAwKmtXKQ0KMS0wOjMyLjcuMCgyMzEuMCpWKQ0KMS0wOjMxLjcuMCgwMDIqQSkNCjAtMDo5Ni4zLjEwKDEpDQowLTA6MTcuMC4wKDk5OS45KmtXKQ0KMS0wOjMxLjQuMCg5OTkqQSkNCjAtMDo5Ni4xMy4wKCkNCjAtMToyNC4xLjAoMDAzKQ0KMC0xOjk2LjEuMSgzNzQ2NEM0RjMyMzEzMTM5MzAzMzM3MzAzNTM5KQ0KMC0xOjI0LjQuMCgxKQ0KMC0xOjI0LjIuMygyMDA2MjIyMDQ1NTVTKSgwMDUyMi44NDgqbTMpDQohMEI0OA0K'
[2020-06-22 20:49:27,265] DEBUG    Received telegram:
/FLU5\253770234_A

0-0:96.1.4(50213)
0-0:96.1.1(3153XXXXXXXXXXXXXXXXXXXXX532)
0-0:1.0.0(200622204950S)
1-0:1.8.1(000640.970*kWh)
1-0:1.8.2(001130.177*kWh)
1-0:2.8.1(002001.997*kWh)
1-0:2.8.2(000685.623*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.491*kW)
1-0:2.7.0(00.000*kW)
1-0:32.7.0(231.0*V)
1-0:31.7.0(002*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(3746XXXXXXXXXXXXXXXXXXXXX539)
0-1:24.4.0(1)
0-1:24.2.3(200622204555S)(00522.848*m3)
!0B48

DEBUG:commands:Received telegram:
/FLU5\253770234_A

0-0:96.1.4(50213)
0-0:96.1.1(3153XXXXXXXXXXXXXXXXXXXXX532)
0-0:1.0.0(200622204950S)
1-0:1.8.1(000640.970*kWh)
1-0:1.8.2(001130.177*kWh)
1-0:2.8.1(002001.997*kWh)
1-0:2.8.2(000685.623*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.491*kW)
1-0:2.7.0(00.000*kW)
1-0:32.7.0(231.0*V)
1-0:31.7.0(002*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(3746XXXXXXXXXXXXXXXXXXXXX539)
0-1:24.4.0(1)
0-1:24.2.3(200622204555S)(00522.848*m3)
!0B48

[2020-06-22 20:49:27,468] ERROR    dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 90, in run_once
    self.run(data=self.data, **options)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 21, in run
    dsmr_datalogger.services.datalogger.read_and_process_telegram()
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services/datalogger.py", line 73, in read_and_process_telegram
    dsmr_datalogger.services.datalogger.telegram_to_reading(data=telegram)
[... 37 lines cut ...]
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 433, in create
    obj.save(force_insert=True, using=self.db)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 746, in save
    force_update=force_update, update_fields=update_fields)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 795, in save_base
    update_fields=update_fields, raw=raw, using=using,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/dispatch/dispatcher.py", line 175, in send
    for receiver in self._live_receivers(sender)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/dispatch/dispatcher.py", line 175, in <listcomp>
    for receiver in self._live_receivers(sender)
  File "/home/dsmr/dsmr-reader/dsmr_mqtt/apps.py", line 67, in _on_dsmrreading_created_signal
    instance = DsmrReading.objects.get(pk=instance.pk)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 422, in get
    num if not limit or num < limit else 'more than %s' % (limit - 1),
dsmr_datalogger.models.reading.DsmrReading.MultipleObjectsReturned: get() returned more than one DsmrReading -- it returned 2!

[2020-06-22 20:49:27,468] INFO     dsmr_datalogger.management.commands.dsmr_datalogger: Exiting on next run...
INFO:commands:dsmr_datalogger.management.commands.dsmr_datalogger: Exiting on next run...
[2020-06-22 20:49:27,468] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Exited
DEBUG:commands:dsmr_datalogger.management.commands.dsmr_datalogger: Exited
```

Op de admin pagina staat de meter op Belgium - Fluvius (gas meter fix). Deze setting heeft hij meegenomen bij het terugzetten van de backup denk ik, net als enkele andere settings, want zo stond het ook voor de migratie en er waren nog wel enkele settings die me opvielen dat ze al juist stonden bij het omzetten.

dennissiemensma, 2020-06-22:
Voor de verwerking kun je het beste op de Status-pagina kijken. Dan heb je uitsluitsel of die wel of niet achterloopt. Los van binnenkomende telegrammen. Als die grafiek telkens overeenkomt met de huidige tijd, dan zal het deels wel kloppen.

Verder is het opmerkelijk dat je die fout ` get() returned more than one DsmrReading` krijgt. Ik zie niet in hoe dat eerlijk gezegd gebeurt. Had je het datalogger-proces in Supervisor nog uitgezet toen je het handmatig draaide?

TheMClaeys, 2020-06-23:
Dag Dennis,

Ik heb beiden getest, met de datalogger stopped en met de datalogger nog aan. De error blijft hetzelfde:
`dsmr_datalogger.models.reading.DsmrReading.MultipleObjectsReturned: get() returned more than one DsmrReading -- it returned 2!`

De statuspagina is een mix van groen en rood:
[image]

dennissiemensma, 2020-06-23:
Ik weet niet of dat helpt dat de data niet klopt, maar technisch gezien zou het onmogelijk moeten zijn dat er twee dezelfde ID's zijn.

Kun je dit eens doen om te kijken of er dubbele ID's zijn?

```
sudo su - postgres
psql dsmrreader

select id, count(id) as aantal from dsmr_datalogger_dsmrreading group by id having count(id) > 1;
```

dennissiemensma, 2020-06-23:
Verder kun je DEBUG-logging (weer) aanzetten en in de logfile kijken van `dsmr_backend` om te zien waar een deel van de dataprocessing op wacht. Ik vermoed dat het ook komt omdat er geen nieuwe data is.

TheMClaeys, 2020-06-24:
Dag Dennis,

Er zijn blijkbaar dubbele ID's, of interpreteer ik dit mis?
```
   id   | aantal 
--------+--------
      1 |      2
      2 |      2
      3 |      2
      4 |      2
      5 |      2
      6 |      2
      7 |      2
      8 |      2
      9 |      2
     10 |      2
     11 |      2
     12 |      2
     13 |      2
     14 |      2
     15 |      2
     16 |      2
     17 |      2
     18 |      2
     19 |      2
     20 |      2
     21 |      2
     22 |      2
     23 |      2
     24 |      2
     25 |      2
     26 |      2
     27 |      2
     28 |      2
     29 |      2
     30 |      2
     31 |      2
     32 |      2
     33 |      2
     34 |      2
     35 |      2
     36 |      2
     37 |      2
     38 |      2
     39 |      2
     40 |      2
     41 |      2
     42 |      2
     43 |      2
     44 |      2
     45 |      2
     46 |      2
     47 |      2
     48 |      2
     49 |      2
     50 |      2
     51 |      2
     52 |      2
     53 |      2
     54 |      2
     55 |      2
     56 |      2
     57 |      2
     58 |      2
     59 |      2
     60 |      2
     61 |      2
     62 |      2
     63 |      2
     64 |      2
     65 |      2
     66 |      2
:
```

Debug logging staat nog steeds aan.
Dit staat in de dsmr_backend log:
```
root@dsmr:~# tail /var/log/supervisor/dsmr_backend.log
[2020-06-24 07:35:09,564] DEBUG    SP: 0 backend service(s) ready to run
[2020-06-24 07:35:09,568] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-06-24 07:35:10,571] DEBUG    SP: 0 backend service(s) ready to run
[2020-06-24 07:35:10,575] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-06-24 07:35:11,578] DEBUG    SP: 0 backend service(s) ready to run
[2020-06-24 07:35:11,581] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-06-24 07:35:12,584] DEBUG    SP: 0 backend service(s) ready to run
[2020-06-24 07:35:12,588] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-06-24 07:35:13,591] DEBUG    SP: 0 backend service(s) ready to run
[2020-06-24 07:35:13,595] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
root@dsmr:~# 
```

dennissiemensma, 2020-06-24:
Ik heb dit nog niet eerder gezien. Wellicht kun je deze query uitvoeren:

```
select id, timestamp, extra_device_timestamp, electricity_delivered_1 from dsmr_datalogger_dsmrreading order by id asc limit 10;
```

Ik ben benieuwd of de waardes van dezelfde ID's ook gelijk zijn. Dan weten we of het inhoudelijk dubbele records zijn of andere records met hetzelfde ID.

----

Hoe dan ook komt het er uiteindelijk op neer dat je de tabel (deels) moet leeggooien denk ik. Het is dan zaak om uit te zoeken of het een deel betreft of alle data:

```
select id, count(id) as aantal from dsmr_datalogger_dsmrreading group by id having count(id) > 1 order by id desc limit 5;

select id from dsmr_datalogger_dsmrreading order by id desc limit 5;
```

Je kunt de hoogste ID's van beide queries vergelijken om te kijken of het in de hele tabel zo is of tot een bepaalde hoogte.

----

Het verwerken van de dagstatistieken wacht overigens inderdaad op nieuwe data: `Compact: Waiting for newer readings before grouping data`.

TheMClaeys, 2020-06-24:
Dag Dennis,

Bedankt voor de queries. Ik heb deze uitgevoerd:
```

dsmrreader=# select id, timestamp, extra_device_timestamp, electricity_delivered_1 from dsmr_datalogger_dsmrreading order by id asc limit 10;
 id |       timestamp        | extra_device_timestamp | electricity_delivered_1 
----+------------------------+------------------------+-------------------------
  1 | 2020-02-11 22:21:44+01 | 2020-02-11 22:20:00+01 |                 279.896
  1 | 2020-06-13 13:22:15+02 | 2020-06-13 13:20:04+02 |                 617.670
  2 | 2020-06-13 13:22:18+02 | 2020-06-13 13:20:04+02 |                 617.670
  2 | 2020-02-11 22:21:45+01 | 2020-02-11 22:20:00+01 |                 279.896
  3 | 2020-02-11 22:21:46+01 | 2020-02-11 22:20:00+01 |                 279.896
  3 | 2020-06-13 13:22:20+02 | 2020-06-13 13:20:04+02 |                 617.670
  4 | 2020-02-11 22:21:47+01 | 2020-02-11 22:20:00+01 |                 279.896
  4 | 2020-06-13 13:22:22+02 | 2020-06-13 13:20:04+02 |                 617.670
  5 | 2020-06-13 13:22:24+02 | 2020-06-13 13:20:04+02 |                 617.670
  5 | 2020-02-11 22:21:48+01 | 2020-02-11 22:20:00+01 |                 279.896
(10 rows)

```

Als ik de 2 laatste goed begrijp zijn het precies niet de laatste ID's die het probleem hebben, de ID's komen niet overeen:

```
dsmrreader=# select id, count(id) as aantal from dsmr_datalogger_dsmrreading group by id having count(id) > 1 order by id desc limit 5;
   id   | aantal 
--------+--------
 357809 |      2
 357808 |      2
 357807 |      2
 357806 |      2
 357805 |      2
(5 rows)

dsmrreader=# select id from dsmr_datalogger_dsmrreading order by id desc limit 5;
   id    
---------
 4976066
 4976065
 4976064
 4976063
 4976062
(5 rows)
```

Het zou jammer zijn moest de data niet te redden zijn, maar dan is dat maar zo. Ik gok dat het op men VM toch stabieler zal draaien dan op men PI3b. Op de PI had ik al eender wat problemen, mogelijk hardware related.

dennissiemensma, 2020-06-24:
Ik denk dat het wel mee zal vallen qua hoeveelheid data die weg moet. Tussen deze ID's lijkt namelijk een aardig verschil in grootte, dus het is een relatief klein deel:
```
357809
4976066
```

Ik denk dat je dan:
- Een DB-backup moet maken zodat je de data nog wel ergens hebt. (en sowieso goed om een backup te hebben als je data gaat weggooien)

- Daarna alle ID's kleiner dan de dubbele moet weggooien:
```
delete from dsmr_datalogger_dsmrreading where id <= 357809;
```

Het zou zomaar kunnen dat dat het oplost en DSMR-reader misschien dan zelfs de eventuele achterstand inloopt, maar dat is geen belofte aangezien ik niet weet wat wel en niet opgeslagen is de afgelopen twee maanden.

dennissiemensma, 2020-06-24:
Bij nader inzien, zie ik nu dat de data gemixt is. Misschien is het nog de moeite waarde om iets gerichter op te schonen.

dennissiemensma, 2020-06-24:
Voer sowieso dit ook even uit:

```
sudo su - dsmr
./manage.py dsmr_sqlsequencereset
```

dennissiemensma, 2020-06-24:
Kun je deze query uitvoeren?

```
select id, timestamp, extra_device_timestamp, electricity_delivered_1, processed from dsmr_datalogger_dsmrreading order by id asc limit 20;
```

Mogelijk zit er nog een verschil in gegevens die wel of niet verwerkt zijn met `processed`. Als dat zo is, kunnen we daar mogelijk onderscheid in maken.

TheMClaeys, 2020-06-24:
Dag Dennis,

Dit is de output:
`dsmrreader=# select id, timestamp, extra_device_timestamp, electricity_delivered_1, processed from dsmr_datalogger_dsmrreading order by id asc limit 20;
 id |       timestamp        | extra_device_timestamp | electricity_delivered_1 | processed 
----+------------------------+------------------------+-------------------------+-----------
  1 | 2020-06-13 13:22:15+02 | 2020-06-13 13:20:04+02 |                 617.670 | t
  1 | 2020-02-11 22:21:44+01 | 2020-02-11 22:20:00+01 |                 279.896 | t
  2 | 2020-02-11 22:21:45+01 | 2020-02-11 22:20:00+01 |                 279.896 | t
  2 | 2020-06-13 13:22:18+02 | 2020-06-13 13:20:04+02 |                 617.670 | t
  3 | 2020-06-13 13:22:20+02 | 2020-06-13 13:20:04+02 |                 617.670 | t
  3 | 2020-02-11 22:21:46+01 | 2020-02-11 22:20:00+01 |                 279.896 | t
  4 | 2020-02-11 22:21:47+01 | 2020-02-11 22:20:00+01 |                 279.896 | t
  4 | 2020-06-13 13:22:22+02 | 2020-06-13 13:20:04+02 |                 617.670 | t
  5 | 2020-06-13 13:22:24+02 | 2020-06-13 13:20:04+02 |                 617.670 | t
  5 | 2020-02-11 22:21:48+01 | 2020-02-11 22:20:00+01 |                 279.896 | t
  6 | 2020-02-11 22:21:49+01 | 2020-02-11 22:20:00+01 |                 279.896 | t
  6 | 2020-06-13 13:22:29+02 | 2020-06-13 13:20:04+02 |                 617.670 | t
  7 | 2020-06-13 13:22:31+02 | 2020-06-13 13:20:04+02 |                 617.670 | t
  7 | 2020-02-11 22:21:50+01 | 2020-02-11 22:20:00+01 |                 279.896 | t
  8 | 2020-02-11 22:21:51+01 | 2020-02-11 22:20:00+01 |                 279.896 | t
  8 | 2020-06-13 13:22:33+02 | 2020-06-13 13:20:04+02 |                 617.670 | t
  9 | 2020-06-13 13:22:35+02 | 2020-06-13 13:20:04+02 |                 617.670 | t
  9 | 2020-02-11 22:21:52+01 | 2020-02-11 22:20:00+01 |                 279.896 | t
 10 | 2020-02-11 22:21:53+01 | 2020-02-11 22:20:00+01 |                 279.896 | t
 10 | 2020-06-13 13:22:37+02 | 2020-06-13 13:20:04+02 |                 617.670 | t
(20 rows)`

dennissiemensma, 2020-06-24:
Als de data al verwerkt is (en ik bedenk me net dat dat ook op de Status pagina stond), dan kun je dit rustig uitvoeren:

-    Een DB-backup maken zodat je de data nog wel ergens hebt. (en sowieso goed om een backup te hebben als je data gaat weggooien)

-    Daarna alle ID's tot en met de hoogste dubbele weggooien:
```
delete from dsmr_datalogger_dsmrreading where id <= 357809;
```

Het zou zomaar kunnen dat dat het oplost en DSMR-reader misschien dan zelfs de eventuele achterstand inloopt, maar dat is geen belofte aangezien ik niet weet wat wel en niet opgeslagen is de afgelopen twee maanden.

En vergeet niet om die `dsmr_sqlsequencereset` van mijn eerdere comment uit te voeren. Die reset de autoincrement ID's.

TheMClaeys, 2020-06-24:
Dag Dennis,

Bedankt! Ik zie nu bovenaan terug live de waardes, bij status staat alles groen, de live graphs zijn leeg maar beginnen terug te vullen en de service blijft mooi runnen (op MQTT na, maar die configureer ik nog wel even opnieuw van 0 af aan).
[image]

Misschien was er iets mis gelopen bij het terugzetten van de backup, maar ik ben blij dat nu alles terug leesbaar is :)

dennissiemensma, 2020-06-24:
Bedankt voor je update en fijn om te horen. Zijn de dagtotalen nu ook met terugwerkende kracht gegenereerd voor de afgelopen twee maanden? Voor zover die data aanwezig was.

TheMClaeys, 2020-06-24:
Net even nagekeken, deze zitten er allemaal in. De enige periode die ontbreekt is de periode dat de Pi er uit lag en de VM nog niet op stond, dus dat is logisch. Ondertussen tonen de live graphs ook opnieuw netjes alles. De backups naar Dropbox zijn al die tijd gewoon blijven lopen. Ik dacht dat ze geen nut gingen hebben omdat ik geen live waarden zag, maar blijkbaar bevatten deze dus wel degelijk gewoon de data.

MQTT intussen ook opnieuw ingesteld, ik zie nu in de queue dat hij terug mqtt messages stuurt en de service blijft draaien. De payload is wel leeg.

`[2020-06-24 20:46:32,868] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 5588
[2020-06-24 20:46:32,868] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m5589), 'b'dsmr/meter-stats/long_power_failure_count'' (NULL payload)
[2020-06-24 20:46:32,869] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 5589
[2020-06-24 20:46:32,869] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m5590), 'b'dsmr/meter-stats/power_failure_count'' (NULL payload)
[2020-06-24 20:46:32,869] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 5590
[2020-06-24 20:46:32,870] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m5591), 'b'dsmr/meter-stats/dsmr_version'' (NULL payload)
[2020-06-24 20:46:32,870] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 5591
[2020-06-24 20:46:32,871] DEBUG    dsmr_mqtt.management.commands.dsmr_mqtt: Sleeping 1.0s
[2020-06-24 20:46:33,408] INFO     dsmr_mqtt.management.commands.dsmr_mqtt: Detected signal #1
[2020-06-24 20:46:33,408] INFO     dsmr_mqtt.management.commands.dsmr_mqtt: Exiting on next run...
[2020-06-24 20:46:33,872] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending DISCONNECT
[2020-06-24 20:46:33,872] DEBUG    MQTT: Paho client: on_disconnect(userdata, rc) None 0
[2020-06-24 20:46:33,872] DEBUG    dsmr_mqtt.management.commands.dsmr_mqtt: Exited
`
[image]

dennissiemensma, 2020-06-24:
Je zou eens kunnen kijken in `/admin/dsmr_datalogger/meterstatistics/1/change/` of in `/statistics`, maar ik vermoed dat die meterstatistieken daar ook leeg zijn. 

Ik weet niet zo 1-2-3 of die ook meekomen in telegrammen van je meter. Als ik de telegrammen bekijk die je eerder stuurde, lijken ze aanzienlijk korter dan degene die we in NL hebben.

En dat verschil zie je hier ook wel:
- https://github.com/dennissiemensma/dsmr-reader/blob/v3/dsmr_datalogger/tests/datalogger/test_fluvius.py#L23
- https://github.com/dennissiemensma/dsmr-reader/blob/v3/dsmr_datalogger/tests/datalogger/test_example_dsmr50.py#L19

In de test voor Fluvius zie ik ook dat ik het destijds allemaal op NULL heb gezet, vermoedelijk omdat ik ze niet kon vinden: https://github.com/dennissiemensma/dsmr-reader/blob/v3/dsmr_datalogger/tests/datalogger/test_fluvius.py#L92

closed 2020-06-24 by dennissiemensma