dsmrreader/dsmr-reader #1035: Migrate settings.py to env vars

dennissiemensma, 2020-06-23:
I found a neat little library for this: https://pypi.org/project/python-decouple/

Seems to work perfectly, allowing me to drop duplicate setting files, the inclusion hell and support env vars by default.

dennissiemensma, 2020-06-24:
It was quite a ride, but I think it's solid now.

closed 2020-06-24 by dennissiemensma

LarsVenema95, 2020-08-01:
@dennissiemensma I follow the guide and run "cp dsmrreader/provisioning/django/postgresql.py dsmrreader/settings.py" this command, The command line states that the files are missing. However, those two files are not available on the current [repo](https://github.com/dennissiemensma/dsmr-reader/tree/v4/dsmrreader/provisioning/django)?

On the [older repo](https://github.com/dennissiemensma/dsmr-reader/tree/v3/dsmrreader/provisioning/django) postgresql.py is available. Am I doing anything wrong?

dennissiemensma, 2020-08-01:
@LarsVenema95 I'm not sure what guide you are following. I cannot find a reference to the file in the v4 branch.

These should suffice:
- Upgrading from v3 to v4: https://dsmr-reader.readthedocs.io/en/v4/faq/v4_upgrade.html
- Installing v4 (quick): https://dsmr-reader.readthedocs.io/en/v4/installation/quick.html
- Installing v4 (explained): https://dsmr-reader.readthedocs.io/en/v4/installation/explained.html

LarsVenema95, 2020-08-01:
@dennissiemensma 
I was following the /latest/ guide: https://dsmr-reader.readthedocs.io/en/latest/installation/explained.html

I will try the v4 guide. That one will probably work. I let you know! :)

dennissiemensma, 2020-08-02:
@LarsVenema95 thanks for the heads up. For some reason I had to force ReadTheDocs to rebuild 'latest', even though it already pointed to v4. It's updated now.