dsmrreader/dsmr-reader #1038: 504 Gateway Time-out /status + retention vragen

D1StrX, 2020-06-24:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/v3/troubleshooting.html

Wat gebruik je?

DSMR-reader versie (links- of rechtsbovenin applicatie zichtbaar):
v3.11
Hardware (RaspberryPi 3, 4 of iets anders): RaspberryPi ...
RaspberryPi 3b
Native of Docker (Native is de installatie direct op je RaspberryPi):
Native

---

Kan helaas niet precies de vinger erop leggen. Maar sinds dat ik op v3.10/11 zit laad de pagina Status niet meer.
Krijg een melding dat de 504 Gateway Time-out is. Andere pagina's zoals API docs, Feeback, Help/FAQ laden wel weer na een reboot. [https://imgur.com/FJhMi2D](url) 

Andere vraag is;
Heb nu retention aangezet omdat mn 14,9gb backup opslag vol zit. Nu elke 2 dagen (3x nu in totaal) de vacuum draaien op de DB. Toch loopt het nog niet vlot af wat er aan ruimte wordt bespaard. Retention op 3 maanden staan, compressie op 2.
Zou ik nou nog handmatig backups kunnen verwijderen?

dennissiemensma, 2020-06-24:
Bedankt voor je vraag. Je kunt een timeout krijgen als de Pi het niet kan bijbenen. 

De meest voorkomende oorzaak is wanneer je een DSMR-5 meter hebt. Heb je die? 

Zo ja dan kun je ervoor kiezen om de datalogger wat trager te laten lopen, zodat die niet elke seconde een telegram probeert te lezen, maar bijvoorbeeld elke 5 of 10 seconden. Dat helpt ook mee in het beperken van de dataset grootte.
En dat sluit ook aan met dat je geen ruimte bespaard ziet wordt, omdat de data vrij snel oploopt. Wanneer de retentie bij is, verkleint de dataset namelijk ook niet meer. 

Klopt het dat je een v5 meter hebt? En anders zullen we verder moeten zoeken.

Backups kun je sowieso verwijderen als je zorgt dat je er altijd eentje hebt. Het is een sneeuwbal-effect als je veel data hebt, want de backups zijn dan ook relatief groot.

dennissiemensma, 2020-06-24:
Je kunt trouwens met de info uit onderstaande comment je DB-grootte zien voor je beeld. Ook staat mijn situatie erbij ter referentie (dat is wel DSMR-v4 met korte retentie).

https://github.com/dennissiemensma/dsmr-reader/issues/1034#issuecomment-647780961

D1StrX, 2020-06-24:
Bedankt voor je snelle reactie. Mijn meter zegt voorop ESMR 5. En kan ook herinneren dat het elke seconden uitleest. Mijn DB gaf 4,7gb aan. Dat dan wel op een usb stick draait met de maximale doorvoer snelheid van usb2.0 (~50mb/s). Maar toch is het niet voldoende. Heb wat backups verwijdert zodat nieuwe gemaakt konden worden. 
Maar hoe kan ik de datalogger vertragen?

D1StrX, 2020-06-28:
Top, heb alle mogelijke wijzigingen doorgevoerd. Alles werkt weer naar behoren/sneller, en de database lijkt een stuk kleiner geworden te zijn. Dus de Pi kan "voorlopig" weer vooruit.

closed 2020-06-28 by D1StrX

dennissiemensma, 2020-06-28:
Fijn om te horen, dank voor de terugkoppeling!