dsmrreader/dsmr-reader #1041: dsmr-reader als add-on in Home Assistant

lweijl, 2020-07-03:
Is er een mogelijkheid om deze fantastische applicatie als add-on binnen HomeAssistant te draaien is? Zo niet, zou ik deze mogelijkheid graag zien, dit zou naar mijn mening een welkome toevoeging zijn.

dennissiemensma, 2020-07-03:
Bedankt voor je vraag. Voor zover ik te horen heb gekregen is er eentje die je zou kunnen proberen: https://dsmr-reader.readthedocs.io/en/latest/home_assistant.html

closed 2020-07-03 by dennissiemensma

lweijl, 2020-07-03:
Bedankt voor je reactie Dennis. Documentatie had ik inderdaad gelezen, maar dit is niet waar ik op doel. Ik doel op een add-on zoals die van AdGuard, Node-RED, deCONZ of TasmoAdmin. Daarmee zou ik binnen Home Assistant je applicatie kunnen draaien, zonder daar zelf nog iets voor te hoeven doen.

dennissiemensma, 2020-07-03:
Bedankt voor je toevoeging. Ik gebruik zelf geen HA, dus het is voor mij soms wat gissen.

Ik denk dat het in dat geval wat lastig is om er een add-on voor te maken, gezien er nu voor DSMR-reader uberhaupt geen automatisch installatie-script is en ik verwacht dat dat de eerste stap is. Wellicht dat het wat makkelijker is om zoiets via Docker te installeren en draaien, maar daar zou dan iemand anders mee bezig moeten, die zowel daarover als van HA meer af weet dan ik. :]

trizz, 2020-08-03:
@lweijl Het 'probleem' is dat HA standaard geen Postgres addon heeft en DSMR-Reader weer geen MariaDB/MySQL ondersteund. Een DSMR-Reader addon zal dus ook moeten voorzien in een DB container. Daarnaast zullen bestaande Docker containers voor DSMR-Reader waarschijnlijk niet afdoende zijn, aangezien, voor hoever ik weet, HA Addons als basis weer een andere template hebben plus dat e.e.a. via HA ingesteld moet kunnen worden. En in het mooiste geval moet de DSMR-Reader-admin-interface werken via Ingress.

Voor mezelf zou ik een HA addon ook de mooiste oplossing vinden, wellicht dat ik me er eens aan ga wagen, naast de remote-datalogger, maar het ontwikkelen voor HA is voor mij nieuw terrein.

dennissiemensma, 2020-08-04:
@trizz je kunt rustig MySQL of MariaDB gebruiken voor DSMR-reader.

Vanaf DSMR-reader [v4.x is het makkelijker](https://dsmr-reader.readthedocs.io/en/latest/env_settings.html#db-engine) om dit in te stellen. Zie [bijvoorbeeld de tests](https://github.com/dennissiemensma/dsmr-reader/blob/v4/tools/test-all.sh#L44-L47) die standaard ook [in MySQL 5.x](https://github.com/dennissiemensma/dsmr-reader/blob/v4/.circleci/config.yml#L122) draaien.

Zorg er wel voor [dat je tijdzone-ondersteuning inschakelt in de DB](https://dsmr-reader.readthedocs.io/en/latest/faq.html#how-do-i-enable-timezone-support-for-mysql), dit staat standaard niet aan! Dat was namelijk de oorzaak van recente issues van andere gebruikers die ook niet-Postgres draaien (#974, #909).

Echter officieel ondersteun ik het niet omdat het behoorlijk wat werk is om het in alle versies te onderhouden. Plus dat het in het verleden een gedoe was met tijdzones, wat standaard in PostgreSQL zit, maar handmatig dus in MySQL/MariaDB ingeschakeld moet worden.