dsmrreader/dsmr-reader #1043: Probleempje na database restore

ph0real, 2020-07-06:
Hoi Dennis,

Onlangs is een USB schijfje gecrashed; helaas stond hier mijn postgreSQL db ook op.
Na wat gepuzzel heb ik een DB restore kunnen doen; ik heb de procedure van DSMR hiervoor opgevolgd. Echter krijg ik na het starten van DSMR nog steeds wat errors. De Postgresql DB is online en die log is vrij rustig. In de applicatie krijg ik onderstaande. Wat te doen?

Server Error

Sorry, something unexpected happened.
Exception:

MultipleObjectsReturned: get() returned more than one WeatherSettings -- it returned 2!
Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/handlers/base.py", line 113, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 71, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 97, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 158, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/dashboard.py", line 18, in get_context_data
    context_data['capabilities'] = dsmr_backend.services.backend.get_capabilities()

  File "/home/dsmr/dsmr-reader/dsmr_backend/services/backend.py", line 53, in get_capabilities
    'weather': WeatherSettings.get_solo().track and TemperatureReading.objects.exists()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/solo/models.py", line 54, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/query.py", line 559, in get_or_create
    return self.get(**kwargs), False

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/query.py", line 422, in get
    num if not limit or num < limit else 'more than %s' % (limit - 1),

dennissiemensma, 2020-07-06:
Bedankt voor je melding. Kan het zijn dat je eerst DSMR-reader helemaal hebt geinstalleerd en toen de backup teruggezet? In dat geval kun je dubbele data tegenkomen.

Je kunt proberen om die te verwijderen, maar als het te diep zit, kun je het wellicht het beste alsnog de backup terugzetten op een lege database.

Je kunt kijken of dit het oplost:
```
sudo su - postgresl
psql dsmrreader

select * from dsmr_weather_weathersettings;
```

Je krijgt daar als het goed is meer dan 1 terug. Verwijder vervolgens eentje (maakt niet uit welk ID):
```
delete from dsmr_weather_weathersettings where id = 2;
```

ph0real, 2020-07-07:
Het lijkt erop dat de database inderdaad niet leeg was. Dubbele entries met dezelfde ID's.
Wat ik heb gedaan is, de applicatie installatie intact gehouden, en omdat de database verloren was, die opnieuw aangemaakt. Maar ik vermoed dat ik daarna per ongeluk de applicatie gestart heb, waardoor de database gevuld werd. En daarna de restore, wat resulteerde in dubbele db items.

Ik ga opnieuw een lege DB regelen, en daarna weer de restore proberen.

ph0real, 2020-07-07:
En hij doet t nu wel weer. Bedankt voor het zetje in de juiste richting...

closed 2020-07-07 by ph0real