dsmrreader/dsmr-reader #1044: I can't get MQTT working :-(

bduijnhouwer, 2020-07-11:
At first, I installed the full DSMR suite on one Raspberry Pi. That went well and everything worked.
But now, I separated the datalogger from the reader, using two separate Raspberry Pi's.

Now I can't get the MQTT working. I verified that the process is running, but in the web interface I still
see: "0 Outgoing MQTT messages"

I used exactly the same credentials as in my first installation, so that's not the problem.
Does the MQTT process need to be running on the datalogger or on the reader?

dennissiemensma, 2020-07-11:
It depends on whether you see MQTT-messages in your MQTT broker. The "0 Outgoing MQTT messages" is normally what you'd like to see, as it indicates that it's not lagging behind. 

The Pi connected to the P1 port should [only run the datalogger client](https://dsmr-reader.readthedocs.io/en/latest/installation/datalogger.html). The webinterface, backend and MQTT processes should run on the full installation you're sending the telegrams to.

### MQTT
DSMR-reader stores outgoing MQTT messages in the database first, and the Status page indicates how many are stored. But simultaneously (if `dsmr_mqtt` is running) the `dsmr_mqtt` proces will read and send those stored MQTT messages to your MQTT broker. 
After doing that it will **delete** them from the database. Sometimes this even happens within the same second, so the Status page could still display `0`, which is good.

### Log
However, if you do not see any messages in the MQTT broker, a good start would be to enable DEBUG-logging on the reader: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#logging
And then check the `/var/log/supervisor/dsmr_mqtt.log` log to see what the process is doing and whether it's sending any messages at all.

Example:
```
[2020-07-11 11:51:49,541] DEBUG    MQTT: Initializing insecure connection (no TLS)
[2020-07-11 11:51:49,542] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending CONNECT (u0, p0, wr0, wq0, wf0, c1, k60) client_id=b'test'
[2020-07-11 11:51:49,542] DEBUG    dsmr_client.management.commands.dsmr_client: Starting infinite command loop...
[2020-07-11 11:51:49,545] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m1), 'b'dsmr/reading/id'', ... (8 bytes)
[2020-07-11 11:51:49,545] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 1
[2020-07-11 11:51:49,549] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m2), 'b'dsmr/reading/phase_power_current_l3'', ... (1 bytes)
[2020-07-11 11:51:49,549] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 2
[2020-07-11 11:51:49,550] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m3), 'b'dsmr/reading/phase_power_current_l2'', ... (1 bytes)
[2020-07-11 11:51:49,550] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 3
[2020-07-11 11:51:49,550] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m4), 'b'dsmr/reading/phase_power_current_l1'', ... (1 bytes)
[2020-07-11 11:51:49,550] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 4
[2020-07-11 11:51:49,551] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m5), 'b'dsmr/reading/phase_voltage_l3'', ... (5 bytes)
[2020-07-11 11:51:49,551] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 5
[2020-07-11 11:51:49,552] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m6), 'b'dsmr/reading/phase_voltage_l2'', ... (5 bytes)
[2020-07-11 11:51:49,552] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 6
[2020-07-11 11:51:49,553] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m7), 'b'dsmr/reading/phase_voltage_l1'', ... (5 bytes)
[2020-07-11 11:51:49,553] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 7
[2020-07-11 11:51:49,554] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m8), 'b'dsmr/reading/phase_currently_returned_l3'', ... (5 bytes)
[2020-07-11 11:51:49,555] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 8
[2020-07-11 11:51:49,555] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m9), 'b'dsmr/reading/phase_currently_returned_l2'', ... (5 bytes)
[2020-07-11 11:51:49,556] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 9
[2020-07-11 11:51:49,556] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m10), 'b'dsmr/reading/phase_currently_returned_l1'', ... (5 bytes)
[2020-07-11 11:51:49,556] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 10
[2020-07-11 11:51:49,557] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m11), 'b'dsmr/reading/extra_device_delivered'', ... (8 bytes)
[2020-07-11 11:51:49,557] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 11
[2020-07-11 11:51:49,558] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m12), 'b'dsmr/reading/extra_device_timestamp'', ... (25 bytes)
[2020-07-11 11:51:49,558] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 12
[2020-07-11 11:51:49,563] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m13), 'b'dsmr/reading/phase_currently_delivered_l3'', ... (5 bytes)
[2020-07-11 11:51:49,564] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 13
[2020-07-11 11:51:49,565] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m14), 'b'dsmr/reading/phase_currently_delivered_l2'', ... (5 bytes)
[2020-07-11 11:51:49,565] DEBUG    MQTT: Paho client: on_publish(userdata, mid) None 14
[... cut ...]
[2020-07-11 11:51:49,580] DEBUG    MQTT: Paho client: on_log(userdata, level, buf) None 16 Received CONNACK (0, 0)
[2020-07-11 11:51:49,580] DEBUG    MQTT: Paho client: on_connect(userdata, flags, rc) None {'session present': 0} 0
[2020-07-11 11:51:49,580] DEBUG    MQTT: --- localhost : 1883 -> Connection successful
[2020-07-11 11:51:49,580] DEBUG    dsmr_client.management.commands.dsmr_client: Sleeping 1.0s
```

# Config
Also, make sure you've enabled the MQTT data sources in the admin interface of DSMR-reader.

bduijnhouwer, 2020-07-11:
It works now, but I don't quite get it.
In "configuration / MQTT" I didn't enable anything.
After I did, it all worked.

But I'm 95% sure 😇 that when I had the datalogger and the reader on
one Raspberry Pi, I didn't enable anything and it still worked.

Is that possible or ... am I getting old? 😂

As far as I'm concerned, this issue can be closed.

dennissiemensma, 2020-07-11:
Great! Let's say it's "bug" then...

closed 2020-07-11 by dennissiemensma