dsmrreader/dsmr-reader #1045: Teruggeleverde stroom wordt aangeduid als laagtarief

rraymakers, 2020-07-13:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/v3/troubleshooting.html

Wat gebruik je?
- DSMR-reader versie:   3.11
- Type RaspberryPi or server:   RaspberryPi
- Standaardinstallatie of Docker:   standaardinstallatie
- DSMR-protocol slimme meter v4/v5:  Protocol V5 (T210-D ESMR5)

---

Standaard installatie gedaan, 1 energiecontract aangemaakt. Geen speciale setting ingesteld.

Probleem:
Teruggeleverde stroom wordt aangeduid als laagtarief terwijl de teruglevering gebeurde tijdens de hoogtariefperiode.

[image]
[image]

dennissiemensma, 2020-07-13:
Bedankt voor je melding. Ik denk dat je morgen (of nu nog voordat 'dal' ingaat) het beste even kunt spieken op: `/admin/dsmr_datalogger/meterstatistics/1/change/`.

Onderaan de pagina staat het laatste telegram. Je kunt aan deze regel zien wat het huidige tarief is:

```
# Laag
0-0:96.14.0(0001)

# Hoog
0-0:96.14.0(0002)
```

Daarnaast kun je ook kijken bij teruglevering (morgen dan) welke meterstand oploopt:

```
# Laag
1-0:2.8.1(000000.000*kWh)

# Hoog
1-0:2.8.2(000000.000*kWh)
```

dennissiemensma, 2020-07-13:
Nog een ander belangrijk detail: In het **weekend** wordt er doorgaans altijd **laagtarief** gehanteerd. De hele dag door!

Ik weet niet of dat in heel NL is, maar dat kan ook nog een verklaring zijn. ;-)

rraymakers, 2020-07-14:
Het was inderdaad laagtarief in het weekend. Had wel de tijden opgezocht maar daar stond niks over weekend bij.

Issues is opgost.

closed 2020-07-14 by rraymakers