dsmrreader/dsmr-reader #1046: dsmr reader

Zwaal022, 2020-07-13:
Hello. 
  I wil graag de slimme meter thuis gaan gebruiken om te zien hoeveel watt vermogen ik teruglever aan het net als en wanneer ik teruglever. Ditverieert dus per seconde. Ik wil deze waarde gaan gebruiken om prcies deze hoeveelheid enrgie (bijv 352 Watt op dat specifieke moment) te gaan "divergeren" naar een elektrisch verwarmings element (mbv pulsbreedte modulatie).
   Nu lees ik in het project van de dsmr reader dat er slimme meters bestaan (de meesten denk ik ) die ieder 10 seconden een meting doorgeven maar ik wil dat graag vaker. Er zijn dus blijkbaar slimme meters die het per 1 seconde doen. Maar die moeten dan (van zichzelf zo begrijp ik) een bepaald specifiek programma in zich hebben. Ik ga nu proberen bij Liander om een meter van die soort te krijgen (ik heb nu nog geen slimme meter maar ze nodigen me uit om er 1 te laten plaatsen)
     Als dit alles lukt dan zal ik de output kunnen lezen; maar daarmee heb ik echter nog geen (geschikt ?) signaal om mijn pulsbreedte mee aan te sturen. 
     Nu ben ik helemaal niet goed thuis (zeg maar totaal niet) in programmeren........Dus ik ben op zoek naar hulp hierbij.

     Is dit misschien de weg om die hulp te vinden ?    Of kan iemand me een suggestie doen waar ik die dan kan vinden ?

   MvG , Chris

dennissiemensma, 2020-07-13:
Bedankt voor je vraag. Ik kan een deel voor je beantwoorden:

----

Tegenwoordig zijn eigenlijk alle nieuwe meters DSMR-v5 meters. Deze communiceren elke seconde de elektrastanden naar de P1-poort (die je kunt uitlezen). Daarnaast werkt een eventueel gekoppelde gasmeter doorgaans elke 5 minuten de standen bij op de P1-poort.
Mijn meter uit ~2015 is not een DSMR-v4 meter en was destijds de standaard. Deze communiceren de elektrastanden elke 10 seconden en de gasmeterstand wordt elk uur bijgewerkt.

Je kunt dus het beste informeren of je een DSMR-v5 meter krijgt, mocht je die dan al niet standaard krijgen.

----

Voor teruglevering is het belangrijk om te weten dat je omvormer de exacte **bruto** opbrengt meet/kan meten, maar een slimme meter kan alleen de **netto** opbrengst meten. De netto opbrengst is dus het 'overschot' dat niet opgenomen wordt in je huishouden, dus wellicht is dat ook exact wat je zoekt.

----

Voor de rest verwijs ik je graag door naar Tweakers, want ik heb zelf geen PV, dus ik kan je beperkt helpen. Een mogelijk startpunt is om je verhaal hier te plaatsen: https://gathering.tweakers.net/forum/list_messages/1578510/0

closed 2020-07-13 by dennissiemensma

Zwaal022, 2020-07-14:
Hallo Dennis,
     Hartstikke fijn dat je me beantwoord hebt !

   Aangezien jij weet te vertellen dat vrijwel alle nieuwe meters DSMR v5
meters zijn , ga ik er gewoon maar 1 laten plaatsen.
      Ik wil inderdaad exact weten hoeveel mijn "netto" opbrengst is.Ik wil
nml proberen om ervoor te zorgen dat er vrijwel geen vermogen door mijn
meter naar de straat / het net gaat.  Oftewel (als er teruggeleverd wordt ,
dat dit "weggevangen"wordt (of anders geformuleerd: gedivergeerd ) en naar
mijn verwarmingselementen gaat.
    Op dat moment is het niet van belang waar dit vermogen vandaan komt
(in mijn geval pv, maar het zou ook een generator kunnen zijn)
   Ik zoek nu echter hulp om de elektra meter (van Liander in mijn geval)
uit te lezen, en niet alleen dat , maar de waarde (in Watt) die daar
gepresenteert wordt om te zetten van een digitaal signaal naar een analoog
stuursignaal waarmee ik mijn vermogen naar de verwarmingselementen kan gaan
regelen .
    Ik zal mijn vraag eens op tweakers stellen , hopelijk zitten daar
mensen die me kunnen helpen !

       Nogmaal reuze bedankt voor respons !!

       Vriendelijke Groet , Chris