dsmrreader/dsmr-reader #1047: Reading in DSMR data from external source

nmrsmn, 2020-07-14:
Is it possible to use the DSMR-reader with telegrams from an external source like another DSMR reader/loggers (like the DSMRLoggerAPI see: https://mrwheel-docs.gitbook.io/dsmrloggerapi/)?

I've tried to find the answer to my question in the documentation but I couldn't find anything except for the "statement" that it should be possible to read data from another DSMR-reader instance, but no guide or whatsoever on how to get this done.

dennissiemensma, 2020-07-14:
Thanks for asking. DSMR-reader has an API: https://dsmr-reader.readthedocs.io/en/latest/api.html
Unfortunately the API docs are hosted locally in every DSMR-reader installation. 

You can insert data into DSMR-reader by either:
- using the v1 (remote datalogger) API, which only has a single endpoint that accepts the **entire raw telegram**. This will parse and process the data as if DSMR-reader was directly connected to your smart meter. This also prevents you from having to parse the telegram manually at all.
- using the v2 (rest) API, which has an endpoint that accepts the data of the telegram **separately on a per-field basis**. This means you **will** have to parse the telegram yourself and assign the right data to the right field in the API. 

----

- You could load the [Postman-collection/env](https://github.com/dennissiemensma/dsmr-reader/tree/v3/dsmr_frontend/static/dsmr_frontend/postman) for a glance.
- Or check the screenshot of the docs below:

[image]

dennissiemensma, 2020-07-14:
Don't hesitate in asking for any futher assistance when needed.

closed 2020-07-14 by dennissiemensma