dsmrreader/dsmr-reader #1048: Feature request: vaste leveringskosten per dag

bikeymouse, 2020-07-14:
Hi, ik heb nog een kleine verbetering voor het tonen van de energiekosten:

Zoals denk ik alle energiecontracten bestaan de kosten bij mijn leverancier uit kosten per Kw of M3 en vaste leveringskosten per dag. Ik kan echter alleen de variabele kosten invoeren. Om toch de vaste kosten mee te laten rekenen kijk ik nu hoeveel ik per jaar gebruik om dan de vaste kosten om te slaan in het bedrag per Kw of M3 van de variabele tarieven.

Ik zou het echter leuk vinden het mogelijk zou zijn om ook de vaste leveringskosten per dag voor electra en gas in te stellen. Dan zouden de maand en jaarbedragen nog beter (of zelfs volledig) moeten kloppen met de rekening die ik krijg :-)

dennissiemensma, 2020-07-14:
Bedankt voor je verzoek. Ik heb de afgelopen tijd aangegeven dat wat minder wenselijk en makkelijk is: #1042 #1020 #830 #573

Echter, gezien het blijft terugkeren als vraag, zal ik bij een toekomstige release opnieuw onderzoeken wat de mogelijkheden zijn.

bikeymouse, 2020-07-14:
Oops, ik had niet goed gezocht, had ook je eerdere antwoorden moeten vinden...In ieder geval dank voor je snelle reactie, en hopelijk dat je op een gegeven moment tijd voor vindt om het te realiseren.
Verder niets dan lof over DSMR, zit echt mooi in elkaar!

dennissiemensma, 2020-08-25:
@bikeymouse 
@Webserve (#1042) 
@ridderr (#1020) 
@quadrophenianl (#830)
@BlackDex (#573)
@poudenes (#573)

Deze staat op de planning voor de v4.3-release. Hieronder een work in progress. **Is dit werkbaar voor jullie?**

(*Ik ben dus niet van plan om voor elke korting en (vast) tariefje een apart veld te maken, om de complete facturatie van leveranciers na te bootsen. Dit is puur een hulpmiddel om het jaartotaal wat beter te benaderen.*)

## Instellen
[image]

## Dashboard
[image]

## Archief
### Dag
[image]
### Maand
[image]
### Jaar
[image]

## Contracten
[image]

ridderr, 2020-08-25:
Bedankt, dit ziet er perfect uit.

BlackDex, 2020-08-25:
Ziet er goed uit :).
Thx voor de toevoeging

bikeymouse, 2020-08-25:
Helemaal prima zo, goed opgelost. Dank!

dennissiemensma, 2020-08-26:
Dank voor jullie terugkoppeling. Dan neem ik het voor nu in deze vorm mee in de volgende release.

Webserve, 2020-08-26:
Ziet er goed uit top 👍

dennissiemensma, 2020-08-26:
Notificaties ook nog bijgewerkt. Het geheel is nu gemerged naar v4.3 release branch.

closed 2020-08-26 by dennissiemensma

dennissiemensma, 2020-08-28:
Uitgebracht in [v4.3](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.3.0).

- Voor terugwerkende kracht zie https://dsmr-reader.readthedocs.io/en/v4/faq.html#recalculate-prices-retroactively