dsmrreader/dsmr-reader #1052: Recente Voltage Fasen en Recente Stroomsterktes grafiek tonen enkel data voor L1

mrombouts, 2020-07-22:
Hi, 

Ik heb DSMR reader geïnstalleerd op een PI voor het uitlezen van een Belgische meter. Tijdens het aanmaken van de grafieken tonen de 'Recente Voltage Fasen' en 'Recente Stroomsterktes' enkel de metingen van L1. L2 en L3 ontbreken. 

Env: DSMrreader 3.11 op Pi4, leest data van Belgische Meter. 
Protocol is ingesteld op 'België - Fluvius (fix voor gasmeter)

Voorbeeld van een telegram tonen dat het correct wordt uitgelezen van de meter:
`(dsmrreader) dsmr@pi-2:~/dsmr-reader $ cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q
Connected.
/FLU5\253769484_A

0-0:96.1.4(50213)
0-0:96.1.1(3153414733313030313435393432)
0-0:1.0.0(200722121313S)
1-0:1.8.1(000013.085*kWh)
1-0:1.8.2(000008.176*kWh)
1-0:2.8.1(000015.995*kWh)
1-0:2.8.2(000000.184*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.000*kW)
1-0:2.7.0(01.228*kW)
1-0:32.7.0(241.9*V)
1-0:52.7.0(236.3*V)
1-0:72.7.0(237.1*V)
1-0:31.7.0(007*A)
1-0:51.7.0(002*A)
1-0:71.7.0(001*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(37464C4F32313230313039373837)
0-1:24.4.0(1)
0-1:24.2.3(200722121006S)(00000.316*m3)
!2A11
/FLU5\253769484_A
`

Ook in de database lijken de data points correct te worden opgeslagen:
` dsmrreader=# SELECT id, timestamp, phase_voltage_l1, phase_voltage_l2, phase_voltage_l3, phase_power_current_l1, phase_power_current_l2, phase_power_current_l3 FROM dsmr_datalogger_dsmrreading WHERE phase_power_current_l1 > 0 AND phase_power_current_l2 > 0 AND phase_power_current_l3 > 0;
  id   |       timestamp        | phase_voltage_l1 | phase_voltage_l2 | phase_voltage_l3 | phase_power_current_l1 | phase_power_current_l2 | phase_power_current_l3
-------+------------------------+------------------+------------------+------------------+------------------------+------------------------+------------------------
   839 | 2020-07-21 13:17:45+02 |            232.2 |            229.8 |            232.6 |                      1 |                      5 |                      1
   840 | 2020-07-21 13:17:46+02 |            232.2 |            229.0 |            233.1 |                      1 |                      9 |                      1
   841 | 2020-07-21 13:17:47+02 |            232.3 |            229.1 |            233.0 |                      1 |                      9 |                      1
   842 | 2020-07-21 13:17:48+02 |            232.3 |            229.3 |            232.9 |                      1 |                      9 |                      1
   843 | 2020-07-21 13:17:49+02 |            232.3 |            229.3 |            232.8 |                      1 |                      9 |                      1
   844 | 2020-07-21 13:17:50+02 |            232.4 |            229.4 |            232.8 |                      1 |                      9 |                      1
   845 | 2020-07-21 13:17:51+02 |            232.4 |            229.3 |            232.7 |                      1 |                      9 |                      1
   846 | 2020-07-21 13:17:52+02 |            232.4 |            229.0 |            232.9 |                      1 |                      9 |                      1
`

Enig idee waar het misloopt? 
Thanks!

dennissiemensma, 2020-07-22:
Bedankt voor je melding.

De weergave wordt bepaald door de 'features' van een slimme meter, zoals multi-fase, voltages en teruglevering. Het kan goed zijn dat [de huidige DSMR-reader-detectie op de features](https://github.com/dennissiemensma/dsmr-reader/blob/v3/dsmr_backend/services/backend.py#L42) van een meter minder goed werkt op niet-NL meters.

Het lijkt erop dat de huidige detectie kijkt of een van de velden `phase_currently_delivered_l2` / `phase_currently_delivered_l3` niet-`NULL` is. Deze geven daarmee dan aan dat de meter multi-fase is.
Deze velden lijken [wel in een NL-telegram te staan](https://github.com/dennissiemensma/dsmr-reader/blob/v3/dsmr_datalogger/tests/datalogger/test_example_dsmr50.py#L48-L53), maar niet in die van jou. Vandaar dat DSMR-reader denkt dat het een meter is met slechts 1 fase.

Ik zal de detectie wat uitbreiden in de v4.1-rease. Tot die tijd lijkt het er inderdaad op dat de gegevens in ieder geval al wel worden opgeslagen, dus dan volgt een visuele fix later.

mrombouts, 2020-07-23:
Hi Dennis, 
Bedankt voor de update.

dennissiemensma, 2020-07-29:
Opgelost in aanstaande v4.1 release.

closed 2020-07-29 by dennissiemensma

dennissiemensma, 2020-08-03:
Uitgebracht in [v4.1](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.1.0)