dsmrreader/dsmr-reader #1053: Archief data selectie - onthouden van maand / jaar

RutgerLubbers, 2020-07-23:
Hoi, 

als je binnen het archief bladert naar een specifiek dag, maand of jaar, en je selecteert dan een ander detail niveau (dus zeg van dag ga je naar maand) dan is standaard de huidige datum geselecteerd.

Als de geselecteerde datum bewaard wordt, dan kan je makkelijker switchen van detail niveau, dus een dag in een maand bekijken, teruggaan naar die maand en een andere dag selecteren.

Ik kan hier een PR voor maken als je wilt.

dennissiemensma, 2020-07-23:
Bedankt voor je verzoek en voorstel. Ik heb hier vorig jaar een keer een proof of concept voor gemaakt. 

Ik probeerde het destijds via de [history API](https://developer.mozilla.org/en-US/docs/Web/API/History_API) van browsers met `pushState()`, maar ik kreeg het niet helemaal goed. 
Het doel was namelijk om de datum persistent in de URL te bewaren en bij te werken (zonder page refresh oid). Daardoor kon ik namelijk realiseren wat je hier voorstelt en werd het namelijk ook mogelijk om te deeplinken tussen pagina's of vanuit het archief. En dat sluit aan op je andere verzoek in #1054.

Je mag zeker een poging wagen en een opzet maken, maar dan zul je het dus wel via die of een soortgelijke methode moeten doen. Ik zal er zelf ook nog een keertje opnieuw induiken. Wellicht kan het wat makkelijker met een combinatie van de hash-variabele in de URL en Local Storage, wat tegenwoordig breed ondersteunt wordt en een verademing is tov cookies.

RutgerLubbers, 2020-07-31:
Deep linking heb ik niet mogelijk gemaakt nog, onderwater wordt de datum geselecteerd en bijgehouden. Dit door aanroep van een methode.
Ik zou nog kunnen kijken of deep linking mogelijk is, door in die methode ook de URL aan te passen. En uit te lezen bij initialisatie.

dennissiemensma, 2020-08-01:
Prima! En anders mag je het eventueel aanbieden zoals je het al hebt hoor. Dan kijk ik later verder voor de afronding.

dennissiemensma, 2020-11-10:
@RutgerLubbers heb je hier (of voor #1054) nog een update? Als je er niet aan toegekomen bent is dat ook prima hoor, in dat geval zal ik ze dan (voor nu) sluiten.

closed 2020-12-17 by dennissiemensma