dsmrreader/dsmr-reader #1054: Archief navigatie via grafiek(en)

RutgerLubbers, 2020-07-23:
Hoi, 

binnen het archief zou het handig zijn als je kan klikken op een specifieke data punt (maand, of dag) en dat je dan springt naar dat detail niveau. 

Zo hoef je niet zelf eerst een ander niveau te selecteren en daarna de gewenste dag / maand ...

Ik kan hier een PR voor maken als je wilt.

dennissiemensma, 2020-07-23:
Bedankt voor je verzoek en voorstel. Zie vooral #1053. Als dat eenmaal werkt, dan vermoed ik dat het relatief weinig werk is om te deeplinken vanuit grafieken.

RutgerLubbers, 2020-07-31:
Voor een voorbeeld: http://dsmr.allowed.org/archive 

Ik zal ook kijken naar #1053.

dennissiemensma, 2020-11-10:
@RutgerLubbers heb je hier (of voor #1053) nog een update? Als je er tussendoor niet aan toegekomen bent is dat ook prima hoor, in dat geval zal ik ze dan (voor nu) sluiten.

closed 2020-12-17 by dennissiemensma