dsmrreader/dsmr-reader #1055: InfluxDB: Met erugwerkende kracht importeren

dennissiemensma, 2020-07-23:
Console command maken waarmee iemand met terugwerkende kracht metingen in InfluxDB kan zetten.

Wellicht met als parameter een InfluxDB-database, zodat iemand het eerst even los/ernaast kan proberen.

dennissiemensma, 2020-07-23:
Eerste opzet is klaar. Ik zal het in v4 mergen, maar niet documenteren.

dennissiemensma, 2020-07-23:
Ik zal het sowieso even een nachtje draaien voor de zekerheid. Werking is handmatig.

Database is verplicht om op te geven en advies is om hier initieel een **aparte database** voor te gebruiken als test, dus niet de database gebruiken waar het al automatisch heen gezet wordt. Kijk eerst maar of het uberhaupt goed verwerkt wordt.
```
./manage.py dsmr_influxdb_export_all_readings --to-influx-database export-test
./manage.py dsmr_influxdb_export_all_readings --to-influx-database export-test-2
```

Wegens de performance van het script + afhankelijkheid van hardware pakt het script standaard max 50 batches van 100 metingen. Dus 5k metingen per keer.

Dit is variabel te gebruiken:
```
./manage.py dsmr_influxdb_export_all_readings --to-influx-database export-test-2 --max-batches 100
./manage.py dsmr_influxdb_export_all_readings --to-influx-database export-test-2 --max-batches 200
```

Voor een export van DSMR-reader op mijn PC naar InfluxDB op dezelfde PC kon die makkelijk een limit van 1000+ batches aan, zonder performance issues. 
Echter mijn Pi 4 ging redelijk tot de 50 (vandaar de standaard) en daarna werd het snel slechter, zonder herstart van het script.

Tot slot wordt er een measurement `export_progress_meta` gebruikt om de voortgang na een batch op te slaan. Deze measurement kan na voltooiing van de export verwijderd worden.

dennissiemensma, 2020-07-24:
@slashw je kunt bovenstaande proberen. Daarvoor zul je moeten updaten naar `v4.0rc2`:

```
sudo su - dsmr
./deploy.sh
```

slashw, 2020-07-25:
Met de script komt (alle)data in de database, hiervoor een aparte database aangemaakt.
Heb max 1000 batches op mijn pi4 geprobeerd, het ging op het laatst niet van harte.
1000 batches zijn ong 2 dagen.
Heb jij enig idee hoe ik alles zou kunnen overzetten naar de database?
Voor de duidelijkheid, aan deze kant geen programmeer ervaring.

dennissiemensma, 2020-07-25:
@slashw dedankt voor het proberen. Het vergroten van de batches gaat sowieso niet werken op de Pi. Als je miljoenen metingen hebt die je wilt overzetten, zal dat sowieso even duren.

Je kunt het beste proberen om het script daarom, met de standaard batchgrootte, heel vaak achter elkaar uit te voeren op de achtergrond. Je kunt hier een bashscript voor gebruiken, of simpelweg iets als `watch`:
```
watch -n0 <command>
```

Waarbij die het command telkens na elkaar uitvoert. 

----

Uitdaging is dan nog om ervoor te zorgen dat je niet de terminal hoeft open te houden (of als de verbinding wegvalt). Hiervoor kun je de standaardtool `screen` gebruiken, die zorgt voor persistent SSH-sessies die je later kan hervatten. 

Wat je daarmee doet is inloggen op je Pi en dan een screen-sessie starten. Installeren kan met:

```
sudo apt-get install screen
```

Screen kan wat onduidelijk zijn in gebruik, maar in de basis is het heel simpel:
- Je start een screen-sessie
- Je voert iets uit
- Je gaat uit de sessie maar laat hem draaien
- Later kun je terug in de sessie.

Je kunt altijd zien welke sessies er zijn met:
```
screen -list
```

Een nieuwe sessie start je met:
```
screen -S <naam>

# bijvoorbeeld:
screen -S export
``` 

Je terminal wordt leeg en je kunt nu doen wat je wilt:
```
sudo su - dsmr
watch -n1 ./manage.py dsmr_influxdb_export_all_readings --to-influx-database <NAAM-DATABASE>
```

Vervolgens gaat die bezig en daar wil je niet op wachten dus ga je uit de sessie met:
```
CTRL + A, daarna CTRL + D
```

Je ziet dan iets als:
```
[detached from 2735.export]
```

Dat is goed.

Als je op een later moment weer in de sessie wilt (je kunt dus rustig uit SSH-terminal gaan), dan log je in op de Pi en doe je:
```
screen -list
```

Je sessie zou nog moeten draaien en je zou zoiets moeten zien:
```
There is a screen on:
	2735.export	(07/25/2020 06:14:32 PM)	(Detached)
1 Socket in /run/screen/S-pi.
```

Ga terug in de sessie met:
```
screen -r <naam>

# bijvoorbeeld:
screen -r export
```

Daarmee zit je weer in de sessie en zou het `watch` command nog steeds moeten draaien.

Vergeet niet om weer `CTRL + A, daarna CTRL + D` als je er weer uit wilt. 
Als je het `watch` command wilt stoppen kun je `CTRL + C` doen en daarna een paar keer `CTRL + D` om de sessie te beeindigen.

slashw, 2020-07-25:
Dankjewel voor de duidelijke uitleg, mijn database stroomt nu langzamerhand voller en de data wordt zichtbaar in de grafieken. 
Het enigste wat niet lukt is uit de sessie te gaan.
Maar dit is geen probleem.

dennissiemensma, 2020-07-25:
Je kunt eventueel gewoon de terminal sluiten. De sessie zou dan moeten blijven bestaan.

dennissiemensma, 2020-07-27:
Uitgebracht in v4-release. Bugs en aanvullingen kunnen het beste via een los issue denk ik. Ik ga er vanuit dat deze eerste opzet 'okay' werkt, maar nog niet perfect is.

dennissiemensma, 2020-07-27:
@slashw heeft het script inmiddels alle history voor je kunnen overzetten?

slashw, 2020-07-27:
Dat heeft het, 1e keer ging het overzetten niet goed, kreeg met nog 3 dagen te gaan een foutmelding. Weet niet meer wat het was, kan het ook niet reproduceren.
Gisteren weer geprobeerd, ik zag net dat de test database netjes gevuld was.
Top, het is wat mij betreft geslaagd.
Dankjewel.

dennissiemensma, 2020-07-27:
Fijn, bedankt voor de terugkoppeling. Dan houd ik het voor nu even ze.

closed 2020-07-27 by dennissiemensma

poudenes, 2021-01-09:
Wanneer weet je dat alle data in InfluxDB is overgezet?