dsmrreader/dsmr-reader #1056: Failed to read Buienradar API: Failed to connect to or read from Buienradar API

squadramunter, 2020-07-23:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/v3/troubleshooting.html

Wat gebruik je?
- DSMR-reader versie:   3.11
- Type RaspberryPi or server:   Raspberry Pi 3b+
- Standaardinstallatie of Docker:   Standaardinstallatie
- DSMR-protocol slimme meter v4/v5:   ???

---

Ik maak nu al een tijdje gebruik van DSMR Reader maar ben tot de ontdekking gekomen dat de Buienradar API niet goed functioneert. Hij komt steeds met een foutmelding in de logs. Klopt de API code nog wel?

[image]

dennissiemensma, 2020-07-23:
Bedankt voor je melding. Het is denk ik belangrijk dat je controleert of de grafiek van Buienradar het nog doet. Hier werkt die nog wel goed voor het meetstation die ik gebruik. Bij jou ook? Zie je recente uurmetingen?

De reden dat de foutmeldingen gelogd worden is dat het voor mij lastig is om te bepalen wat handig is en wat niet. De ene gebruiker wil het graag weten als er even een hikje is (waar Buienradar veel last van heeft) en de andere juist weer niet.

squadramunter, 2020-07-23:
Ja bij mij werkt alles gewoon naar behoren. Maar zou wel nice zijn als je een optie hebt waardoor je die meldingen uit kan zetten.

dennissiemensma, 2020-07-23:
Ik zal het in een toekomstige versie uitschakelen

squadramunter, 2020-07-23:
Misschien wel handig om dit in de terminal logging te kunnen bijhouden. Niet zozeer als meldingen. API waarschuwingen zijn opzich niet heel bijzonder maar kunnen wel handig zijn.

dennissiemensma, 2020-07-29:
Vanaf v4.1 worden deze meldingen niet meer getoond op het dashboard. Ze blijven dus nog wel gelogd in de logfile, dat was eerder ook al zo.

closed 2020-07-29 by dennissiemensma

dennissiemensma, 2020-08-03:
Uitgebracht in [v4.1](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.1.0)