dsmrreader/dsmr-reader #1058: check wether dsmr reader receives data Synology docker container

githubnoobjk, 2020-07-26:
I am running logger on a raspberry pi and reader in synoloy docker.
It seems I don't receive data using the API. how to check ?
Because I don't use a usb-cable I assume I don't have to use the usb-drivers 

Before switching to this project I used ser2net to receive data in home assistant. So when it is possible to use that again what should be set under devices in docker-compose.

This is what I have in my docker-compose.yaml

`version: '3.2'

services:
  dsmrdb:
    image: postgres:10.5-alpine
    container_name: dsmrdb
    restart: always
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./dsmrdb:/var/lib/postgresql/data
    environment:
      - TZ=Europe/Amsterdam
      - PG_TZ=Europe/Amsterdam
      - POSTGRES_USER=dsmrreader
      - POSTGRES_PASSWORD=dsmrreader
      - POSTGRES_DB=dsmrreader

  dsmr:
#    build: .
    image: xirixiz/dsmr-reader-docker
    #:test-latest
    container_name: dsmr
    depends_on:
      - dsmrdb
    cap_add:
      - NET_ADMIN    
    links:
      - dsmrdb
    restart: always
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./dsmr_backups:/dsmr/backups
      - ./modules/forward_telegram_to_api.py:/dsmr/dsmr_plugins/modules/forward_telegram_to_api.py
      - ./datalogger:/var/log/supervisor/dsmr_datalogger.log
      - ./dsmr_backend_log:/var/log/supervisor/dsmr_backend.log
      
    environment:
      - TZ=Europe/Amsterdam
      - VIRTUAL_HOST=localhost
      - SD_AUTOSTART_MQTT=false
      - SD_AUTORESTART_MQTT=false
      - DSMRREADER_PLUGINS=dsmr_plugins.modules.forward_telegram_to_api
      - DSMRREADER_LOGLEVEL=DEBUG
    ports:
      - 7777:80
      - 7779:443
    #devices:
   #   - /dev/ttyUSB0:/dev/ttyUSB0

#volumes:
#   dsmrdb:
#   dsmrdb_backups:`

dennissiemensma, 2020-07-26:
To use a remote datalogger, you can [install the remote datalogger script](https://dsmr-reader.readthedocs.io/en/latest/installation/datalogger.html) on your Pi. It will read the serial port and forward the telegram to the DSMR-reader API on your Synology. You will have to enable the API in DSMR-reader and use the API key provided there. 
Afterwards, you can set the API-URL and API-key in the remote datalogger script.

Also, I do not think you need the `dsmr_plugins.modules.forward_telegram_to_api` plugin, unless you want to forward it again after being processed by DSMR-reader.

githubnoobjk, 2020-07-26:
Hello Dennis, 
Thank you for your quick reply. I did install the remote datalogger script. I assume host-ip has to be the ip of the receiving dsmr-reader ? I changed the API-key on both side's because i installed the logger first. Does the API-key has to have a certain length ?

dennissiemensma, 2020-07-26:
Yeah you can use any api key you wish in `API_SERVERS`. By default a random one is generated, but feel free to use one that's shorter or longer.

githubnoobjk, 2020-07-26:
Ok, thanks. But hat still does not answer my question how to check whether the telegrams are send and/or received.

dennissiemensma, 2020-07-26:
On the datalogger Pi you can either check the `/var/log/supervisor/dsmr_remote_datalogger.log` logfile or simply run the command manually to see whether there are any errors:

```
sudo su - dsmr
python3 -u /home/dsmr/dsmr_datalogger_api_client.py
```

When telegrams are received it should be displayed on the dashboard, Status page or the graphs.

githubnoobjk, 2020-07-27:
This is the error I get : ` python3 -u /home/dsmr/dsmr_datalogger_api_clie                                                                                                                                                             nt.py
INFO:root:[2020-07-27 14:52:39.891752] Starting...
INFO:root:[2020-07-27 14:52:39.895891] Opening serial port: /dev/ttyUSB0
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/serial/se                                                                                                                                                             rialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned                                                                                                                                                              no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dsmr/dsmr_datalogger_api_client.py", line 126, in <module>
    main()
  File "/home/dsmr/dsmr_datalogger_api_client.py", line 106, in main
    for telegram in read_serial_port(**SERIAL_SETTINGS):
  File "/home/dsmr/dsmr_datalogger_api_client.py", line 61, in read_serial_port
    data = serial_handle.readline()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/serial/se                                                                                                                                                             rialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: device reports readiness to read                                                                                                                                                              but returned no data (device disconnected or multiple access on port?)
`
As said I use the API, should I comment other lines ?

dennissiemensma, 2020-07-27:
I'm not sure if I understand properly what you mean with 'the API'. The datalogger script mentioned before does use the API, you should only need to run it.

The error `device reports readiness to read but returned ` could mean that you you are using multiple program to read the same serial port (not 100% sure though). Do you still use `ser2net` at the same time for HA as well? Because that might cause the conflict.

githubnoobjk, 2020-07-27:
No I don't use ser2net on this. 
With "the API' i meant this. When I understand it correctly , a user uses the API when it is not possible to use a USB-cable to dsmr-reader or another way to transfer the telegrams to dsmr-reader. So I thought, it might be necessary to turn off other ways of transferring in the datalogger -script. (just guessing).

I have another error in the script, but I asked Bram van Dartel about that because I believe he is the "docker-specialist"?
That error says :
INFO:root:[2020-07-27 17:22:37.416346] Telegram read
INFO:root:[2020-07-27 17:22:37.419464] Sending telegram to: http://10.0.0.31/api/v1/datalogger/dsmrreading
ERROR:root:[2020-07-27 17:22:37.468062] API error: HTTP 405 - <html>
<head><title>405 Not Allowed</title></head>
<body>
<center><h1>405 Not Allowed</h1></center>
<hr><center>nginx</center>
</body>
</html>
So I asked him if the url is correct (while everyting runs in a container)

Appreciate your quick responses by the way :)

dennissiemensma, 2020-07-27:
Did you enable the API in DSMR-reader admin? It's disabled by default and will [return HTTP 405 Not Allowed](https://github.com/dennissiemensma/dsmr-reader/blob/v3/dsmr_api/views/v1.py#L21) when still disabled.

githubnoobjk, 2020-07-27:
Yes I did enable the API in DSMR-reader

dennissiemensma, 2020-07-27:
What does this output?

```
curl -H :Authorization: Token ENTERAPIKEY" -d "telegram=dummy" -X POST http://ENTERHOST/api/v1/datalogger/dsmrreading
```

Replace `ENTERAPIKEY` and `ENTERHOST` with you API key and host.

githubnoobjk, 2020-07-28:
Gives the same output, shouldn't there be something like portnumber or an extended url. Or are api allways saved in the same folders ?
 curl -H "Authorization: Token 356ABXX715" -d "telegram=dummy" -X POST http://10.0.0.31/api/v1/datalogger/dsmrreading
<html>
<head><title>405 Not Allowed</title></head>
<body>
<center><h1>405 Not Allowed</h1></center>
<hr><center>nginx</center>
</body>
</html>

dennissiemensma, 2020-07-28:
My bad. You will need to use the same (base) URL/hostname as you're using for the DSMR-reader webinterface. This includes any non-standard port.

You could try appending `:7777` to the hostname:
```
http://10.0.0.31:7777/api/v1/datalogger/dsmrreading
```

githubnoobjk, 2020-07-28:
Hurray.. added portnumber and the telegrams flush in. Thank you very much.
I suggest you modify the installation instructions for other noobs like me :)

closed 2020-07-28 by githubnoobjk

dennissiemensma, 2020-07-28:
Will do, thanks for the heads up.