dsmrreader/dsmr-reader #1060: Upgrade to v4.0 step 6. Drop dsmr_mqtt

bertus24, 2020-07-28:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

Wat gebruik je?
- DSMR-reader versie:   4.0
- Type RaspberryPi or server:   RapberryPI - VERSION="10 (buster)"
- Standaardinstallatie of Docker:   Standaardinstallatie
- DSMR-protocol slimme meter v4/v5:   Waar vind ik dit?

---
Tijdens de uitvoer van het update process bij stap 6. Drop dsmr_mqtt

sudo supervisorctl status geeft 

dsmr_backend                     STOPPED   Jul 28 09:51 AM
dsmr_datalogger                  STOPPED   Jul 28 09:51 AM
dsmr_mqtt                        STOPPED   Not started
dsmr_webinterface                STOPPED   Jul 28 09:51 AM

In instructie staat "Is dsmr_mqtt listed? If not listed, skip this chapter. Otherwise remove it:" in dit geval is mqtt wel listed maar een "sudo rm /etc/supervisor/conf.d/dsmr_mqtt.conf" geeft als resultaat :
rm: cannot remove '/etc/supervisor/conf.d/dsmr_mqtt.conf': No such file or directory"

De dsmr_mqtt.conf is dan ook niet aanwezig
sudo ls /etc/supervisor/conf.d/dsmr_*
/etc/supervisor/conf.d/dsmr_backend.conf  /etc/supervisor/conf.d/dsmr_datalogger.conf  /etc/supervisor/conf.d/dsmr_webinterface.conf

Bij het doorgaan met :
sudo supervisorctl reread
sudo supervisorctl update
sudo supervisorctl restart all

Krijg je het mqtt process nu als draaiend te zien.

sudo supervisorctl status
dsmr_backend                     RUNNING   pid 17605, uptime 0:00:05
dsmr_datalogger                  RUNNING   pid 16302, uptime 0:00:25
dsmr_mqtt                        RUNNING   pid 17714, uptime 0:00:04
dsmr_webinterface                RUNNING   pid 16301, uptime 0:00:25

Gaat het bij mij niet goed of klopt dit resultaat wel en zou wellicht de instructie wat aangepast moeten worden.

"You should not see dsmr_mqtt anymore.
Also, the other processes should be running as well again."

dennissiemensma, 2020-07-28:
Bedankt voor je melding. De situatie die je schetst is wat vreemd, dus ik wacht even af of meer hier last van hebben.

Wat je kunt doen is even kijken of `dsmr_mqtt` wellicht in een ander bestand verstopt zit.

Geeft dit output?

```
sudo grep dsmr_mqtt /etc/supervisor/conf.d/*
```

----

Zo niet dan kun je het eens handmatig proberen:
```
sudo supervisorctl stop dsmr_mqtt
sudo supervisorctl remove dsmr_mqtt
sudo supervisorctl reread
sudo supervisorctl update
```

Komt `dsmr_mqtt` dan nog steeds tevoorschijn?

bertus24, 2020-07-28:
Hi Dennis,

Zie hieronder.

Geeft dit output?

sudo grep dsmr_mqtt /etc/supervisor/conf.d/*
/etc/supervisor/conf.d/dsmr-reader.conf:[program:dsmr_mqtt]
/etc/supervisor/conf.d/dsmr-reader.conf:command=/usr/bin/nice -n 15 /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/dsmr-reader/manage.py dsmr_mqtt

Dit geeft inderdaad resultaat kennelijk staat er iets in dsmr-reader.conf. Versie 4.0 draait verder wel goed.

Heeft deze configuratie nog een extra actie nodig of staat dit verder wel goed?

dennissiemensma, 2020-07-28:
Bedankt voor de update, dat verklaart het, want die miste in je eerdere lijst met bestanden.

Als de volgende bestanden aanwezig zijn:
```
pi@rpi4:~ $ sudo ls -l /etc/supervisor/conf.d/
total 12
-rw-r--r-- 1 root root 757 Feb 11 20:14 dsmr_backend.conf
-rw-r--r-- 1 root root 621 Feb 11 20:14 dsmr_datalogger.conf
-rw-r--r-- 1 root root 617 Feb 11 20:14 dsmr_webinterface.conf
```

Dan kun je deze verwijderen:
```
/etc/supervisor/conf.d/dsmr-reader.conf
```

En dit uitvoeren:
```
sudo supervisorctl reread
sudo supervisorctl update
```

Als het goed is verdwijnt dit dan.

----

Als het niet werkt of het helemaal stuk gaat, dan kun je het beste **al die bestanden daar verwijderen** en ze weer vers uit de repo kopieren:
```
sudo cp /home/dsmr/dsmr-reader/dsmrreader/provisioning/supervisor/dsmr_datalogger.conf /etc/supervisor/conf.d/
sudo cp /home/dsmr/dsmr-reader/dsmrreader/provisioning/supervisor/dsmr_backend.conf /etc/supervisor/conf.d/
sudo cp /home/dsmr/dsmr-reader/dsmrreader/provisioning/supervisor/dsmr_webinterface.conf /etc/supervisor/conf.d/

sudo supervisorctl reread
sudo supervisorctl update
sudo supervisorctl restart all
```

bertus24, 2020-07-28:
Hi Dennis,

Na het verwijderen van /etc/supervisor/conf.d/dsmr-reader.conf een reread en update ziet het er goed uit.

pi@energie:~ $ sudo supervisorctl status
dsmr_backend                     RUNNING   pid 608, uptime 0:01:02
dsmr_datalogger                  RUNNING   pid 607, uptime 0:01:02
dsmr_webinterface                RUNNING   pid 606, uptime 0:01:02

dennissiemensma, 2020-07-28:
Fijn, goed om te horen!

closed 2020-07-28 by dennissiemensma

glsf91, 2020-08-03:
Misschien handig om in de handleiding op te nemen. Ik had exact hetzelfde probleem.

dennissiemensma, 2020-08-03:
Bedankt voor je melding. Ik zal het bijwerken. 

Waarschijnlijk komt dit omdat er een periode is geweest dat alle processen in 1 bestand stonden en daar zijn jullie tegenaan gelopen.

dennissiemensma, 2020-08-03:
Docs bijgewerkt met aanvulling: https://dsmr-reader.readthedocs.io/en/latest/faq/v4_upgrade.html#drop-dsmr-mqtt