dsmrreader/dsmr-reader #1063: Notificaties via Telegram werken niet?

mthmulders, 2020-07-30:
Allereerst, dank voor deze mooie software! Ik gebruik het al enkele maanden met veel plezier.

Als ik het goed begrijp zou het mogelijk moeten zijn om elke nacht een bericht te krijgen op Telegram (de chat applicatie) met de verbruiksstatistieken. Om dat in te regelen heb ik een bot geregistreerd bij Telegram, en heb ik onder **/admin/dsmr_notification/notificationsetting/** de Telegram api key (tien cijfers, dubbele punt, random tekst) ingesteld en ook het Chat ID (een numerieke waarde die het gesprek met mijn telefoon aanduidt).

Kort na het opslaan krijg ik deze twee meldingen in het hoofdscherm:

[image]

De instellingen zijn daarna ook gewist.

Ik heb logging op DEBUG gezet (volgens [deze instructies](https://dsmr-reader.readthedocs.io/en/v3/troubleshooting.html)) en het nogmaals geprobeerd. In **/var/log/supervisor/dsmr_backend.log** kom ik dan dit tegen

    [2020-07-30 11:43:13,562] ERROR     - Notification API returned client error, wiping settings...
    [2020-07-30 11:43:13,569] ERROR    (AssertionError) <bound method NotificationAppConfig._on_backend_called_signal of <NotificationAppConfig: dsmr_notification>> errored: Notify API call failed: {"ok":false,"error_code":400,"description":"Bad Request: chat not found"} (HTTP 400)

Maar daar wordt ik nog niet heel veel wijzer van. Kies ik ergens een verkeerde waarde? Is de Telegram Bot API misschien gewijzigd waardoor DSMR Reader er niet goed meer mee overweg kan?

Ik wil gerust meer informatie aanleveren als dat nodig is, maar zelfs met debug logging kreeg ik er niet meer uit dan dit...

dennissiemensma, 2020-07-30:
Bedankt voor je melding. Ik gebruik het zelf niet en moest er ook weer induiken. Uiteindelijk lijkt de API hetzelfde gebleven, maar is de help-tekst van het Chat-ID veld onduidelijk in DSMR-reader.

Ik heb het nu weer werkend gekregen met zowel een DM als een groep. **Hiervoor zul je eerst de bot direct of in een groep een bericht moeten sturen.** 

Het bericht komt dan voorbij als 'update' voor de bot en daar kun je het Chat ID uithalen. Je kunt de laatste update naar de bot gewoon simpel uitlezen in je browser:
```
https://api.telegram.org/bot<APIkey>/getUpdates?offset=-1
```

Bijvoorbeeld met deze fictieve API key `1268827887:ABCDEFGqdtsgtts-JDASIJDqewqkqewaq`:
```
https://api.telegram.org/bot1268827887:ABCDEFGqdtsgtts-JDASIJDqewqkqewaq/getUpdates?offset=-1
```

Let op dat er geen extra teken of slash achter `https://api.telegram.org/bot` staat, maar gewoon direct je API key. Je krijgt als het goed is zoiets te zien:

[image]

Waarbij de Chat ID die je zoekt hier `123456789` is.

Je kunt dit overigens wat makkelijker testen met:

```
sudo su - dsmr
./manage.py dsmr_notification_test
```

Na een gefaalde notificatie resetten de instellingen wel, maar als je gewoon DSMR-reader admin openhoudt en na invoeren klikt op `Save and continue editing` dan kun je dat telkens blijven doen. Beetje workaround voor het mechanisme.

dennissiemensma, 2020-07-30:
Ik zal de docs bijwerken met deze info.

mthmulders, 2020-07-30:
Wauw, dank voor het speurwerk! Ik heb mijn fout ook ontdekt: ik had `@<chat-id>` gebruikt, en dat moet dus `<chat-id>` zijn. Nu werkt het als een zonnetje. Ook fijn om te weten dat er een commando is om mee te testen.

dennissiemensma, 2020-07-30:
Bedankt voor de info! Ik heb de docs aangepast met bovenstaande verhaal en de beschrijving bij het veld aangepast zodat helder is dat die `@` er niet bij hoort.

dennissiemensma, 2020-07-30:
Gaat mee met v4.1

closed 2020-07-30 by dennissiemensma

dennissiemensma, 2020-08-03:
Uitgebracht in [v4.1](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.1.0)