dsmrreader/dsmr-reader #1064: Grafieken inhoud vraagje

jorkzijlstra, 2020-07-31:
Wat gebruik je?
- DSMR-reader versie:   3.11
- Type RaspberryPi or server:   RockPi 4B
- Standaardinstallatie of Docker:   Docker
- DSMR-protocol slimme meter v4/v5:   V5

---

[image]

Zoals je kan zien werkt alles perfect. Ik had alleen een vraag over de inhoud van de grafiek op de live graphs pagina. 

De 2e grafiek toont de per fase waarbij je netjes de + en de - verbruik zit. Klopt het dat de eerste grafiek deze waardes van elkaar lijkt af te trekken en dus het verschil laat zien?

dennissiemensma, 2020-08-01:
Bedankt voor je vraag. De eerste grafiek toont wat je je slimme meter voor het hele huishouden leest als verbruik en/of teruglevering. 

Ik weet niet wat de slimme meter intern doet. Dus of die de waardes van de tweede grafiek gebruikt of dat het losse tellers zijn. In de docs staat het volgende.

## Eerste grafiek
[image]

## Tweede grafiek
[image]

dennissiemensma, 2020-08-01:
Dus voor de duidelijkheid. De waardes worden als het goed is getoond zoals doorgegeven door de slimme meter.

jorkzijlstra, 2020-08-02:
Zo een vermoeden had ik al. Bedankt voor de info.

closed 2020-08-02 by jorkzijlstra