dsmrreader/dsmr-reader #1065: Trouble starting dsmr_webinterface

pwesters, 2020-07-31:
Trying a fresh install with the docs but after step 10 the web interface gives an error

dsmr_webinterface FATAL can't find command '/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn'

dsmr_backend is running according to supervisor> status

jeroenpeters1986, 2020-07-31:
That's possible, becuase the backend does not use `gunicorn`. 
I am assuming you have used v4's setup, like https://dsmr-reader.readthedocs.io/en/v4/installation/explained.html#application-configuration-setup 

It's important that the step `pip3 install -r dsmrreader/provisioning/requirements/base.txt` didn't produce any errors. 
In this step, Gunicorn (which is needed for the webinterface) would be installed.

Could you try this step again, or do you still have the output of that?

pwesters, 2020-07-31:
I did use the v4's setup
Ran the pip3 install step again and it produced the following output

```
Requirement` already satisfied: coreapi==2.3.3 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 1)) (2.3.3)
Requirement already satisfied: crcmod==1.7 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 2)) (1.7)
Requirement already satisfied: django==3.0.8 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 3)) (3.0.8)
Requirement already satisfied: django-admin-rangefilter==0.6.1 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 4)) (0.6.1)
Requirement already satisfied: django-colorfield==0.3.2 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 5)) (0.3.2)
Requirement already satisfied: django-debug-toolbar==2.2 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 6)) (2.2)
Requirement already satisfied: django-filter==2.3.0 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 7)) (2.3.0)
Requirement already satisfied: djangorestframework==3.11 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 8)) (3.11.0)
Requirement already satisfied: django-solo==1.1.3 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 9)) (1.1.3)
Requirement already satisfied: dropbox==10.3.0 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 10)) (10.3.0)
Requirement already satisfied: gunicorn==20.0.4 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 11)) (20.0.4)
Requirement already satisfied: influxdb==5.3.0 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 12)) (5.3.0)
Requirement already satisfied: paho-mqtt==1.5.0 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 13)) (1.5.0)
Requirement already satisfied: pyserial==3.4 in /usr/lib/python3/dist-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 14)) (3.4)
Requirement already satisfied: pyserial-asyncio==0.4 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 15)) (0.4)
Requirement already satisfied: python-dateutil==2.8.1 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 16)) (2.8.1)
Requirement already satisfied: python-decouple==3.3 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 17)) (3.3)
Requirement already satisfied: pytz==2020.1 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 18)) (2020.1)
Requirement already satisfied: requests==2.24.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 19)) (2.24.0)
Requirement already satisfied: urllib3==1.25.9 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 20)) (1.25.9)
Requirement already satisfied: coreschema in /home/dsmr/.local/lib/python3.8/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (0.0.4)
Requirement already satisfied: uritemplate in /home/dsmr/.local/lib/python3.8/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (3.0.1)
Requirement already satisfied: itypes in /home/dsmr/.local/lib/python3.8/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (1.2.0)
Requirement already satisfied: asgiref~=3.2 in /home/dsmr/.local/lib/python3.8/site-packages (from django==3.0.8->-r dsmrreader/provisioning/requirements/base.txt (line 3)) (3.2.10)
Requirement already satisfied: sqlparse>=0.2.2 in /home/dsmr/.local/lib/python3.8/site-packages (from django==3.0.8->-r dsmrreader/provisioning/requirements/base.txt (line 3)) (0.3.1)
Requirement already satisfied: six>=1.12.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from dropbox==10.3.0->-r dsmrreader/provisioning/requirements/base.txt (line 10)) (1.14.0)
Requirement already satisfied: setuptools>=3.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from gunicorn==20.0.4->-r dsmrreader/provisioning/requirements/base.txt (line 11)) (44.0.0)
Requirement already satisfied: msgpack==0.6.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from influxdb==5.3.0->-r dsmrreader/provisioning/requirements/base.txt (line 12)) (0.6.1)
Requirement already satisfied: chardet<4,>=3.0.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from requests==2.24.0->-r dsmrreader/provisioning/requirements/base.txt (line 19)) (3.0.4)
Requirement already satisfied: idna<3,>=2.5 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from requests==2.24.0->-r dsmrreader/provisioning/requirements/base.txt (line 19)) (2.8)
Requirement already satisfied: certifi>=2017.4.17 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from requests==2.24.0->-r dsmrreader/provisioning/requirements/base.txt (line 19)) (2019.11.28)
Requirement already satisfied: jinja2 in /usr/lib/python3/dist-packages (from coreschema->coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (2.10.1)
```

jeroenpeters1986, 2020-07-31:
It seems that gunicorn is installed succesfully, but it seems that it has not been installed in the correct VirtualEnv:
`Requirement already satisfied: gunicorn==20.0.4 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 11)) (20.0.4)`
Are you sure you installed it while having the virtualenv activated?
This is mention in the note just above step 7 on the page we linked earlier:

> **Note**
> You can easily test whether you’ve configured this correctly by logging out the dsmr user (CTRL + D) and login again using `sudo su - dsmr`.
> 
> You should see the terminal have a `(dsmrreader)` prefix now, for example: `(dsmrreader)dsmr@rasp:~/dsmr-reader $`

Ofcourse I cannot be sure, but it looks like you might have forgotten to activate the virtualenv?

pwesters, 2020-07-31:
I probably made a typo somewhere while creating/activating the virtualenv
I started again and it now works. Thanks for your help

closed 2020-07-31 by pwesters

jeroenpeters1986, 2020-07-31:
You are welcome! Glad it works

dennissiemensma, 2020-08-01:
Thanks JP!