dsmrreader/dsmr-reader #1068: feature request--opbrengst zonnepanelen meenemen.

githubnoobjk, 2020-08-06:
Het zou mooi zijn als de opbrengst van de zonnepanelen meegenomen zou kunnen worden. Je weet dan ook wat dan je eigen verbruik is.
Daarnaast kun je die waarden dan ook weer uploaden naar PVOutput.

Ik heb zelf een Growatt omvormer en in HomeAssistant kun je de uitvoerwaarden daarvan al opnemen. (loopt via de growatt server geloof ik).
Helaas ontbeer ik (nog) de programmeervaardigheden om zelf iets te ontwikkelem

jeroenpeters1986, 2020-08-06:
Hoi @githubnoobjk, deze vraag is al vaker voorbij gekomen. Omdat DSMR-Reader alleen bedoeld is voor het uitlezen van de slimmer meter (DSMR is het protocol) is de kans niet groot dat dat erin komt. Dennis [heeft dat eerder al mooi verwoord](https://github.com/dennissiemensma/dsmr-reader/issues/649#issuecomment-498408272) als: _Dit project focust zich alleen op het uitlezen van de slimme meter[...]_

Precies deze feature request, maar dan voor SolarEdge ipv Growatt, is ook al binnengekomen in [issue #621](https://github.com/dennissiemensma/dsmr-reader/issues/621#issuecomment-485078540), daarin staat overigens wel een ander pakket genoemd die dat voor SolarEdge kan, misschien is het er ook voor Growatt.

Ik zal het issue even openlatten voor Dennis zelf, maar weet in ieder geval dat dit eerder al gevraagd is

githubnoobjk, 2020-08-06:
Dank voor je reactie. Het leek me al onwaarschijnlijk dta het niet eerder was geoppperd, maar ik kon het niet vinden. Ben niet heel erg thuis in Github .

jeroenpeters1986, 2020-08-06:
Geen enkel probleem! Als je bij het issue overzicht ziet wordt er standaard
alleen gezocht in alle openstaande issues vandaar. Als je daar “ is:closed”
bij zet zal er wel gezocht worden in eerder gesloten issues.
-- 
Groeten,

Jeroen Peters

dennissiemensma, 2020-08-06:
Bedankt JP! Dat is inderdaad altijd de reden geweest (en nu nog steeds) :]

closed 2020-08-06 by dennissiemensma