dsmrreader/dsmr-reader #1071: Gaat automatisch opruimen goed?

vincedh, 2020-08-07:
Wat gebruik je?
- DSMR-reader versie:   4.0
- Type RaspberryPi or server:   ubuntu server
- Standaardinstallatie of Docker:   standaard
- DSMR-protocol slimme meter v4/v5:   v5

---

Hoi Dennis,
Ik kwam in een ander issue tegen hoe je de grootte van je database kunt zien en ik dacht, ik zal bij mezelf eens kijken hoeveel data er staat.
De retentieconfiguratie staat op 'verwijder de meeste metingen na een half jaar'
Ik heb DSMR sinds 1-jun-2019 en m'n DB is nu 3800MB (public.dsmr_datalogger_dsmrreading)
Dat vind ik vrij veel data voor ruim een jaar, waarvan de meeste metingen na een half jaar verwijderd worden, of valt dat wel mee?
Kan ik ergens zien dat het 'opruimen van oude data' goed gaat?

Bedankt voor je antwoord alvast en een fijn weekend.

dennissiemensma, 2020-08-07:
Bedankt voor je vraag. Dit hangt af van twee zaken:
- DSMR-versie van je meter
- Sleep tijd in datalogger
- Groeperen van telegrammen per minuut

Als je DSMR-versie 5 hebt, krijg je elke seconde een telegram. Indien je dat niet limiteert met de 'datalogger sleep', krijg je dus in theorie:
- Elke seconde een telegram
- Elke dag 86.400 telegrammen
- Elke week 604.800 telegrammen
- Elk halfjaar 15.724.800 telegrammen

Afhankelijk van wel of niet 'groeperen per minuut' heeft elke minuut 1 tot 60 extra records. Dus in het ergste geval heb je dubbel het aantal telegrammen.

Om te kijken hoe de aantallen verspreid zijn kun je de volgende queries uitvoeren:

```
sudo su - postgres
psql dsmrreader
```

## Telegrammen
```
select date_trunc('month', timestamp) as month, count(id) as total from dsmr_datalogger_dsmrreading group by month order by month desc;
```

## Elektra-records (wellicht gegroepeerd per minuut)
```
select date_trunc('month', read_at) as month, count(id) as total from dsmr_consumption_electricityconsumption group by month order by month desc;
```

## Gas-records (waarschijnlijk elke 5 minuten eentje)
```
select date_trunc('month', read_at) as month, count(id) as total from dsmr_consumption_gasconsumption group by month order by month desc;
```

dennissiemensma, 2020-08-07:
Vergeten te vermelden, maar de queries laten de totalen per maand zien. Je zou bij jou retentie-instelling dus een significant verschil in aantal moeten zien na 6 a 7 maanden.

dennissiemensma, 2020-08-07:
Ter vergelijking, ik draai het met retentie van een week:

```
         month          | total 
------------------------+-------
 2020-08-01 00:00:00+01 | 57339
 2020-07-01 00:00:00+01 |  2866
 2020-06-01 00:00:00+01 |  1440
 2020-05-01 00:00:00+01 |  1488
 2020-04-01 00:00:00+01 |  1440
 2020-03-01 00:00:00+00 |  1486
 2020-02-01 00:00:00+00 |  1392
```

vincedh, 2020-08-07:
Hier de output van de telegrammen:

 2020-08-01 00:00:00+00 |  590788
 2020-07-01 00:00:00+00 | 2668134
 2020-06-01 00:00:00+00 | 2583863
 2020-05-01 00:00:00+00 | 2662781
 2020-04-01 00:00:00+00 | 2585443
 2020-03-01 00:00:00+00 | 2639380
 2020-02-01 00:00:00+00 |  710104
 2020-01-01 00:00:00+00 |    1488
 2019-12-01 00:00:00+00 |    1488
 2019-11-01 00:00:00+00 |    1440
 2019-10-01 00:00:00+00 |    1488
 2019-09-01 00:00:00+00 |    1440
 2019-08-01 00:00:00+00 |    1488
 2019-07-01 00:00:00+00 |    1488
 2019-06-01 00:00:00+00 |    1440
 2019-05-01 00:00:00+00 |     300
(16 rows)

Lijkt dus helemaal goed te gaan, hartelijk dank voor je snelle reactie!

dennissiemensma, 2020-08-07:
Jij ook bedankt voor de snelle terugkoppeling en het dubbelchecken.

closed 2020-08-07 by dennissiemensma