dsmrreader/dsmr-reader #1073: psycopg2.IntegrityError: duplicate key value violates unique constraint "auth_permission_pkey" na upgrade van v3 -> v4

megapearl, 2020-08-09:
- DSMR-reader versie:  4.1.1
- Type RaspberryPi or server:   Debian 10
- Standaardinstallatie of Docker:   Standaard in virtualenv
- DSMR-protocol slimme meter v4/v5:  v5
---
Waarschijnlijk iets in de database niet lekker, ik heb dit soort meldingen weleens vaker gehad.
Hoe is de duplicate ook alweer te verwijderen uit de postgresql db? of kan ik deze fout negeren?

```
psycopg2.IntegrityError: duplicate key value violates unique constraint "auth_permission_pkey"
DETAIL:  Key (id)=(20) already exists.

```
```
root@mediaserver:/home/dsmr/dsmr-reader# sudo su - dsmr
(dsmrreader) dsmr@mediaserver:~/dsmr-reader$ ./deploy.sh

 --- You are currently running version:
4.1.1

 --- Checking Python version.
Running Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0]

 --- Pulling remote repository for new commits...

 --- Merging/updating checkout.
Already up to date.

 >>> Running post-deployment script. <<<

 --- Checking whether VirtualEnv is activated.

 --- Checking Python version.
Running Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0]

 --- Checking & synchronizing base requirements for changes.
Requirement already up-to-date: coreapi==2.3.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 1)) (2.3.3)
Requirement already up-to-date: crcmod==1.7 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 2)) (1.7)
Requirement already up-to-date: django==3.0.8 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 3)) (3.0.8)
Requirement already up-to-date: django-admin-rangefilter==0.6.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 4)) (0.6.1)
Requirement already up-to-date: django-colorfield==0.3.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 5)) (0.3.2)
Requirement already up-to-date: django-debug-toolbar==2.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 6)) (2.2)
Requirement already up-to-date: django-filter==2.3.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 7)) (2.3.0)
Requirement already up-to-date: djangorestframework==3.11 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 8)) (3.11.0)
Requirement already up-to-date: django-solo==1.1.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 9)) (1.1.3)
Requirement already up-to-date: dropbox==10.3.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 10)) (10.3.0)
Requirement already up-to-date: gunicorn==20.0.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 11)) (20.0.4)
Requirement already up-to-date: influxdb==5.3.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 12)) (5.3.0)
Requirement already up-to-date: paho-mqtt==1.5.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 13)) (1.5.0)
Requirement already up-to-date: pyserial==3.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 14)) (3.4)
Requirement already up-to-date: pyserial-asyncio==0.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 15)) (0.4)
Requirement already up-to-date: python-dateutil==2.8.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 16)) (2.8.1)
Requirement already up-to-date: python-decouple==3.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 17)) (3.3)
Requirement already up-to-date: pytz==2020.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 18)) (2020.1)
Requirement already up-to-date: requests==2.24.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 19)) (2.24.0)
Requirement already up-to-date: urllib3==1.25.10 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 20)) (1.25.10)
Requirement already satisfied, skipping upgrade: coreschema in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (0.0.4)
Requirement already satisfied, skipping upgrade: itypes in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (1.2.0)
Requirement already satisfied, skipping upgrade: uritemplate in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (3.0.1)
Requirement already satisfied, skipping upgrade: asgiref~=3.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from django==3.0.8->-r dsmrreader/provisioning/requirements/base.txt (line 3)) (3.2.10)
Requirement already satisfied, skipping upgrade: sqlparse>=0.2.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from django==3.0.8->-r dsmrreader/provisioning/requirements/base.txt (line 3)) (0.3.1)
Requirement already satisfied, skipping upgrade: six>=1.12.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from dropbox==10.3.0->-r dsmrreader/provisioning/requirements/base.txt (line 10)) (1.15.0)
Requirement already satisfied, skipping upgrade: setuptools>=3.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from gunicorn==20.0.4->-r dsmrreader/provisioning/requirements/base.txt (line 11)) (49.2.1)
Requirement already satisfied, skipping upgrade: msgpack==0.6.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from influxdb==5.3.0->-r dsmrreader/provisioning/requirements/base.txt (line 12)) (0.6.1)
Requirement already satisfied, skipping upgrade: chardet<4,>=3.0.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from requests==2.24.0->-r dsmrreader/provisioning/requirements/base.txt (line 19)) (3.0.4)
Requirement already satisfied, skipping upgrade: certifi>=2017.4.17 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from requests==2.24.0->-r dsmrreader/provisioning/requirements/base.txt (line 19)) (2020.6.20)
Requirement already satisfied, skipping upgrade: idna<3,>=2.5 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from requests==2.24.0->-r dsmrreader/provisioning/requirements/base.txt (line 19)) (2.10)
Requirement already satisfied, skipping upgrade: jinja2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from coreschema->coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (2.11.2)
[... 66 lines cut ...]
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 86, in _execute
    return self.cursor.execute(sql, params)
django.db.utils.IntegrityError: duplicate key value violates unique constraint "auth_permission_pkey"
DETAIL:  Key (id)=(20) already exists.

 --- Checking & synchronizing static file changes.

170 static files copied to '/var/www/dsmrreader/static', 387 unmodified.

 --- Reloading running apps...

 * Reloading process: dsmr_webinterface (Gunicorn)   [OK]
 * Reloading process: dsmr_backend                   [OK]
 * Reloading process: dsmr_datalogger                [!!] PID file does not exist or the datalogger is disabled (Should it run? As sudo user, try 'sudo supervisorctl start dsmr_datalogger')

 --- Clearing cache...

 --- Deployed version:
4.1.1

 >>> Deployment complete. <<<
(dsmrreader) dsmr@mediaserver:~/dsmr-reader$

```

dennissiemensma, 2020-08-09:
Probeer: https://dsmr-reader.readthedocs.io/en/v4/faq.html#how-do-i-fix-errors-such-as-detail-key-id-123-already-exists . En als dat het niet oplost, wat geeft de output van de commands uit de volgende comment? https://github.com/dennissiemensma/dsmr-reader/issues/201#issuecomment-269916675

megapearl, 2020-08-09:
Hoi Dennis,

```
# Note: dsmr_sqlsequencereset is only available in DSMR-reader v3.3.0 and higher
sudo su - dsmr
./manage.py dsmr_sqlsequencereset
```
Heeft het inderdaad opgelost, dank je wel!

closed 2020-08-09 by megapearl