dsmrreader/dsmr-reader #1074: Vraag, historische waarde van Voltages per fase

RualdOrdelman, 2020-08-10:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

Wat gebruik je?
- DSMR-reader versie:   3.12
- Type RaspberryPi or server:   Intel NUC
- Standaardinstallatie of Docker:   Docker compose
- DSMR-protocol slimme meter v4/v5:   Versie 4/5

---

DSMR reader draait hier nu al enkele jaren met veel plezier. Ik post de belangrijkste waardes via MQTT naar Home Assistant. Nu heb ik tijdens een probleem tijdens veel zonuren dat de voltages in de wijk te hoog worden, door de vele zonnepanelen in de wijk. In het live dashboard kan ik mooi de voltages zien per fase van de afgelopen 24 uur. Echter wil ik graag de waardes terugkijken van specifieke dagen in mei, juni etc. Echter kan ik in het tabblad historie de voltages niet terugkijken. 

Is er een manier om de historische voltage waardes alsnog zichtbaar te maken?

dennissiemensma, 2020-08-10:
Bedankt voor je vraag. Ik heb helaas geen grafische oplossing voor je.

Wat je wel kunt doen is zoeken op de commandline van de database. De cli client van Postgres is vrij goed en makkelijk in gebruik:

```
sudo su - postgres
psql dsmrreader

select read_at, phase_voltage_l1, phase_voltage_l2, phase_voltage_l3 from dsmr_consumption_electricityconsumption where read_at between '2020-05-01 05:00:00'::timestamp and '2020-05-01 21:00:00'::timestamp;
```

Afhankelijk van je groepering van de metingen zul je zelf moeten spelen met wat werkt voor je. Je kunt selecties maken per uur, minuut of een dag(deel) of meerdere dagen. Pas vanzelfsprekend dan `between '2020-05-01 05:00:00'::timestamp and '2020-05-01 21:00:00'::timestamp` aan.

Als je veel resultaten krijgt te zien kun je met up/down of page-up/-down (of je muis) scrollen en met `q` weer terug naar je query-invoer. Laat maar weten of je er uit komt.

Dit kun je uiteraard ook doen met een grafische database-client, zoals PgAdmin3, mits de poort open staat.

RualdOrdelman, 2020-08-11:
Dank je. Het is gelukt met eigen queries en die naar een CSV bestand.

dennissiemensma, 2020-08-11:
Oke top! PostgreSQL kan [zelf trouwens ook CSV's maken](https://www.postgresqltutorial.com/export-postgresql-table-to-csv-file/), mocht je het op een andere manier gedaan hebben dan die in de link.

closed 2020-08-11 by dennissiemensma