dsmrreader/dsmr-reader #1075: Influxdb start niet (goed)?

Helloitsme, 2020-08-10:
Hallo

Ik heb zojuist v4.1.1. geïnstalleerd (komende vanaf de laatste versie van 3..)  en probeer met Grafana de influxdb data te benaderen. Dat werkt echter niet (Bad Gateway(502) melding)

Als ik op de admin pagina van dsmr kijk dan zie ik geen vinkje staan bij "InfluxDB-integratie wijzigen". als ik zelf dat vinkje zet dan lijkt het alsof er 1x data wordt geschreven (onder Influxdb measurements ) maar daarna stopt het weer?
Als ik dan weer op InfluxDB-integratie wijzigen kijk, dan staat het vinkje weer uit.

dennissiemensma, 2020-08-10:
Bedankt voor je vraag. Zou je eens in de logfile van dsmr_backend willen kijken? Vermoedelijk is daar de reden te vinden: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#supervisor

De foutafhandeling van deze en andere integraties is vrij behouden, om te voorkomen dat DSMR-reader oneindig iets probeert te doen. Vandaar de wat aggresieve uitschakeling bij fouten. Ik zal nog eens kijken of ik daar iets beters op kan verzinnen.

dennissiemensma, 2020-08-10:
En als je helemaal uitgebreide logging wilt zien dan kun je eventueel tijdelijk debug logging inschakelen. Dat staat op dezelfde pagina toegelicht.

Helloitsme, 2020-08-10:
selectie uit de logfile (nadat debug aangezet is)

2020-08-10 18:45:14,246] ERROR    Failed to instantiate InfluxDB connection, disabling InfluxDB integration
[2020-08-10 18:45:14,247] ERROR    HTTPSConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?q=CREATE+DATABASE+%22dsmrreader_measurements%22&db=dsmrreader_measurements (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x73f62450>: Failed to establish a new connection: [Errno 111] Connection refused'))
NoneType: None
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
[2020-08-10 18:47:18,343] ERROR    Failed to instantiate InfluxDB connection, disabling InfluxDB integration
[2020-08-10 18:47:18,345] ERROR    HTTPSConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?q=CREATE+DATABASE+%22dsmrreader_measurements%22&db=dsmrreader_measurements (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x73f40430>: Failed to establish a new connection: [Errno 111] Connection refused'))
NoneType: None
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
[2020-08-10 19:57:58,315] ERROR    Failed to instantiate InfluxDB connection, disabling InfluxDB integration
[2020-08-10 19:57:58,316] ERROR    HTTPSConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?q=CREATE+DATABASE+%22dsmrreader_measurements%22&db=dsmrreader_measurements (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x73fde3d0>: Failed to establish a new connection: [Errno 111] Connection refused'))
NoneType: None
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
[2020-08-10 19:58:17,644] ERROR    Failed to instantiate InfluxDB connection, disabling InfluxDB integration
[2020-08-10 19:58:17,646] ERROR    HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?q=CREATE+DATABASE+%22dsmrreader_measurements%22&db=dsmrreader_measurements (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x73f896b0>: Failed to establish a new connection: [Errno 111] Connection refused'))
NoneType: None
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
[2020-08-10 20:03:10,751] ERROR    Failed to instantiate InfluxDB connection, disabling InfluxDB integration
[2020-08-10 20:03:10,752] ERROR    HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?q=CREATE+DATABASE+%22dsmrreader_measurements%22&db=dsmrreader_measurements (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7400c710>: Failed to establish a new connection: [Errno 111] Connection refused'))
NoneType: None
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
[2020-08-10 20:16:15,794] ERROR    Failed to instantiate InfluxDB connection, disabling InfluxDB integration
[2020-08-10 20:16:15,795] ERROR    HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?q=CREATE+DATABASE+%22dsmrreader_measurements%22&db=dsmrreader_measurements (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x740026d0>: Failed to establish a new connection: [Errno 111] Connection refused'))
NoneType: None
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
[2020-08-10 20:24:31,216] ERROR    Failed to instantiate InfluxDB connection, disabling InfluxDB integration
[2020-08-10 20:24:31,219] ERROR    HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?q=CREATE+DATABASE+%22dsmrreader_measurements%22&db=dsmrreader_measurements (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7403d710>: Failed to establish a new connection: [Errno 111] Connection refused'))
NoneType: None
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
[2020-08-10 20:31:28,918] ERROR    Failed to instantiate InfluxDB connection, disabling InfluxDB integration
[2020-08-10 20:31:28,920] ERROR    HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?q=CREATE+DATABASE+%22dsmrreader_measurements%22&db=dsmrreader_measurements (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7400f750>: Failed to establish a new connection: [Errno 111] Connection refused'))
NoneType: None
[2020-08-10 20:37:06,464] ERROR    dsmr_backend.management.commands.dsmr_backend: [!] Exception raised. Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 86, in _execute
[... cut ...]
[2020-08-10 22:08:06,935] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-08-10 22:08:07,952] DEBUG    SP: 0 backend service(s) ready to run
[2020-08-10 22:08:08,041] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-08-10 22:08:09,057] DEBUG    SP: 0 backend service(s) ready to run
[2020-08-10 22:08:09,149] DEBUG    dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
[2020-08-10 22:08:10,165] DEBUG    SP: 0 backend service(s) ready to run

dennissiemensma, 2020-08-10:
De host en/of poort klopt niet. Of influxdb draait niet op localhost. Kun je daar wat mee? Debug logging kan trouwens weer uit. Deze fout zou altijd zichtbaar moeten zijn.

dennissiemensma, 2020-08-10:
Of je hebt Https geselecteerd met verkeerde poort. Of https terwijl influxdb dat niet heeft. Dat kan ook trouwens nog.

Helloitsme, 2020-08-10:
vraag: moet ik influxdb zelf nog installeren? (heb ik namelijk niet gedaan) of zou dat tijdens de installatie van versie 4 automatisch gebeurd moeten zijn

Helloitsme, 2020-08-10:
daar lijkt het probleem te zitten. "Assumption is the mother of all fuckups..." sorry voor de overlast

dennissiemensma, 2020-08-11:
Oke dank voor de opheldering. Je zult inderdaad zelf InfluxDb moeten installeren. Doorgaans wil je die namelijk ook niet op je Pi hebben staan, tenzij je een niet-SD-kaart als storage gebruikt. Vandaar.

closed 2020-08-11 by dennissiemensma