dsmrreader/dsmr-reader #1076: "This site can’t be reached" na kale Docker installatie & een backlog issue

Erik050572, 2020-08-10:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

Wat gebruik je?
- DSMR-reader versie:   Stand alone V4.1.1 &
                                       Docker V4.1.0 (met plug-in naar Stand alone.)
- Type RaspberryPi or server:   Raspberry Pi 3B V7
- Standaardinstallatie of Docker:   Beide
- DSMR-protocol slimme meter v4/v5:   V5

---

Hallo Dennis,

Naar aanleiding van de upload versie 4.1.0 voor Docker van @ualex73 heb ik een kale install gedaan, met onderstaande yml-bestand.

> version: '3'
>services:
>
>   dsmr-influxdb:
>     image: influxdb:latest
>     container_name: dsmr-influxdb
>     ports:
>      - '8086:8086'
>     volumes:
>      - influxdb-storage:/var/lib/influxdb
>     environment:
>      - INFLUXDB_DB=dsmrreader
>      - INFLUXDB_ADMIN_USER=dsmrreader
>      - INFLUXDB_ADMIN_PASSWORD=dsmrreader
>     volumes:
>     - /docker/db:/var/lib/postgresql/data
>     depends_on:
>      - dsmr-reader
>     restart: unless-stopped
>     networks:>
>      - net-dsmr
>
>   dsmr-grafana:
>     container_name: dsmr-grafana
>     image: grafana/grafana:latest
>     ports:
>      - '3000:3000'
>    volumes:
>      - grafana-storage:/var/lib/grafana
>      - ./grafana-provisioning/:/etc/grafana/provisioning
 >    depends_on:
>     - dsmr-influxdb
>     environment:
>      - GF_SECURITY_ADMIN_USER=dsmr
>      - GF_SECURITY_ADMIN_PASSWORD=dsmr
>     restart: unless-stopped
>     networks:
>      - net-dsmr
>      - grafana-storage:/var/lib/grafana
>      - ./grafana-provisioning/:/etc/grafana/provisioning
>     depends_on:
>      - dsmr-influxdb
>     environment:
>      - GF_SECURITY_ADMIN_USER=dsmr
>      - GF_SECURITY_ADMIN_PASSWORD=dsmr
>     restart: unless-stopped
>     networks:
>      - net-dsmr
>
>   dsmr-reader:
>    container_name: dsmr-reader
>    image: ualex73/dsmr-reader-docker:arm32v6
>    restart: unless-stopped
>    volumes:
>      - ./dsmr/backups:/home/dsmr/app/backups
>      - /dev:/dev
>    environment:
>      - DB_HOST=dsmr-influxdb
>      - DSMR_USER=admin
>      - DSMR_EMAIL=root@localhost
>      - DSMR_PASSWORD=admin
>      - VIRTUAL_HOST=localhost
>    ports:
>      - 8888:80
>      - 8889:443
>    networks:
>      - net-dsmr
>networks:
>  net-dsmr:
>    driver: bridge
>volumes:
>  influxdb-storage:
>  grafana-storage:
>  backups:

Vervolgens heb ik de onderstaande installatie commando's uitgevoerd:
sudo docker-compose -f docker-compose.yaml up -d dsmr-grafana
sudo docker-compose -f docker-compose.yaml up -d dsmr-influxdb
sudo docker-compose -f docker-compose.yaml up -d dsmr-reader

Tot zo ver geen problemen.  Docker PS geeft zelfs aan dat alles werkt
[image]

Als ik via een webbrowser poort 3000 op de Pi benader, komt keurig Grafana op.
Dit geldt helaas niet voor poort 443  of op poort 8888. Dan krijg ik de mededeling te zien dat de connectie geweigerd is.
Ook via Telnet krijg ik een foutmelding: telnet: Unable to connect to remote host: Connection refused

Zou je mij kunnen helpen met:
1. Deze fout vinden & oplossen;
2. Omdat dit een kale installatie betrof, zou ik InfluxDB willen laden met de data van de standalone (of van Dropbox)
    Zou je kunnen uitleggen hoe dat moet? &
3. (Misschien moet ik hier een andere topic voor aanmaken:)
    De stand-alone draait redelijk probleemloos. Tot het moment dat er weer een nieuwe versie beschikbaar is.:
    Waarom weet ik niet, maar dan worden de telegrammen niet meer verwerkt en de webpagina is ook niet te benaderen (ik weet even de foutcode niet). Als ik dan vervolgens de nieuwe versie heb geïnstalleerd, wordt de backlog weer verwerkt en is de webpagina weer beschikbaar. 
    Vanaf dat moment wordt ook de upload naar PVoutput hervat. Hierbij wordt dan niet de backlog geupload. Deze data ben ik dan gewoon kwijt. Zou je iets kunnen verzinnen dat deze backlog dmv. bijvoorbeeld een mass-update toch nog alsnog bij PVoutput terecht komt?

Alvast heel erg bedankt.
Met vriendelijke groet,
Erik

ualex73, 2020-08-11:
@Erik050572 In principe moet je poort 80, dus jouw port 8888 gebruiken. De 443 is gekoppeld aan 8889 in jouw voorbeeld?

Erik050572, 2020-08-11:
Goedenmorgen @ualex73 ,
Dat klopt: Poort 80 is gekoppeld aan poort 8888 en Poort 443 is gekoppeld aan 8889.

ualex73, 2020-08-11:
Ok, ik heb wel een idee waardoor het komt. Het init script controleert of de PostgreSQL DB online is, als dit niet het geval is, start die niet door.

Ik weet niet of ik het verkeerd gelezen heb, maar de InfluxDB draait naast de PostgreSQL en vervangt het niet? Dus je DSMR setup is mogelijk niet correct.

Erik050572, 2020-08-11:
Ik heb PostgreSQL toegevoegd en volgens docker PS draaien alle processen.
maar helaas is het resultaat, op dit moment hetzelfde,

Het yaml bestand ziet er nu zo uit

>version: '3.3'
>
>services:
>
>   dsmr-postgreSQL:
>     container_name: dsmr-postgreSQL
>     image: arm32v7/postgres:latest
>    restart: unless-stopped
>     ports:
>        - '5432:5432'
>     volumes:
>        - ./db/:/var/lib/postgresql/data
>     environment:
>        - POSTGRES_USER=dsmrreder
>        - POSTGRES_PASSWORD=dsmrreader
>        - POSTGRES_DB=dsmrreader
>     depends_on:
>        - dsmr-reader
>     networks:
>        - net-dsmr
>
>   dsmr-influxdb:
>     container_name: dsmr-influxdb
>     image: influxdb:latest
>     restart: unless-stopped
>     ports:
>        - '8086:8086'
>     volumes:
>        - influxdb-storage:/var/lib/influxdb
>     environment:
>        - INFLUXDB_DB=dsmrreader
>        - INFLUXDB_ADMIN_USER=dsmrreader
>        - INFLUXDB_ADMIN_PASSWORD=dsmrreader
>     depends_on:
>        - dsmr-postgreSQL
>     networks:
>        - net-dsmr
>
>   dsmr-grafana:
>     container_name: dsmr-grafana
>     image: grafana/grafana:latest
>     restart: unless-stopped
>     ports:
>        - '3000:3000'
>     volumes:
>        - grafana-storage:/var/lib/grafana
>        - ./grafana-provisioning/:/etc/grafana/provisioning
>     depends_on:
>        - dsmr-influxdb
>     environment:
>        - GF_SECURITY_ADMIN_USER=dsmr
>        - GF_SECURITY_ADMIN_PASSWORD=dsmr
>     networks:
>        - net-dsmr
>
>   dsmr-reader:
>    container_name: dsmr-reader
>    image: ualex73/dsmr-reader-docker:arm32v6
>    restart: unless-stopped
>    volumes:
>        - ./dsmr/backups:/home/dsmr/app/backups
>        - /dev:/dev
>    environment:
>        - DB_HOST=dsmr-influxd
>        - DSMR_USER=admin
>        - DSMR_EMAIL=root@localhost
>        - DSMR_PASSWORD=admin
>        - VIRTUAL_HOST=localhost
>    ports:
>        - '8888:80'
>        - '8889:443'
>    networks:
>        - net-dsmr
>
>networks:
>  net-dsmr:
>    driver: bridge
>
>volumes:
>  influxdb-storage:
>  grafana-storage:
>  grafana-provisioning:
>  backups:
>  dev:
>  db:
>

ualex73, 2020-08-12:
@Erik050572 als de web interface niet opkomt, moet je de docker logfile checken via "docker logs -f dsmr-reader".

Maar in je docker-compose file staat meerder fouten:
1)"DB_HOST=dsmr-influxd" moet de PostgreSQL zijn, dus moet "DB_HOST=dsmr-postgreSQL" worden. dit werkt alleen als je netwerk "net-dsmr" gebruikt, anders moet je je host IP gebruiken.
2) "POSTGRES_USER=dsmrreder" bevat een typo
3) De PostgreSQL heeft "depends_on: - dsmr-reader" ... Da's raar, volgens mij is de dsmr-reader afhankelijk van de SQL DB en niet andersom.

Erik050572, 2020-08-12:
Hoi @ualex73 ,

Ik heb je aanpassingen verwerkt en het docker logfile geeft meerdere foutmelding. Maar het komt hier op neer:
django.db.utils.OperationalError: FATAL:  password authentication failed for user "dsmrreader"

Om welke wachtwoord zou dit moeten gaan?

dennissiemensma, 2020-08-12:
Dat is een foutmelding van DSMR-reader die niet in de database van Postgres kan komen.

Erik050572, 2020-08-12:
Welk wachtwoord moet ik dan aanpassen? 
Die van Postgres neem ik aan. Maar die heb ik, volgens mij, letterlijk uit het voorbeeld script overgenomen en dus niet aangepast.

Of is er een andere manier om dit probleem op te lossen?

dennissiemensma, 2020-08-12:
In principe maakt het niet zoveel uit wat je gebruikt, zolang je maar dezelfde toegangsgegevens in zowel DSMR-reader als PostgreSQL instelt.

Bij een reguliere installatie [stel je als een van de eerste dingen](https://dsmr-reader.readthedocs.io/en/latest/installation/explained.html#database-backend-postgresql) de gebruikersnaam en wachtwoord in. Later gebruik je [dezelfde gegevens](https://dsmr-reader.readthedocs.io/en/latest/installation/explained.html#application-configuration-setup) in DSMR-reader, al staan ze daar al in de config template, er vanuitgaande dat men de gebruikersnaam en wachtwoord uit de handleiding gebruikt.

Als ik het zo zie lijk je in je script geen DB-gebruikersnaam en DB-wachtwoord voor DSMR-reader zelf in te stellen, alleen voor de admingebruiker van de webinterface. Mogelijk mis je dus nog variabelen.

Erik050572, 2020-08-12:
ik zou niet weten welke...
ik ben uitgegaan van dit [script](https://github.com/dennissiemensma/dsmr-reader/issues/797#issuecomment-566227647).
Deze werkte voor die versie en daaraan heb ik de secties Influxdb & Grafana toegevoegd.
De secties  dsmr-reader & db-dsmr (hernoemt naar dsmr-postgreSQL) zijn, volgens mij, praktisch gelijk gebleven.

dennissiemensma, 2020-08-13:
Je kunt hier zien wat je kunt instellen: https://dsmr-reader.readthedocs.io/en/latest/env_settings.html

Zowel daar als [in de repo van ualex](https://github.com/ualex73/dsmr-reader-docker/blob/master/Dockerfile.amd64) staan o.a. deze DSMR-reader variabelen die ik niet terugzie in je verhaal:

- `DB_USER`
- `DB_PASS`
- `DB_NAME`

Als je die niet via een andere manier instelt, kan het zo zijn dat je die mist.

Erik050572, 2020-08-13:
Deze newbee heeft een klein succesje:
Ik heb die nu toegevoegd en het docker log zegt nu het volgende:

...........................................................
2020-08-13 06:28:56
Start DSMR Reader - Mode=SERVER

Maar de webpagina komt nog niet op, want volgens mij miste ik er nog een paar:
SECRET_KEY: Hoe genereer ik deze in docker? Of kan ik dit als variabele toevoegen, het bestand, met een willekeurige waarde?
DB_ENGINE: Wat zou ik hier moeten invullen? 'dsmr-postgreSQL'  of 'django.db.backends.postgresql' of totaal wat anders?
DB_PORT: Uitgaande van docker PS is dit poort 5432, denk ik.

ualex73, 2020-08-13:
@Erik050572 

De succesvolle logging zou er zo uit moeten zien:

```
2020-08-11 02:51:23
Start DSMR Reader - Mode=SERVER
Connected to database successfully
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_influxdb, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  No migrations to apply.

557 static files copied to '/var/www/dsmrreader/static'.
Updating password of superuser "dsmrreader" (DSMR_USER)
2020-08-11 02:51:26,268 INFO Included extra file "/etc/supervisor.d/supervisord.ini" during parsing
2020-08-11 02:51:26,268 INFO Set uid to user 0 succeeded
2020-08-11 02:51:26,272 INFO RPC interface 'supervisor' initialized
2020-08-11 02:51:26,272 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2020-08-11 02:51:26,273 INFO supervisord started with pid 21
2020-08-11 02:51:27,278 INFO spawned: 'dsmr_backend' with pid 23
2020-08-11 02:51:27,281 INFO spawned: 'dsmr_datalogger' with pid 24
2020-08-11 02:51:27,283 INFO spawned: 'dsmr_webinterface' with pid 25
2020-08-11 02:51:27,284 INFO spawned: 'nginx' with pid 26
2020-08-11 02:51:28,959 INFO success: dsmr_backend entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-11 02:51:28,960 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-11 02:51:28,960 INFO success: nginx entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-11 02:51:30,963 INFO success: dsmr_webinterface entered RUNNING state, process has stayed up for > than 3 seconds (startsecs)

```

Indien er database connectie problemen zijn, zou je iets moeten zien als:
```
2020-08-11 02:51:23
Start DSMR Reader - Mode=SERVER
.............................................................. Could not connect to database server. Aborting...
```

ualex73, 2020-08-13:
Dus er lijkt nog een issue te zijn met je DB connectie/credentials.

BTW - had je ook "POSTGRES_USER=dsmrreder" gefixt (mist een "a")? Ik weet alleen niet of je de DB moet droppen om dit te fixen. Het is ook een mogelijk om dat te proberen, want je heb nog geen data erin staan.

Erik050572, 2020-08-13:
Het docker log laat het gedeelte zien van het gedeelte van de issues met database connectie, tot en met de puntjes, zien.
En daarna komt de lunix prompt weer op.

En de typefout had ik er ook uitgehaald.

(zal ik nog eens het YAML-bestand laten zien?)

Erik050572, 2020-08-13:
De onderstaande regels uit de sector dsmr-reader zijn verwijderd, omdat ze nu opmerkingen zijn: 
devices:/dev/ttyUSB0:/dev/ttyUSB0; 
DB_PORT:5432; 
SECRET_KEY:xyz; 
DB_ENGINE:dsmr-postgreSQL

version: '3.3'                                                                      
services:

   dsmr-postgreSQL:
     container_name: dsmr-postgreSQL
     image: arm32v7/postgres:latest
     restart: unless-stopped
     ports:
        - '5432:5432'
     volumes:
        - ./db/:/var/lib/postgresql/data
     environment:
        - POSTGRES_USER=dsmrreader
        - POSTGRES_PASSWORD=dsmrreader
        - POSTGRES_DB=dsmrreader
     networks:
        - net-dsmr

   dsmr-influxdb:
     container_name: dsmr-influxdb
     image: influxdb:latest
     restart: unless-stopped
     ports:
        - '8086:8086'
     volumes:
        - influxdb-storage:/var/lib/influxdb
     environment:
        - INFLUXDB_DB=dsmrreader
        - INFLUXDB_ADMIN_USER=dsmrreader
        - INFLUXDB_ADMIN_PASSWORD=dsmrreader
     depends_on:
        - dsmr-postgreSQL
     networks:
        - net-dsmr

   dsmr-grafana:
     container_name: dsmr-grafana
     image: grafana/grafana:latest
     restart: unless-stopped
     ports:
        - '3000:3000'
     volumes:
        - grafana-storage:/var/lib/grafana
        - ./grafana-provisioning/:/etc/grafana/provisioning
     depends_on:
        - dsmr-influxdb
     environment:
        - GF_SECURITY_ADMIN_USER=dsmrreader                   
        - GF_SECURITY_ADMIN_PASSWORD=dsmrreader
     networks:
        - net-dsmr
   dsmr-reader:
    container_name: dsmr-reader
    image: ualex73/dsmr-reader-docker:arm32v6
    restart: unless-stopped
    volumes:
        - ./dsmr/backups:/home/dsmr/app/backups
        - /dev:/dev
    environment:
        - DB_NAME:dsmrreader
        - DB_USER:dsmrreader
        - DB_PASS:dsmrreader
        - DB_HOST:dsmr-postgreSQL
        - DSMR_USER=admin
        - DSMR_PASSWORD:admin
        - DSMR_EMAIL=root@localhost
        - DSMR_PASSWORD=admin
        - VIRTUAL_HOST=localhost
        - TZ=Europe/amsterdam
    depends_on:
        - dsmr-postgreSQL
    ports:
        - '8888:80'
        - '8889:443'
    networks:
        - net-dsmr

networks:
  net-dsmr:
    driver: bridge

volumes:
  influxdb-storage:
  grafana-storage:
  grafana-provisioning:
  backups:
  dev:
  db

ualex73, 2020-08-13:
Zie je iets anders in de "docker logs dsmr-reader" output?

BTW - De "SECRET_KEY" is optioneel, als je die niet opgeeft, genereert de Docker container hiervoor een random value ;-)

Erik050572, 2020-08-13:
Dit is alles wat ik zie:

...........................................................
2020-08-13 11:10:53
Start DSMR Reader - Mode=SERVER
...........................................................
2020-08-13 11:11:58
Start DSMR Reader - Mode=SERVER
...........................................................
2020-08-13 11:13:03
Start DSMR Reader - Mode=SERVER
...........................................................
2020-08-13 11:14:10
Start DSMR Reader - Mode=SERVER
...........................................................
2020-08-13 11:15:16
Start DSMR Reader - Mode=SERVER
...........................................................
2020-08-13 11:16:20
Start DSMR Reader - Mode=SERVER

ualex73, 2020-08-13:
Er is nog steeds een probleem met je connectie naar je PostgreSQL server.

Probeer het volgende, om je in je container te komen (je zou na <60 seconden eruit worden gegooid, bij een herstart van de container):
```
docker exec -it dsmr-reader /bin/bash
```

En dan uitvoeren:
```
pg_isready -h dsmr-postgreSQL -p 5432
```

Dan hoort de output te zijn:
```
dsmr-postgreSQL:5432 - accepting connections
```

ualex73, 2020-08-13:
BTW, je environment lijkt fout:

```
environment:
- DB_NAME:dsmrreader
- DB_USER:dsmrreader
- DB_PASS:dsmrreader
- DB_HOST:dsmr-postgreSQL
```

Dit hoort volgens mij te zijn:
```
environment:
- DB_NAME=dsmrreader
- DB_USER=dsmrreader
- DB_PASS=dsmrreader
- DB_HOST=dsmr-postgreSQL
```

Erik050572, 2020-08-13:
Dit is de output als ik de ":" vervang door "=":

ERROR: The Compose file '././docker-compose/DSMR-reader.yaml' is invalid because:
services.dsmr-reader.environment contains non-unique items, please remove duplicates from ['DB_NAME=dsmrreader', 'DB_USER=dsmrreader', 'DB_PASS=dsmrreader', 'DB_HOST=dsmr-postgreSQL', 'DSMR_USER=admin', 'DSMR_PASSWORD=admin', 'DSMR_EMAIL=root@localhost', 'DSMR_PASSWORD=admin', 'VIRTUAL_HOST=localhost', 'TZ=Europe/amsterdam']

ualex73, 2020-08-13:
Je hebt een typo gemaakt :-)

```
'DB_NAME=dsmrreader',
'DB_USER=dsmrreader',
'DB_PASS=dsmrreader',
'DB_HOST=dsmr-postgreSQL',
'DSMR_USER=admin',
'DSMR_PASSWORD=admin',
'DSMR_EMAIL=root@localhost', 
'DSMR_PASSWORD=admin', <<<----- dubbel
'VIRTUAL_HOST=localhost',
'TZ=Europe/amsterdam'
```

Docker heeft jammer genoeg gelijk.

Erik050572, 2020-08-13:
Volgens mij ben ik nu in een cirkeltje gegaan.....

psycopg2.OperationalError: FATAL:  password authentication failed for user "dsmrreader"
django.db.utils.OperationalError: FATAL:  password authentication failed for user "dsmrreader"

ualex73, 2020-08-13:
kan je je PostgreSQL database nieuw aanmaken?

Erik050572, 2020-08-13:
net gedaan (al 3 keer :) )
nu krijg ik de foutmelding niet, maar hij start niet door.

Erik050572, 2020-08-13:
Ik denk dat ik nog een foutje van mezelf heb hersteld:
- DB_ENGINE=django.db.backends.postgresql

Erik050572, 2020-08-13:
2020-08-13 19:13:02
Start DSMR Reader - Mode=SERVER
Connected to database successfully
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/django/db/backends/base/base.py", line 220, in ensure_connection
    self.connect()
  File "/usr/local/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/django/db/backends/base/base.py", line 197, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/usr/local/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/django/db/backends/postgresql/base.py", line 185, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/usr/local/lib/python3.8/site-packages/psycopg2/__init__.py", line 127, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: FATAL:  password authentication failed for user "dsmrreader"

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/usr/local/lib/python3.8/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/usr/local/lib/python3.8/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/usr/local/lib/python3.8/site-packages/django/core/management/base.py", line 328, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/usr/local/lib/python3.8/site-packages/django/core/management/base.py", line 369, in execute
    output = self.handle(*args, **options)
  File "/usr/local/lib/python3.8/site-packages/django/core/management/base.py", line 83, in wrapped
    res = handle_func(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/django/core/management/commands/migrate.py", line 86, in handle
    executor = MigrationExecutor(connection, self.migration_progress_callback)
  File "/usr/local/lib/python3.8/site-packages/django/db/migrations/executor.py", line 18, in __init__
    self.loader = MigrationLoader(self.connection)
  File "/usr/local/lib/python3.8/site-packages/django/db/migrations/loader.py", line 49, in __init__
    self.build_graph()
  File "/usr/local/lib/python3.8/site-packages/django/db/migrations/loader.py", line 212, in build_graph
    self.applied_migrations = recorder.applied_migrations()
  File "/usr/local/lib/python3.8/site-packages/django/db/migrations/recorder.py", line 76, in applied_migrations
    if self.has_table():
  File "/usr/local/lib/python3.8/site-packages/django/db/migrations/recorder.py", line 56, in has_table
    return self.Migration._meta.db_table in self.connection.introspection.table_names(self.connection.cursor())
  File "/usr/local/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/django/db/backends/base/base.py", line 260, in cursor
    return self._cursor()
  File "/usr/local/lib/python3.8/site-packages/django/db/backends/base/base.py", line 236, in _cursor
    self.ensure_connection()
  File "/usr/local/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/django/db/backends/base/base.py", line 220, in ensure_connection
    self.connect()
  File "/usr/local/lib/python3.8/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/usr/local/lib/python3.8/site-packages/django/db/backends/base/base.py", line 220, in ensure_connection
    self.connect()
  File "/usr/local/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/django/db/backends/base/base.py", line 197, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/usr/local/lib/python3.8/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/django/db/backends/postgresql/base.py", line 185, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/usr/local/lib/python3.8/site-packages/psycopg2/__init__.py", line 127, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
django.db.utils.OperationalError: FATAL:  password authentication failed for user "dsmrreader"

dennissiemensma, 2020-08-13:
@ualex73 klopt het dat je je eigen repo inhoudelijk niet bijwerkt? V4 heeft wel wat wijzigingen namelijk. Zie ook [deze PR voor een beeld van de verschillen](https://github.com/xirixiz/dsmr-reader-docker/pull/118/files), al is die nog niet klaar.

En ook het datalogger-script is flink op de schop gegaan in de recente releases.

dennissiemensma, 2020-08-13:
@Erik050572 ik weet eerlijk gezegd niet of je er op deze manier uit gaat komen. Ter info:
- De standaardinstellingen staan hier: https://github.com/dennissiemensma/dsmr-reader/blob/v4/.env.template.
- Houd er rekening mee dat een eventuele oude config voor DSMR-reader v3.x **niet** zomaar te gebruiken is voor v4.x. Er is daar een [upgrade guide voor](https://dsmr-reader.readthedocs.io/en/latest/faq/v4_upgrade.html).
- Wellicht zou je je situatie beter op bijvoorbeeld Tweakers forum kunnen voorleggen. Want hier op Github lezen maar een beperkt aantal mensen mee en zelf gebruik ik docker niet actief genoeg om je inhoudelijk te helpen.

ualex73, 2020-08-13:
De github is niet bijgewerkt, dat klopt. Ik moet die nog pushen. De
datalogger is uit de image, zoals je al zegt, veel veranderd (hierdoor heb
ik het maar verwijderd, want ik gebruik het niet). De v4 wijzigen zijn
doorgevoerd, kostte wel even wat moeite ✌️

notifications@github.com>

ualex73, 2020-08-14:
@Erik050572 

Ik heb het bij mij getest met PostgreSQL 11.2 en dit werkt goed. Alleen met "latest" (12.x) PostgreSQL lijken dingen niet te werken.

Kan je je image aanpassen naar:
image: arm32v7/postgres:11.2

ualex73, 2020-08-14:
@Erik050572 

Ik keek nog naar je docker-compose en het viel me op dat er meer fouten inzitten.

Volgende is een directory, niet een docker volume:
```
volumes:
  - ./db/:/var/lib/postgresql/data
```

Volgende is wel een docker volume:
```
volumes:
  - db:/var/lib/postgresql/data
```

Erik050572, 2020-08-14:
Dankjewel.
we komen er wel wel 👍 

Ik zit nu in de loop dat de container "dsmr-reader" om de 3 zoveel seconden opnieuw opstart

dennissiemensma, 2020-08-17:
@Erik050572 ben je hier nog uiteindelijk uit gekomen?

Erik050572, 2020-08-19:
Nee, maar met de Development versie van @xirixiz komt ik wel verder.
Helaas stopt hij bij de volgende foutmelding:
Successfully installed psycopg2-2.8.5
[ INFO ] Verifying if Postgres in running...
[ FAIL ] Could not connect to database server. Aborting...

Erik050572, 2020-08-19:
Weer een foutje gevonden:
DB_HOST heb ik veranderd van LOCALHOST naar dsmr-postgreSQL

Beide images laden nu verder en komen volgens mij nu bij hetzelfde punt tot stilstand
De foutmeldingen op dit moment:

Bij de image van @ualex73:
2020-08-19 22:21:59,557 INFO spawned: 'dsmr_datalogger' with pid 98
2020-08-19 22:22:00,563 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-19 22:22:09,893 INFO exited: dsmr_datalogger (exit status 0; expected)

Bij de image van @xirixiz:
2020-08-19 22:29:44,680 INFO spawned: 'dsmr_webinterface' with pid 227
2020-08-19 22:29:45,695 INFO spawned: 'dsmr_backend' with pid 228
2020-08-19 22:29:45,705 INFO spawned: 'dsmr_datalogger' with pid 229
2020-08-19 22:29:46,547 INFO exited: dsmr_backend (exit status 1; not expected)
2020-08-19 22:29:46,551 INFO exited: dsmr_datalogger (exit status 1; not expected)
2020-08-19 22:29:47,211 INFO exited: dsmr_webinterface (exit status 3; not expected)

xirixiz, 2020-08-20:
Hoi Erik,

Kan je inloggen en kijken wat 

Logfiles can be found at: /var/log/supervisor/*.log and /tmp/supervisord.log

xirixiz, 2020-08-20:
> @Erik050572
> 
> Ik keek nog naar je docker-compose en het viel me op dat er meer fouten inzitten.
> 
> Volgende is een directory, niet een docker volume:
> 
> ```
> volumes:
>   - ./db/:/var/lib/postgresql/data
> ```
> 
> Volgende is wel een docker volume:
> 
> ```
> volumes:
>   - db:/var/lib/postgresql/data
> ```

Dat zou niet uit moeten maken, dat gaat meer om persoonlijke voorkeur. Sommigen prefereren het niet om Docker volumes te gebruiken en houden de data liever op een locatie naar keuze, zolang het pad maar bestaat en de Docker container voldoende rechten heeft.

xirixiz, 2020-08-20:
Dit zou naar behoren moeten werken. Ik denk dat je vergeten was ```DSMR_RELEASE=latest_tag``` als environment variabele te zetten. De development image werkt alleen met versie 4 en hoger, maar ben nog bezig het ook compatible te maken/houden met v3.

DSRM-Reader:
```
dsmr-reader:
  container_name: dsmr-reader
  image: xirixiz/dsmr-reader-docker:DEVELOPMENT-arm32v7
  restart: unless-stopped
  volumes:
    - /etc/localtime:/etc/localtime:ro
    - ./dsmr/backups:/dsmr/backups
    - /dev:/dev
  environment:
    - DB_HOST=dsmr-postgreSQL
    - DB_HOST=dsmrdb
    - DB_USER=dsmrreader
    - DB_PASS=dsmrreader
    - DB_PORT=5432
    - SD_LOGLEVEL=info
    - SD_AUTOSTART_MQTT=false
    - SD_AUTORESTART_MQTT=false
    - DSMRREADER_LOGLEVEL=WARNING
    - DSMR_USER=admin
    - DSMR_EMAIL=root@localhost
    - DSMR_RELEASE=latest_tag
    - DSMR_PASSWORD=admin
  ports:
    - '8888:80'
    - '8889:443'
  depends_on:
    - dsmr-postgreSQL
  cap_add:
    - NET_ADMIN        
  networks:
    - net-dsmr
```

DSMR-DB
```
dsmr-postgreSQL:
  container_name: dsmr-postgreSQL
  image: arm32v7/postgres:latest
  restart: unless-stopped
  ports:
    - '5432:5432'
  volumes:
    - /etc/localtime:/etc/localtime:ro
    - ./db/:/var/lib/postgresql/data
  environment:
    - POSTGRES_USER=dsmrreader
    - POSTGRES_PASSWORD=dsmrreader
    - POSTGRES_DB=dsmrreader
  networks:
    - net-dsmr
```

Erik050572, 2020-08-20:
> /tmp/supervisord.log

2020-08-20 18:35:44,199 INFO Included extra file "/etc/supervisor.d/supervisord.ini" during parsing
2020-08-20 18:35:44,199 INFO Set uid to user 0 succeeded
2020-08-20 18:35:44,243 INFO RPC interface 'supervisor' initialized
2020-08-20 18:35:44,244 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2020-08-20 18:35:44,247 INFO supervisord started with pid 211
2020-08-20 18:35:45,258 INFO spawned: 'dsmr_backend' with pid 213
2020-08-20 18:35:45,271 INFO spawned: 'dsmr_datalogger' with pid 214
2020-08-20 18:35:45,289 INFO spawned: 'dsmr_webinterface' with pid 215
2020-08-20 18:35:45,302 INFO spawned: 'nginx' with pid 216
2020-08-20 18:35:46,308 INFO success: dsmr_backend entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-20 18:35:46,310 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-20 18:35:46,312 INFO success: nginx entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-20 18:35:48,696 INFO success: dsmr_webinterface entered RUNNING state, process has stayed up for > than 3 seconds (startsecs)
2020-08-20 18:35:56,447 INFO exited: dsmr_datalogger (exit status 0; expected)
2020-08-20 18:35:57,456 INFO spawned: 'dsmr_datalogger' with pid 225
2020-08-20 18:35:58,461 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-20 18:36:07,931 INFO exited: dsmr_datalogger (exit status 0; expected)
2020-08-20 18:36:08,943 INFO spawned: 'dsmr_datalogger' with pid 227
2020-08-20 18:36:09,949 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-20 18:36:19,431 INFO exited: dsmr_datalogger (exit status 0; expected)
2020-08-20 18:36:20,441 INFO spawned: 'dsmr_datalogger' with pid 229
2020-08-20 18:36:21,493 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-20 18:36:31,927 INFO exited: dsmr_datalogger (exit status 0; expected)
2020-08-20 18:36:32,935 INFO spawned: 'dsmr_datalogger' with pid 231
2020-08-20 18:36:33,938 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-20 18:36:43,325 INFO exited: dsmr_datalogger (exit status 0; expected)
2020-08-20 18:36:44,338 INFO spawned: 'dsmr_datalogger' with pid 233
2020-08-20 18:36:45,344 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-20 18:36:54,767 INFO exited: dsmr_datalogger (exit status 0; expected)
2020-08-20 18:36:55,778 INFO spawned: 'dsmr_datalogger' with pid 235
2020-08-20 18:36:56,784 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-20 18:37:06,248 INFO exited: dsmr_datalogger (exit status 0; expected)
2020-08-20 18:37:07,260 INFO spawned: 'dsmr_datalogger' with pid 237
2020-08-20 18:37:08,265 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-20 18:37:17,889 INFO exited: dsmr_datalogger (exit status 0; expected)
2020-08-20 18:37:18,899 INFO spawned: 'dsmr_datalogger' with pid 244
2020-08-20 18:37:19,904 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-08-20 18:37:29,347 INFO exited: dsmr_datalogger (exit status 0; expected)
2020-08-20 18:37:30,355 INFO spawned: 'dsmr_datalogger' with pid 246
2020-08-20 18:37:31,359 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)

Erik050572, 2020-08-20:
Mijn docker-compose.yaml bestand op dit moment

version: '3.6'

services:

   dsmr-postgreSQL:
     container_name: dsmr-postgreSQL
     image: arm32v7/postgres:latest
     restart: unless-stopped
     ports:
        - '5432:5432'
     volumes:
        - /etc/localtime:/etc/localtime:ro
        - /db/:/var/lib/postgresql/data
     environment:
        - POSTGRES_USER=dsmrreader
        - POSTGRES_PASSWORD=dsmrreader
        - POSTGRES_DB=dsmrreader
     networks:
        - net-dsmr

   dsmr-influxdb:
     container_name: dsmr-influxdb
     image: influxdb:latest
     restart: unless-stopped
     ports:
        - '8086:8086'
     volumes:
        - influxdb-storage:/var/lib/influxdb
     environment:
        - INFLUXDB_DB= dsmrreader
        - INFLUXDB_ADMIN_USER= dsmrreader
        - INFLUXDB_ADMIN_PASSWORD= dsmrreader
     depends_on:
        - dsmr-postgreSQL
     networks:
        - net-dsmr

   dsmr-grafana:
     container_name: dsmr-grafana
     image: grafana/grafana:latest
     restart: unless-stopped
     ports:
        - '3000:3000'
     volumes:
        - grafana-storage:/var/lib/grafana
        - /grafana-provisioning/:/etc/grafana/provisioning
     depends_on:
        - dsmr-influxdb
     environment:
        - GF_SECURITY_ADMIN_USER=dsmrreader
        - GF_SECURITY_ADMIN_PASSWORD=dsmrreader
     networks:
        - net-dsmr

   dsmr-reader:
    container_name: dsmr-reader
**#**    image: ualex73/dsmr-reader-docker:arm32v6
    image: xirixiz/dsmr-reader-docker:DEVELOPMENT-arm32v7 
    restart: unless-stopped
    volumes:
        - /etc/localtime:/etc/localtime:ro
        - /dsmr/backups:/home/dsmr/app/backups
        - /dev:/dev
    environment:
        - DB_ENGINE=django.db.backends.postgresql
        - DB_NAME=dsmrreader
        - DB_USER=dsmrreader 
        - DB_PASS=dsmrreader
        - DB_HOST=dsmr-postgreSQL
        - DB_PORT=5432
        - SD_LOGLEVEL=info
        - SD_AUTOSTART_MQTT=false
        - SD_AUTORESTART_MQTT=false
        - DSMRREADER_LOGLEVEL=WARNING
        - CONN_MAX_AGE=60
        - DSMR_USER=admin
        - DSMR_PASSWORD=admin
        - DSMR_EMAIL=oot@localhost 
        - DSMR_RELEASE=latest_tag
        - VIRTUAL_HOST=localhost
        - TZ=Europe/Amsterdam
    depends_on:
        - dsmr-postgreSQL
    cap_add:
        - NET_ADMIN   
    ports:
        - '8888:80'
        - '8889:443'
**#**    devices:
**#**        - /dev/ttyUSB0:/dev/ttyUSB0
    networks:
        - net-dsmr

networks:
  net-dsmr:
    driver: bridge

volumes:
  influxdb-storage:
  grafana-storage:
  grafana-provisioning:
  backups:
  dev:
  db:
  localtime:

xirixiz, 2020-08-20:
@Erik050572 Kan je code tussen quotes zetten, dan krijg je de code in blokken te zien die goed leesbaar zijn, maar dat terzijde :): https://docs.github.com/en/github/writing-on-github/basic-writing-and-formatting-syntax

Kan je wel inloggen op de webinterface, want dat lijkt allemaal netjes te zijn gestart. Mogelijk dat de datalogger instellingen niet goed staan. ```http://<ip address>:8888/admin/dsmr_datalogger/dataloggersettings/```

Wat je zou verder kunnen doen is het loglevel op DEBUG zetten, dat geeft wellicht meer inzicht in het issue. ```DSMRREADER_LOGLEVEL=DEBUG```

dennissiemensma, 2020-08-20:
Nog ter aanvulling, DSMR-reader v4.2 maakt debugging van de datalogger wat extra verbose. 

Met bovenstaande DSMRREADER_LOGLEVEL-suggestie van xirixiz zou je zoiets in de logs van de datalogger moeten zien, dus ook als er geen data is:
```
[2020-08-20 21:09:34,278] INFO     [2020-08-20 21:09:34.278412] Opening serial connection "/dev/ttyUSB0" using options: 
{'baudrate': 115200, 'bytesize': 8, 'parity': 'N', 'stopbits': 1, 'xonxoff': 1, 'rtscts': 0}
[2020-08-20 21:09:34,534] DEBUG    [2020-08-20 21:09:34.534800] Read 1 Byte(s)
[2020-08-20 21:09:34,785] DEBUG    [2020-08-20 21:09:34.785841] Read 0 Byte(s)
[2020-08-20 21:09:35,036] DEBUG    [2020-08-20 21:09:35.036403] Read 0 Byte(s)
[2020-08-20 21:09:35,287] DEBUG    [2020-08-20 21:09:35.286955] Read 0 Byte(s)
[2020-08-20 21:09:35,537] DEBUG    [2020-08-20 21:09:35.537550] Read 0 Byte(s)
[2020-08-20 21:09:35,788] DEBUG    [2020-08-20 21:09:35.788419] Read 0 Byte(s)
[2020-08-20 21:09:36,039] DEBUG    [2020-08-20 21:09:36.039251] Read 0 Byte(s)
[2020-08-20 21:09:36,290] DEBUG    [2020-08-20 21:09:36.290112] Read 0 Byte(s)
[2020-08-20 21:09:36,541] DEBUG    [2020-08-20 21:09:36.540986] Read 0 Byte(s)
[2020-08-20 21:09:36,792] DEBUG    [2020-08-20 21:09:36.792011] Read 0 Byte(s)
[2020-08-20 21:09:37,043] DEBUG    [2020-08-20 21:09:37.042987] Read 0 Byte(s)
[2020-08-20 21:09:37,294] DEBUG    [2020-08-20 21:09:37.294075] Read 0 Byte(s)
[2020-08-20 21:09:37,545] DEBUG    [2020-08-20 21:09:37.545497] Read 0 Byte(s)
[2020-08-20 21:09:37,797] DEBUG    [2020-08-20 21:09:37.797189] Read 0 Byte(s)
[2020-08-20 21:09:38,049] DEBUG    [2020-08-20 21:09:38.048943] Read 0 Byte(s)
[2020-08-20 21:09:38,300] DEBUG    [2020-08-20 21:09:38.300732] Read 0 Byte(s)
[2020-08-20 21:09:38,552] DEBUG    [2020-08-20 21:09:38.552426] Read 0 Byte(s)
[2020-08-20 21:09:38,804] DEBUG    [2020-08-20 21:09:38.804098] Read 0 Byte(s)
[2020-08-20 21:09:39,055] DEBUG    [2020-08-20 21:09:39.055738] Read 0 Byte(s)
[2020-08-20 21:09:39,307] DEBUG    [2020-08-20 21:09:39.307405] Read 0 Byte(s)
[2020-08-20 21:09:39,559] DEBUG    [2020-08-20 21:09:39.559028] Read 0 Byte(s)
[2020-08-20 21:09:39,810] DEBUG    [2020-08-20 21:09:39.810724] Read 0 Byte(s)
[2020-08-20 21:09:40,062] DEBUG    [2020-08-20 21:09:40.062452] Read 0 Byte(s)
[2020-08-20 21:09:40,314] DEBUG    [2020-08-20 21:09:40.314168] Read 0 Byte(s)
[2020-08-20 21:09:40,565] DEBUG    [2020-08-20 21:09:40.565844] Read 0 Byte(s)
[2020-08-20 21:09:40,817] DEBUG    [2020-08-20 21:09:40.817536] Read 0 Byte(s)
[2020-08-20 21:09:41,069] DEBUG    [2020-08-20 21:09:41.069199] Read 0 Byte(s)
[2020-08-20 21:09:41,320] DEBUG    [2020-08-20 21:09:41.320898] Read 0 Byte(s)
[2020-08-20 21:09:41,572] DEBUG    [2020-08-20 21:09:41.572563] Read 0 Byte(s)
[2020-08-20 21:09:41,823] DEBUG    [2020-08-20 21:09:41.823847] Read 0 Byte(s)
[2020-08-20 21:09:42,075] DEBUG    [2020-08-20 21:09:42.075534] Read 0 Byte(s)
[2020-08-20 21:09:42,327] DEBUG    [2020-08-20 21:09:42.327188] Read 0 Byte(s)
[2020-08-20 21:09:42,578] DEBUG    [2020-08-20 21:09:42.578846] Read 0 Byte(s)
[2020-08-20 21:09:42,830] DEBUG    [2020-08-20 21:09:42.830779] Read 0 Byte(s)
[2020-08-20 21:09:43,082] DEBUG    [2020-08-20 21:09:43.082088] Read 66 Byte(s)
[2020-08-20 21:09:43,334] DEBUG    [2020-08-20 21:09:43.334030] Read 480 Byte(s)
[2020-08-20 21:09:43,586] DEBUG    [2020-08-20 21:09:43.586017] Read 316 Byte(s)
[2020-08-20 21:09:43,838] DEBUG    [2020-08-20 21:09:43.838059] Read 7 Byte(s)
[2020-08-20 21:09:43,845] DEBUG    Received telegram:
/XMX5LGBBFFB231117727

... telegram ...

!C6F2

```

Op de eerste regel staan de opties die gebruikt worden, die kun je dubbelchecken. Voor de rest zie je dus telkens pogingen tot uitlezen. In mijn geval duurt het met DSMRv4-meter tot 10 seconden voordat er iets te lezen is.

Als je ook een DSMRv4-meter hebt plus DSMR-reader < v4.2, dan kan het ook zijn dat het proces bewust na elke uitlezing zichzelf exit. Dat zie je ook aan de log: `exited: dsmr_datalogger (exit status 0; expected)`.

xirixiz, 2020-08-20:
Overigens heb ik zojuist de image officieel released. De image naam is nu, al is DEVELOPMENT op dit moment nog exact hetzelfde.

```xirixiz/dsmr-reader-docker:latest-<arch>```

Erik050572, 2020-08-20:
Bedankt voor de link, ik was al aan het zoeken hoe ik een remark niet hals header weergegeven krijg.

http://192.168.1.47:8888/admin/dsmr_datalogger/dataloggersettings/  geeft ook This site can’t be reached

Ik weet niet of ''''tail -f < /dev/<device>'''' voor mij gaat werken omdat deze computer niet direct met de meterkast is verbonden maar via een API de data krijgt

Ik heb het  "development" gedeelte verwijderd uit de image en heb de debug optie toegevoegd.
De hierboven genoemde bestanden zien er niet anders uit.
Of  wordt dat ergens anders in opgeslagen.

Erik050572, 2020-08-27:
Tussentijdse update:
De container "dsmr-reader" werkt intussen.
Ik ga nu proberen om de remote datalogger in het yaml-bestand te verwerken.

ualex73, 2020-08-27:
@Erik050572 dan ben ik toch benieuwd waar het "probleem" zat en wat de oplossing was? :-)

Erik050572, 2020-08-27:
Dat is nu het stomme.... Ik weet het niet....
Ik heb via portainer de vastgelopen container geherstart en toen bleef hij draaien.

Erik050572, 2020-08-29:
Goedemiddag @dennissiemensma ,

Vandaag heb je de update van V4.2 naar V4.3 gepushed
De website van de Pi is gewoon te benaderen en ook geeft het status overzicht geen backlog aan.
Alleen komt de update naar [PVoutput.org](https://pvoutput.org/intraday.jsp?id=40142&sid=36675&dt=20200828&gs=0&m=1) niet door komt.

Hieroner de logbestanden

sudo tail -f /var/log/supervisor/dsmr_webinterface*
==> /var/log/supervisor/dsmr_webinterface.log <==
[2020-08-27 08:23:29 +0200] [29837] [INFO] Booting worker with pid: 29837
[2020-08-27 08:28:53 +0200] [29837] [INFO] Autorestarting worker after current r                                                                                                                                   equest.
[2020-08-27 08:28:53 +0200] [29837] [INFO] Worker exiting (pid: 29837)
[2020-08-27 08:28:54 +0200] [2613] [INFO] Booting worker with pid: 2613
[2020-08-27 08:34:08 +0200] [2613] [INFO] Autorestarting worker after current re                                                                                                                                   quest.
[2020-08-27 08:34:08 +0200] [2613] [INFO] Worker exiting (pid: 2613)
[2020-08-27 08:34:09 +0200] [7492] [INFO] Booting worker with pid: 7492
[2020-08-27 08:39:31 +0200] [7492] [INFO] Autorestarting worker after current re                                                                                                                                   quest.
[2020-08-27 08:39:31 +0200] [7492] [INFO] Worker exiting (pid: 7492)
[2020-08-27 08:39:32 +0200] [12492] [INFO] Booting worker with pid: 12492

==> /var/log/supervisor/dsmr_webinterface.log.1 <==
[2020-07-24 23:26:12 +0200] [31333] [INFO] Booting worker with pid: 31333
[2020-07-24 23:27:12 +0200] [776] [CRITICAL] WORKER TIMEOUT (pid:31333)
[2020-07-24 23:27:12 +0200] [31333] [INFO] Worker exiting (pid: 31333)
[2020-07-24 23:27:14 +0200] [31358] [INFO] Booting worker with pid: 31358
[2020-07-24 23:28:14 +0200] [776] [CRITICAL] WORKER TIMEOUT (pid:31358)
[2020-07-24 23:28:14 +0200] [31358] [INFO] Worker exiting (pid: 31358)
[2020-07-24 23:28:23 +0200] [31372] [INFO] Booting worker with pid: 31372
[2020-07-24 23:29:23 +0200] [776] [CRITICAL] WORKER TIMEOUT (pid:31372)
[2020-07-24 23:29:24 +0200] [31372] [INFO] Worker exiting (pid: 31372)
[2020-07-24 23:29:30 +0200] [31393] [INFO] Booting worker with pid: 31393

sudo tail -f /var/log/supervisor/dsmr_datalogger*
==> /var/log/supervisor/dsmr_datalogger.log <==
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging

==> /var/log/supervisor/dsmr_datalogger.log.1 <==
plugin test
plugin test
plugin test
plugin test
plugin test
plugin test
plugin test
plugin test
plugin test
plugin test

==> /var/log/supervisor/dsmr_datalogger.log.2 <==
plugin test
plugin test
plugin test
plugin test
plugin test
plugin test
plugin test
plugin test
plugin test
plugin test

[... 84 lines cut ...]
django.core.exceptions.ImproperlyConfigured: Error loading psycopg2 module: No module named 'psycopg2'

==> /var/log/supervisor/dsmr_backend.log.4 <==
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/apps/config.py", line 211, in import_models
    self.models_module = import_module(models_module_name)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed

==> /var/log/supervisor/dsmr_backend.log.5 <==
    return getattr(connections[DEFAULT_DB_ALIAS], item)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/utils.py", line 207, in __getitem__
    backend = load_backend(db['ENGINE'])
[... cut ...]
    return import_module('%s.base' % backend_name)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/postgresql/base.py", line 29, in <module>
    raise ImproperlyConfigured("Error loading psycopg2 module: %s" % e)
django.core.exceptions.ImproperlyConfigured: Error loading psycopg2 module: No module named 'psycopg2'

dennissiemensma, 2020-08-29:
Ik denk dat je het beste hier een **apart issue** voor kan maken, gezien Github het beste werkt met opsplitsing van verschillende problemen.

Verder kun je proberen om de logfiles in ` /var/log/supervisor/` weg te gooien, want er lijken wat oude meldingen tussen te zitten. En tot slot debug logging aanzetten zodat je in de log van `backend` kan zien wat er gebeurt qua PVOutput uploads.

closed 2020-08-29 by dennissiemensma

Erik050572, 2020-08-30:
Ik zal dat doen, mocht het nog een keer voorkomen, maar voor de hoofdvraag  heb Ik toch nog wel hulp nodig.

Ik heb de stappen uitgevoerd, van de [werkinstructie](https://dsmr-reader.readthedocs.io/en/latest/installation/datalogger.html)  in RPi3 (meterkast) om de data door te sturen naar RPIi (docker: de "datalogger")

Hoe roep ik het `home/dsmr/dsmr_datalogger_api_client.py` script aan vanuit het Yaml-bestand?
Ik heb wel een [uitleg](https://docs.docker.com/compose/gettingstarted/) gevonden, maar ik denk dat op die manier onnodig ingewikkeld wordt gemaakt.

dennissiemensma, 2020-08-31:
Ik denk dat je vragen over hoe de remote datalogger te draaien via Docker het beste [bij xirixiz](https://github.com/xirixiz/dsmr-reader-docker/issues) (of ualex) kunt leggen.

Als je op de meterkast-Pi **geen Docker** gebruikt, kun je de standaard-instructies volgen: https://dsmr-reader.readthedocs.io/en/latest/installation/datalogger.html

xirixiz, 2020-08-31:
Ik kan daar wel bij helpen, ik kom er morgen op terug.

xirixiz, 2020-08-31:
Inmiddels heb ik de development images bijgewerkt met de functionaliteit om de Docker container in te zetten als remote datalogger via een seriele verbinding of via ipv4. Morgen kan ik het testen, of je kan het zelf testen met de development image als de docu helder genoeg is.. de development image heb je alleen nodig op de locatie waar je de meter gaat uitlezen, aan de andere kant veranderd er niets maar moet je wel de "local" datalogger uitzetten uiteraard.

xirixiz, 2020-09-01:
```https://github.com/xirixiz/dsmr-reader-docker/tree/development``` is bijgewerkt. Dat houdt dus in dat de development Docker image ingezet kan worden als remote datalogger. De README in de development branch is ook bijgewerkt, en met die info moet het vast goedkomen. Succes! Wanneer het werkt merge ik het uiteraard naar de master branch.

Erik050572, 2020-09-01:
We must be nearly there.....  
Hier was ik dus never... nooit niet.... opgekomen!
Wat een heldere uitleg. 👍 

Ik krijg nu een ander foutmelding op webpagina van de "Docker versie"
`Geen gegevens gevonden. Zorg ervoor dat het dsmr_datalogger-proces draait of dat je gegevens aanlevert via die API. 
Op het moment dat ik "REMOTE_DATALOGGER" van false naar true verander, schiet de container weer in de "dsmr_datalogger loop".

Voor de zekerheid:
Op de dataverzendende RPi (de stand-alone versie in de meterkast) heb ik de API aangezet en de daarvermelde code overgenomen in het YAML-bestand. In het dataloggerconfiguratie scherm heb ik niets veranderd, want hier komt de data binnen.
(ook heb ik het dataloggen (nog) niet uit gezet omdat dit tot  nu toe enige werkende versie is in versie 4.)

Het beest van een YAML-bestand  van de dataontvangende RPi ziet er nu inmiddels zo uit
```
version: '3.3'

services:

   dsmr-postgreSQL:
     container_name: dsmr-postgreSQL
     image: arm32v7/postgres:latest
     restart: always
     ports:
        - '5432:5432'
     volumes:
        - /etc/localtime:/etc/localtime:ro
        - /db/:/var/lib/postgresql/data
     environment:
        - TZ=Europe/Amsterdam
        - PG_TZ=Europe/Amsterdam
        - POSTGRES_USER=dsmrreader
        - POSTGRES_PASSWORD=dsmrreader
        - POSTGRES_DB=dsmrreader
     networks:
        - net-dsmr

   dsmr-influxdb:
     container_name: dsmr-influxdb
     image: influxdb:latest
     restart: unless-stopped
     ports:
        - '8086:8086'
     volumes:
        - /influxdb-storage:/var/lib/influxdb
     environment:
        - INFLUXDB_DB= dsmrreader
        - INFLUXDB_ADMIN_USER= dsmrreader
        - INFLUXDB_ADMIN_PASSWORD= dsmrreader
     depends_on:
        - dsmr-postgreSQL
     networks:
        - net-dsmr

   dsmr-grafana:
     container_name: dsmr-grafana
     image: grafana/grafana:latest
     restart: always
     ports:
        - '3000:3000'
     volumes:
        - ./grafana-provisioning/:/etc/grafana/provisioning
     depends_on:
        - dsmr-influxdb
     environment:
        - GF_SECURITY_ADMIN_USER=dsmrreader
        - GF_SECURITY_ADMIN_PASSWORD=dsmrreader
     networks:
        - net-dsmr

   dsmr-reader:
    container_name: dsmr-reader
#    image: ualex73/dsmr-reader-docker:arm32v6
    image: xirixiz/dsmr-reader-docker:development-arm32v7
    restart: always
    volumes:
        - /etc/localtime:/etc/localtime:ro
        - ./dsmr/backups:/home/dsmr/app/backups
        - /dev:/dev
    environment:
#        DB related
        - DB_ENGINE=django.db.backends.postgresql
        - DB_NAME=dsmrreader
        - DB_USER=dsmrreader
        - DB_PASS=dsmrreader
        - DB_HOST=dsmr-postgreSQL
        - DB_PORT=5432
        - CONN_MAX_AGE=60
#        DSMR related
        - DSMR_USER=admin
        - DSMR_PASSWORD=admin
        - DSMRREADER_LOGLEVEL=DEBUG # Valid values are WARNING, INFO, DEBUG
#        - DSMRREADER_PLUGINS=dsmr_plugins.modules.plugin_name1,dsmr_plugins.modules.plugin_name2
        - SECRET_KEY=dsmrreader
        - DSMR_RELEASE=latest       # Can be 'latest', 'latest_tag' or specific version like '4.1.1'
        - VIRTUAL_HOST=localhost
        - TZ=Europe/Amsterdam
#        Nginx related
        - ENABLE_HTTP_AUTH=false
        - HTTP_AUTH_USERNAME=
        - HTTP_AUTH_PASSWORD=
#        Supervisord related
        - SD_LOGLEVEL=info
        - SD_USER=root
        - SD_GROUP=root
        - SD_AUTOSTART_DATALOGGER=true
        - SD_AUTORESTART_DATALOGGER=true
        - SD_AUTOSTART_REMOTE_DATALOGGER=false
        - SD_AUTORESTART_REMOTE_DATALOGGER=false
        - SD_AUTOSTART_BACKEND=true
        - SD_AUTORESTART_BACKEND=true
        - SD_AUTOSTART_WEBINTERFACE=true
        - SD_AUTORESTART_WEBINTERFACE=true
#        Remote DSMR datalogger related (more info: https://dsmr-reader.readthedocs.io/nl/v4/installation/datalogger.html)
        - REMOTE_DATALOGGER=false             # Set to true if datalogger is running on another container
        - DATALOGGER_API_HOSTS=1              # Required. Destination(s) of the DSMR Reader (Docker) host(s)
        - DATALOGGER_API_KEYS=CNM3BS0XEU086TSIX2CGWAFR3F3RCT9M6ALEURKT4SK4AOSN8ZQCLEWCF8WW44CX               # Required. Add the API keys of the DSMR Reader (Docker) destination host(s)
        - DATALOGGER_INPUT_METHOD=ipv4         # Required. Only serial or ipv4 (network) are valid values
#        - DATALOGGER_SERIAL_PORT=/dev/ttyUSB0 # Required if the input method is set to serial
#        - DATALOGGER_SERIAL_BAUDRATE=115200   # Required if the input method is set to serial
        - DATALOGGER_NETWORK_HOST=192.168.1.47 # Required if the input method is set to ipv4
        - DATALOGGER_NETWORK_PORT=80           # Required if the input method is set to ipv4 (Docker port)
#       - DATALOGGER_TIMEOUT=x                # Optional. In seconds
#       - DATALOGGER_SLEEP=x                  # Optional. In seconds
#       - DATALOGGER_DEBUG_LOGGING=false      # Optional.
     depends_on:
        - dsmr-postgreSQL
    cap_add:
        - NET_ADMIN
    links:
        - dsmr-postgreSQL
    ports:
        - '8888:80'
        - '8889:443'
#    devices:
#        - /dev/ttyUSB0:/dev/ttyUSB0
    networks:
        - net-dsmr

networks:
  net-dsmr:
    driver: bridge

volumes:
  influxdb-storage:
  grafana-storage:
  grafana-provisioning:
  backups:
  dev:
  db:
  localtime:

```

xirixiz, 2020-09-02:
Je bent er bijna 👍.

Deze waarden kunnen op de ontvangende container gezet worden (pas ik nog aan, kleine denkfout van mij inderdaad):
```
        - SD_AUTOSTART_DATALOGGER=false
        - SD_AUTORESTART_DATALOGGER=false
        # - REMOTE_DATALOGGER=false
        # - DATALOGGER_API_HOSTS=1
        # - DATALOGGER_API_KEYS=xxxx
        # - DATALOGGER_INPUT_METHOD=ipv4 
        # - DATALOGGER_NETWORK_HOST=192.168.1.47
        # - DATALOGGER_NETWORK_PORT=80
```

Deze waarden moet je op de datalogger container zetten:
```
        - REMOTE_DATALOGGER=true
        - DATALOGGER_API_HOSTS=x.x.x.x
        - DATALOGGER_API_KEYS=xxxxxxxxxxxxx
        - DATALOGGER_INPUT_METHOD=ipv4 
        - DATALOGGER_NETWORK_HOST=192.168.1.47
        - DATALOGGER_NETWORK_PORT=80
```

- Kijk even in de documentatie was de waarde moet zijn voor ```DATALOGGER_API_HOSTS```, daar moet een IP staan: https://dsmr-reader.readthedocs.io/en/latest/installation/datalogger.html
- Draait de Docker daemon van de datalogger host op dit IP adres ```DATALOGGER_NETWORK_HOST=192.168.1.47```?
- Draait de Docker container op poort ```DATALOGGER_NETWORK_PORT=80```, of is die voor de datalogger ook gemapped op poort 8888:80 dus dat de poort 8888 moet zijn?.

@dennissiemensma weet overigens meer over de configuratie zelf hiervan, wellicht dat hij de waarden kan beoordelen.

Succes nog! Je bent er bijna volgens mij!

xirixiz, 2020-09-02:
De devlopement branch heb ik bijgewerkt (code en docs). De eerdere comments die ik heb verzonden vandaag kunnen dus genegeerd worden bij gebruik van de laatste development image

- Removed REMOTE_DATALOGGER
- Added DATALOGGER_MODE

De datalogger mode setting kent drie opties:
- standalone (default, eigenlijk een single setup)
- sender (split setup, een container die alleen als verdender dient, dus de datalogger verzender gekoppeld aan P1)
- receiver (split setup, een contaer die de data ontvangt van de sender container, maar waarbij de local datalogger in de container is uitgeschakeld, die krijgt immers geen data omdat de sender dat verzorgt)

Volgens mij moet je er op deze manier wel uitkomen. 

Aangezien de DATALOGGER_MODE nu alle opstellingen coveren heb ik de volgende opties verwijderd:
- SD_AUTOSTART_.*
- SD_AUTORESTART.*

Erik050572, 2020-09-02:
Goedemorgen @xirixiz ,

Als ik jouw antwoord lees, krijg ik de indruk dat je er van uitgaat dat  mijn setup uit 2 docker opstellingen bestaat.
Op de Rpi3 in de meterkast draait geen docker en is een standalone en daarom heb ik deze [stappen ](https://maxxie85.tweakblogs.net/) niet uitgevoerd.

Ik zal vanavond even nog eens controleren of ik niet een fout heb gemaakt in dit gedeelte van de [werkinstructie](https://dsmr-reader.readthedocs.io/en/latest/installation/datalogger.html#api-config-env).

Zit er eigenlijk een verschil de werking van datavelden omgeven met of zonder ""?

Bijvoorbeeld
```
# Example with non standard port, e.g. Docker:
DATALOGGER_API_HOSTS=http://12.34.56.78:7777
DATALOGGER_API_KEYS=0987654321HGFEDCBA
```
Tegenover
```
# Example with non standard port, e.g. Docker:
DATALOGGER_API_HOSTS="http://12.34.56.78:7777"
DATALOGGER_API_KEYS="0987654321HGFEDCBA"
```

xirixiz, 2020-09-02:
Ah ok, dan moet je de Docker variant, dus instellen als ``receiver``` voor de DATALOGGER_MODE, dan schakel je de datalogger uit op de Docker container. 

De andere waarden zoals DATLOGGER_API_HOSTS en dergelijke worden alleen door de ```sender``` docker container gebruikt, of bij een handmatige configuratie zoals die in de handleiding van @dennissiemensma beschreven staan: https://dsmr-reader.readthedocs.io/en/latest/installation/datalogger.html, maar aangezien je deze stappen hebt gevolgd: https://maxxie85.tweakblogs.net/ is het voor jou niet van toepassing

Het enige dat je dus hoeft in te stellen is, alle overige DATLOGGER.* config kan je uit je compose file verwijderen.
```
        - DATALOGGER_MODE="receiver"
```

Je compose file in de environment sectie zou er dan als volgt uit zien (default waarden zou ik nooit opnemen in de compose file):
```
    environment:
        - DATALOGGER_MODE="receiver"
        - DB_HOST="dsmr-postgreSQL"
        - VIRTUAL_HOST="localhost"
        - TZ="Europe/Amsterdam"
```

Je kan overigens het beste de waarden tussen dubbele quotes zetten in dit geval om issues te voorkomen.

Vanuit de bash handleiding:

```
3.1.2.2 Single Quotes

Enclosing characters in single quotes (') preserves the literal value of each character within the quotes. A single quote may not occur between single quotes, even when preceded by a backslash.

3.1.2.3 Double Quotes

Enclosing characters in double quotes (") preserves the literal value of all characters within the quotes, with the exception of $, `, \, and, when history expansion is enabled, !. The characters $ and ` retain their special meaning within double quotes (see Shell Expansions). The backslash retains its special meaning only when followed by one of the following characters: $, `, ", \, or newline. Within double quotes, backslashes that are followed by one of these characters are removed. Backslashes preceding characters without a special meaning are left unmodified. A double quote may be quoted within double quotes by preceding it with a backslash. If enabled, history expansion will be performed unless an ! appearing in double quotes is escaped using a backslash. The backslash preceding the ! is not removed.

The special parameters * and @ have special meaning when in double quotes (see Shell Parameter Expansion).
```

[14 more replies]