dsmrreader/dsmr-reader #1077: DSMR postgresql backup

hansv50, 2020-08-11:
Dennis,
Hoe zet ik een DB terug van mijn NAS?
PATH TO BACKUP snap het helemaal niet!
De NAS  (QNAP) heeft IP-adres 192.168.1.XXX, de backup staat in de Dir <Backups>.
Heb al van alles geprobeerd, maar hij blijft zeggen 'no such file or directory'
HELP!! d'r staat data in vanaf het begin.

Het toch: //192.168.1.XXX/Backups/dsmrreader-postgresql-backup-Saturday.sql.gz
Of maak ik denkvaud? Moet ik de backup terugzetten naar de SD-card? (hoe doe ik dat?)

Met hartelijke groet, Hans Verbeek

dennissiemensma, 2020-08-11:
Bedankt voor je vraag. Ik begrijp niet helemaal je situatie. 

Heb je alleen de database op de NAs staan? Of ook DSMR-reader? Of alleen de backup op de NAS? Gebruik je daarnaast een Pi?

Ik weet niet hoe je de NAS extern kan benaderen. Wat ik je kan aanraden is om ervoor te zorgen dat je de backup op hetzelfde systeem zet als waar DSMR-reader draait. En dan geef je bij het terugzetten het lokale bestandspad op.

hansv50, 2020-08-11:
Dank voor het antwoord!
Ik heb de backup op de NAS staan en de DSMR-reader op de Pi.
Hoe doe ik de backup naar de Pi zetten?

dennissiemensma, 2020-08-11:
Ik kan je helaas inhoudelijk niet helpen omdat ik zelf geen NAS heb. Je kunt denk ik het beste zoeken naar een manier om de backup naar je Pi te krijgen. 
Probeer het vooral pragmatisch aan te vliegen, bijvoorbeeld met iets eenvoudigs zoals een USB-stick in de NAS en daarna in je Pi.

En anders kun je proberen om met `scp` de gegevens op je Pi te krijgen, maar dat is al een stuk geavanceerder en hangt af van de verbindingsmogelijkheden van je NAS.

hansv50, 2020-08-11:
Bedankt voor je tips. Ga het met een USB proberen

dennissiemensma, 2020-08-17:
@hansv50 is dit je uiteindelijk gelukt?

hansv50, 2020-08-17:
Hoi Dennis,
Ja, het is me gelukt!
Nadat ik de img. op een USB had gekopieerd was het zo gedaan. Ik had nog
wel wat moeite met het mounten van de USB maar daar is internet voor!

Rest me je nog te bedanken en verder nog te verdiepen in de Linux omgeving.
Bedankt!

Hans Verbeek

dennissiemensma, 2020-08-17:
Mooi, dank voor de terugkoppeling!

closed 2020-08-17 by dennissiemensma