dsmrreader/dsmr-reader #1078: DSMR Webinterface error: can't find command '/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn'

JohanEikelenboom, 2020-08-13:
Hi,

This morning I updated my DSMRReader from version 3.x to 4.x only now the webinterface seems to be broken.

When executing a ``sudo supervisorctl status`` I get the following status:
```
dsmr_backend                     RUNNING   pid 1602, uptime 0:01:18
dsmr_datalogger                  RUNNING   pid 1627, uptime 0:00:26
dsmr_webinterface                FATAL     can't find command '/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn'
```

Any hints / solutions how to solve this? 

The backend and datalogger are running as expected as far as I can see (mqtt messages are being sent).
I removed the file /var/log/supervisor/dsmr_webinterface.log and restarted DSMR, but nothing appears in the log

dennissiemensma, 2020-08-13:
Thanks for reporting. Can you try the suggestions in the following issue as a start? 

- #1065

JohanEikelenboom, 2020-08-13:
I just checked the post and it seems that everything is correct:
When I enter ``sudo -su dsmr`` I get the prompt ``(dsmrreader) dsmr@DSMRReader:~/dsmr-reader $`` (that should be right if I'm correct.

When I run ``pip3 install -r dsmrreader/provisioning/requirements/base.txt`` I get the following output:
```
(dsmrreader) dsmr@DSMRReader:~/dsmr-reader $ pip3 install -r dsmrreader/provisioning/requirements/base.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: coreapi==2.3.3 in /home/dsmr/.local/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 1)) (2.3.3)
Requirement already satisfied: crcmod==1.7 in /home/dsmr/.local/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 2)) (1.7)
Requirement already satisfied: django==3.0.8 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 3)) (3.0.8)
Requirement already satisfied: django-admin-rangefilter==0.6.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 4)) (0.6.1)
Requirement already satisfied: django-colorfield==0.3.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 5)) (0.3.2)
Requirement already satisfied: django-debug-toolbar==2.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 6)) (2.2)
Requirement already satisfied: django-filter==2.3.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 7)) (2.3.0)
Requirement already satisfied: djangorestframework==3.11 in /home/dsmr/.local/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 8)) (3.11.0)
Requirement already satisfied: django-solo==1.1.3 in /home/dsmr/.local/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 9)) (1.1.3)
Requirement already satisfied: dropbox==10.3.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 10)) (10.3.0)
Requirement already satisfied: gunicorn==20.0.4 in /home/dsmr/.local/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 11)) (20.0.4)
Requirement already satisfied: influxdb==5.3.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 12)) (5.3.0)
Requirement already satisfied: paho-mqtt==1.5.0 in /home/dsmr/.local/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 13)) (1.5.0)
Requirement already satisfied: pyserial==3.4 in /home/dsmr/.local/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 14)) (3.4)
Requirement already satisfied: pyserial-asyncio==0.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 15)) (0.4)
Requirement already satisfied: python-dateutil==2.8.1 in /home/dsmr/.local/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 16)) (2.8.1)
Requirement already satisfied: python-decouple==3.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 17)) (3.3)
Requirement already satisfied: pytz==2020.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 18)) (2020.1)
Requirement already satisfied: requests==2.24.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 19)) (2.24.0)
Requirement already satisfied: urllib3==1.25.10 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 20)) (1.25.10)
Requirement already satisfied: uritemplate in /home/dsmr/.local/lib/python3.7/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (3.0.1)
Requirement already satisfied: coreschema in /home/dsmr/.local/lib/python3.7/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (0.0.4)
Requirement already satisfied: itypes in /home/dsmr/.local/lib/python3.7/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (1.1.0)
Requirement already satisfied: sqlparse>=0.2.2 in /home/dsmr/.local/lib/python3.7/site-packages (from django==3.0.8->-r dsmrreader/provisioning/requirements/base.txt (line 3)) (0.3.0)
Requirement already satisfied: asgiref~=3.2 in /home/dsmr/.local/lib/python3.7/site-packages (from django==3.0.8->-r dsmrreader/provisioning/requirements/base.txt (line 3)) (3.2.3)
Requirement already satisfied: six>=1.12.0 in /usr/lib/python3/dist-packages (from dropbox==10.3.0->-r dsmrreader/provisioning/requirements/base.txt (line 10)) (1.12.0)
Requirement already satisfied: setuptools>=3.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from gunicorn==20.0.4->-r dsmrreader/provisioning/requirements/base.txt (line 11)) (49.3.2)
[... cut ...]
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests==2.24.0->-r dsmrreader/provisioning/requirements/base.txt (line 19)) (2018.8.24)
Requirement already satisfied: jinja2 in /home/dsmr/.local/lib/python3.7/site-packages (from coreschema->coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (2.10.3)
Requirement already satisfied: MarkupSafe>=0.23 in /home/dsmr/.local/lib/python3.7/site-packages (from jinja2->coreschema->coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (1.1.1)
```

dennissiemensma, 2020-08-13:
So what does this show as output?

```
ls -l /home/dsmr/.virtualenvs/dsmrreader/bin/
```

```
sudo su - dsmr
gunicorn --version
```

dennissiemensma, 2020-08-13:
I think I see the issue causing the error in your output:
```
Requirement already satisfied: gunicorn==20.0.4 in /home/dsmr/.local/lib/python3.7/site-packages
```

The path differs.

I think it is caused by the new `--system-site-packages` when creating the virtual env. Could you try creating a test virtualenv along the existing one? To see whether it solves the issue.

```
sudo su - dsmr
deactivate

virtualenv /home/dsmr/.virtualenvs/test-dsmrreader --python python3
source /home/dsmr/.virtualenvs/test-dsmrreader/bin/activate

pip3 install gunicorn
```

Now does the binary exist here?
```
ls -l /home/dsmr/.virtualenvs/test-dsmrreader/bin/gunicorn
```

dennissiemensma, 2020-08-13:
After testing you can delete the `/home/dsmr/.virtualenvs/test-dsmrreader/` folder and its contents.

JohanEikelenboom, 2020-08-14:
Output ``ls -l /home/dsmr/.virtualenvs/dsmrreader/bin/``:

```
-rw-r--r-- 1 dsmr dsmr    2093 Aug 13 07:32 activate
-rw-r--r-- 1 dsmr dsmr    1035 Aug 13 07:32 activate.csh
-rw-r--r-- 1 dsmr dsmr    2189 Aug 13 07:32 activate.fish
-rw-r--r-- 1 dsmr dsmr    1137 Aug 13 07:32 activate_this.py
-rwxr-xr-x 1 dsmr dsmr     293 Aug 13 07:34 django-admin
-rwxr-xr-x 1 dsmr dsmr     155 Aug 13 07:34 django-admin.py
-rwxr-xr-x 1 dsmr dsmr     263 Aug 13 07:32 easy_install
-rwxr-xr-x 1 dsmr dsmr     263 Aug 13 07:32 easy_install-3.7
-rwxr-xr-x 1 dsmr dsmr     254 Aug 13 07:32 pip
-rwxr-xr-x 1 dsmr dsmr     254 Aug 13 07:32 pip3
-rwxr-xr-x 1 dsmr dsmr     254 Aug 13 07:32 pip3.7
drwxr-xr-x 2 dsmr dsmr    4096 Aug 13 07:34 __pycache__
lrwxrwxrwx 1 dsmr dsmr       7 Aug 13 07:31 python -> python3
-rwxr-xr-x 1 dsmr dsmr 4275580 Aug 13 07:31 python3
lrwxrwxrwx 1 dsmr dsmr       7 Aug 13 07:31 python3.7 -> python3
-rwxr-xr-x 1 dsmr dsmr    2352 Aug 13 07:32 python-config
-rwxr-xr-x 1 dsmr dsmr     241 Aug 13 07:32 wheel
```

gunicorn version: ``gunicorn (version 20.0.4)``

After creating the test virtualenv, the file binary gunicorn exists:
``-rwxr-xr-x 1 dsmr dsmr 252 Aug 14 07:16 /home/dsmr/.virtualenvs/test-dsmrreader/bin/gunicorn``

dennissiemensma, 2020-08-14:
Great. Try recreating the virtualenv without `--system-site-packages`:

```
sudo supervisorctl stop all

sudo su - dsmr
deactivate
cd ~
mv .virtualenvs/dsmrreader .virtualenvs/old-v4-dsmrreader

virtualenv /home/dsmr/.virtualenvs/dsmrreader --system-site-packages --python python3
source ~/.virtualenvs/dsmrreader/bin/activate

pip3 install -r dsmrreader/provisioning/requirements/base.txt

logout
sudo supervisorctl restart all
```

Does that resolve your issue?

JohanEikelenboom, 2020-08-14:
Hmm... This kind of made the problem worse: Now it seems like the dsmr_backend, dsmr_datalogger and dsmr_webinterface are constantly restarting

```
pi@DSMRReader:~ $ sudo supervisorctl status
dsmr_backend                     STARTING
dsmr_datalogger                  RUNNING   pid 1056, uptime 0:00:02
dsmr_webinterface                STARTING
pi@DSMRReader:~ $ sudo supervisorctl status
dsmr_backend                     RUNNING   pid 1069, uptime 0:00:01
dsmr_datalogger                  RUNNING   pid 1056, uptime 0:00:03
dsmr_webinterface                STARTING
pi@DSMRReader:~ $ sudo supervisorctl status
dsmr_backend                     STARTING
dsmr_datalogger                  RUNNING   pid 1259, uptime 0:00:03
dsmr_webinterface                STARTING
pi@DSMRReader:~ $ sudo supervisorctl status
dsmr_backend                     RUNNING   pid 1271, uptime 0:00:03
dsmr_datalogger                  RUNNING   pid 1278, uptime 0:00:01
dsmr_webinterface                STARTING
pi@DSMRReader:~ $ sudo supervisorctl status
dsmr_backend                     RUNNING   pid 1293, uptime 0:00:01
dsmr_datalogger                  RUNNING   pid 1278, uptime 0:00:03
dsmr_webinterface                STARTING
pi@DSMRReader:~ $ sudo supervisorctl status
dsmr_backend                     STARTING
dsmr_datalogger                  RUNNING   pid 1306, uptime 0:00:01
dsmr_webinterface                STARTING
pi@DSMRReader:~ $ sudo supervisorctl status
dsmr_backend                     RUNNING   pid 1316, uptime 0:00:02
dsmr_datalogger                  STARTING
dsmr_webinterface                STARTING
```

dennissiemensma, 2020-08-14:
You will need to look into the logfiles in `/var/log/supervisor/` to get the specifics. 

Can you try this as well:

```
sudo su - dsmr
./deploy.sh
```

dennissiemensma, 2020-08-14:
And this file should exist now, can you check whether it does:
```
ls /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn
```

JohanEikelenboom, 2020-08-14:
Just checked and the file ``ls /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn`` does exist.

When running the ``./deploy.sh`` I get the following error:
```
ModuleNotFoundError: No module named 'psycopg2'
```

JohanEikelenboom, 2020-08-14:
Manually install psycopg2 using ``pip3 install psycopg2`` and now it looks like it's working

JohanEikelenboom, 2020-08-14:
Just confirmed: Everything is working as expected.

Thank you verry much for your help.

Another happy user :)

dennissiemensma, 2020-08-14:
Great! I guess I will need to think of some other way to skip the `--system-site-packages`, because I forgot it excludes the system-wide installed `psycopg2` as a side effect. 

Thanks for trying and debugging anyway!

closed 2020-08-14 by dennissiemensma

JohanEikelenboom, 2020-08-14:
No problem.