dsmrreader/dsmr-reader #1079: Unable to start MQTT and Debugging

mchlrv, 2020-08-13:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

Wat gebruik je?
- DSMR-reader versie:   4.1.1
- Type RaspberryPi or server:   Ubuntu 18.04.2 LTS
- Standaardinstallatie of Docker:   Standaardinstallatie
- DSMR-protocol slimme meter v4/v5:   4

---

Om DSMR-reader in Home Assistant te integreren wil ik MQTT gebruiken. Wanneer ik deze enable en de pagina na een paar seconden refresh is de optie weer disabled. Als ik vervolgens kijk in /var/log/syslog zie ik dat DSMR-reader 2x wordt herstart (zie onderstaande logs). Na de 2e keer is de optie weer disabled.

```
Aug 13 19:38:30 dsmr supervisord[951]: 2020-08-13 19:38:30,033 INFO exited: dsmr_backend (exit status 0; expected)
Aug 13 19:38:31 dsmr supervisord[951]: 2020-08-13 19:38:31,036 INFO spawned: 'dsmr_backend' with pid 20351
Aug 13 19:38:32 dsmr supervisord[951]: 2020-08-13 19:38:32,039 INFO success: dsmr_backend entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
Aug 13 19:38:33 dsmr supervisord[951]: 2020-08-13 19:38:33,849 INFO exited: dsmr_backend (exit status 0; expected)
Aug 13 19:38:34 dsmr supervisord[951]: 2020-08-13 19:38:34,853 INFO spawned: 'dsmr_backend' with pid 20356
Aug 13 19:38:35 dsmr supervisord[951]: 2020-08-13 19:38:35,855 INFO success: dsmr_backend entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
```

Om dit probleem te debuggen heb ik volgens de info op https://dsmr-reader.readthedocs.io/nl/v4/troubleshooting.html?highlight=debug het loglevel op debug gezet en post-deploy gedraaid. Echter nu zie ik nergens debug output (niet in /var/log/syslog, maar ook niet in /home/dsmr/dsmr-reader/logs). Een restart van supervisord biedt helaas ook geen uitkomst.

dennissiemensma, 2020-08-13:
Bedankt voor je melding. Syslog wordt namelijk niet direct gebruikt door de applicatie, alleen Supervisor-logfiles omdat daar alle processen onder draaien.

Heb je ook de logfiles bekeken die op die pagina van de docs staan? Ik kan dat niet expliciet uit je beschrijving halen.
[image]

Ik vermoed dat je de logfile van de `dsmr_backend` hier nodig hebt. Ook zonder debugging ingeschakeld, zou je daar een (fout)melding moeten zien met een indicatie waarom MQTT niet werkt.

mchlrv, 2020-08-13:
Thanks, daar stond inderdaad de debug info. Als ik kijk in de logs zie ik het volgende:

```
[2020-08-13 23:03:41,963] DEBUG    MQTT: Initializing MQTT client for "127.0.0.1:1883"
[2020-08-13 23:03:41,964] DEBUG    MQTT: Using insecure connection (no TLS)
[2020-08-13 23:03:41,964] ERROR    MQTT: Failed to connect to broker (127.0.0.1:1883), disabling MQTT: [Errno 111] Connection refused
```

Andere poorten werken ook niet. Klopt het dat DSMR zelf een MQTT broker is of dien ik die nog apart te installeren?

closed 2020-08-13 by mchlrv

dennissiemensma, 2020-08-13:
DSMR-reader is inderdaad alleen een MQTT-client. Fijn dat je het zo snel gevonden hebt.

mchlrv, 2020-08-13:
Thanks voor je snelle hulp!