dsmrreader/dsmr-reader #1080: Using webserver subpath with DSMR v4

swerveshot, 2020-08-15:
### Wat gebruik je?
* DSMR-reader version: `v4.1.1`
* Hardware: `Odroid U3`
* DSMR protocol: `v5`

### Omschrijving
De upgrade procedure van v3 naar v4 beschijft hoe de instellingen in dsmrreader/settings.py te verplaatsen naar een .env file. Echter in mijn settings.py stond deze config die nog niet door de Env settings wordt ondersteund (uit issue #146).

```
# Configure subpath for use with reverse proxy
STATIC_URL = '/dsmrreader/static/'
FORCE_SCRIPT_NAME = '/dsmrreader'
USE_X_FORWARDED_HOST = True
```

Deze regels zorgen er voor dat ik naast dsmrreader nog meer applicaties op mijn webserver kan hosten.

### Reproduceren issue
Doe een clean install van v4 en zorg dat de volgende regels in de file `/etc/nginx/sites-available/dsmr-webinterface` staan.

```
    location /dsmrreader/static {
        alias /var/www/dsmrreader/static;
    }

    location /dsmrreader/ {
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header Host $http_host;
        proxy_redirect off;
        proxy_pass http://dsmr-webinterface/;
```

### Mogelijke oplossingen
1. Maak mogelijk dat de Env settings de subpath opties bevatten.
2. Blijf deze tweak gebruiken in dsmrreader/settings.py.

Ik neig naar de tweede optie (easiest fix). Waarschijnlijk is deze setting maar relevant voor een klein deel van de gebruikers. En de docker users kunnen gebruik maken van tools als [Traefik](https://containo.us/traefik/). Maar het zou jammer zijn als door een wijziging deze mogelijkheid ineens niet meer zou werken.

Ik hoor graag je feedback @dennissiemensma.

dennissiemensma, 2020-08-15:
Bedankt voor je vraag! Ik denk dat je voorstel, om het wederom in de `settings.py` te zetten, inderdaad het beste is voor nu.

Het zijn namelijk vrij geavanceerde instellingen, dus voel je vrij ze door te voeren voor je eigen situatie. Als meerdere gebruikers er tzt behoefte aan hebben, kan ik het wel mappen vanuit een envvar.

closed 2020-08-15 by dennissiemensma

discofris, 2020-10-03:
Zou echt wel fijn zijn DSMR via reverse proxy te laten werken adhv envvar.
Ook heel handig ze mee te geven vanuit docker-compose zo ...

dennissiemensma, 2020-10-03:
Bedankt voor je aanvulling. Ik ga er ondersteuning voor maken in de volgende release, v4.5, via #1140.

dennissiemensma, 2020-10-03:
Vanaf `v4.5` zijn ze straks in te stellen met:

```
DJANGO_STATIC_URL=
DJANGO_FORCE_SCRIPT_NAME=
DJANGO_USE_X_FORWARDED_HOST=
```