dsmrreader/dsmr-reader #1081: Geen metingen meer na terugzetten backup files.

Schaduwval, 2020-08-17:
DSMR-reader is gemaakt om door jezelf te gebruiken en is daarom niet van buitenaf te debuggen. Daarom is het belangrijk dat je eerst deels zelf onderzoek doet en dat hieronder deelt (voor zover nodig). Dit helpt met het oplossen van jouw probleem. 

https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

Wat gebruik je?
- DSMR-reader versie:   4.1.1
- Type RaspberryPi or server:   RaspberryPi 3B
- Standaardinstallatie of Docker:   Standaard
- DSMR-protocol slimme meter v4/v5:   v5 

---
Beste, 

Ik heb na een frisse installatie de backups terug gezet. Echter nadat de backups waren terug gezet en de dsmr-reader opnieuw opgestart was komen er geen metingen meer binnen. Zoals het lijkt doet alles het maar ik krijg helaas geen metingen meer binnen op dsmr-reader.

Enig idee waaraan dit kan liggen? en hoe dit op te lossen is?

Alvast bedankt voor de medewerking

sudo supervisorctl status
dsmr_backend                     RUNNING   pid 560, uptime 2:01:07
dsmr_datalogger                  RUNNING   pid 3426, uptime 0:00:09
dsmr_webinterface                RUNNING   pid 558, uptime 2:01:07

[image]
[image]

dennissiemensma, 2020-08-17:
Ik zie dat de uptime van de datalogger vrij kort is:
```
 dsmr_datalogger RUNNING pid 3426, uptime 0:00:09
```

Klopt het dat die (telkens) crasht/herstart als je een paar keer opnieuw dat status-overzicht doet? 

Een goed startpunt zijn altijd de logfiles, in dat geval:
```
/var/log/supervisor/dsmr_datalogger.log
```

Schaduwval, 2020-08-17:
Dit is de output van de .log

`Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging`

dennissiemensma, 2020-08-17:
Je kunt het beste dan debug logging aanzetten om te zien wat de datalogger doet: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging-levels

Schaduwval, 2020-08-17:
Ik heb de debug logging aangezet, hoe kan ik deze logfile uitlezen? Mijn kennis van terminal is nog beperkt.

dennissiemensma, 2020-08-17:
Het is dezelfde logfile als eerder. Je kunt doorgaan met tail een logfile live meekijken:

```
sudo tail -f /var/log/supervisor/dsmr_datalogger.log
```

Met `CTRL+C` breek je het volgen af.

Schaduwval, 2020-08-17:
> Dit is de output van de .log
>
> ```
> Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
> Current logging [...]

Dit is wat ik nu ook zie

dennissiemensma, 2020-08-17:
Ah ja, ik was vergeten dat ik de extra debugging pas in v4.2 heb toegevoegd, die nog niet uit is. Je kunt dan wel even wisselen naar die toekomstige versie, ik verwacht dat je dan meer ziet in de log:

```
sudo su - dsmr
git fetch
git checkout -b wip/v4.2 origin/wip/v4.2
./deploy.sh
```

Kun je daarna nogmaals in de log kijken?

Schaduwval, 2020-08-17:
> Ah ja, ik was vergeten dat ik de extra debugging pas in v4.2 heb toegevoegd, die nog niet uit is. Je kunt dan wel even wisselen naar die toekomstige versie, ik verwacht dat je dan meer ziet in de [...]

Ik zie helaas nog steeds dezelfde output. Hoe wissel ik weer naar de huidige versie?

dennissiemensma, 2020-08-17:
Zou je eens deze willen tailen, wellicht staat het in een logfile van een subproces. Ik zie hier dat de originele logfile hier ook leeg blijft, maar deze werkt bij mij wel:

```
sudo tail -f /var/log/supervisor/dsmr_datalogger*.log
```

dennissiemensma, 2020-08-17:
Ter vergelijking, ik zie dan zoiets:

```
!D046
[2020-08-17 22:03:51,687] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 2.0s
[2020-08-17 22:03:53,940] DEBUG    [2020-08-17 22:03:53.940826] Read 0 Byte(s)
[2020-08-17 22:03:54,193] DEBUG    [2020-08-17 22:03:54.192981] Read 0 Byte(s)
[2020-08-17 22:03:54,445] DEBUG    [2020-08-17 22:03:54.445137] Read 0 Byte(s)
[2020-08-17 22:03:54,697] DEBUG    [2020-08-17 22:03:54.697236] Read 0 Byte(s)
[2020-08-17 22:03:54,949] DEBUG    [2020-08-17 22:03:54.949290] Read 0 Byte(s)
[2020-08-17 22:03:55,201] DEBUG    [2020-08-17 22:03:55.201452] Read 0 Byte(s)
[2020-08-17 22:03:55,453] DEBUG    [2020-08-17 22:03:55.453585] Read 0 Byte(s)
[2020-08-17 22:03:55,705] DEBUG    [2020-08-17 22:03:55.705442] Read 380 Byte(s)
[2020-08-17 22:03:55,957] DEBUG    [2020-08-17 22:03:55.957369] Read 441 Byte(s)
[2020-08-17 22:03:56,209] DEBUG    [2020-08-17 22:03:56.209335] Read 48 Byte(s)
[2020-08-17 22:03:56,216] DEBUG    Received telegram:
/XMX5LGBBFFB231117727

1-3:0.2.8(42)
0-0:1.0.0(200817220320S)
...
```

Schaduwval, 2020-08-17:
Ik krijg nu wel andere input zie hieronder:

`==> /var/log/supervisor/dsmr_datalogger-stderr---supervisor-xUcwYI.log <==
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
django.db.utils.IntegrityError: duplicate key value violates unique constraint "dsmr_datalogger_dsmrreading_pkey"
DETAIL:  Key (id)=(16601537) already exists.

[2020-08-17 22:09:04,330] INFO     dsmr_datalogger.management.commands.dsmr_datalogger: Exiting on next run...
[2020-08-17 22:09:04,331] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
[2020-08-17 22:09:09,338] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Exited
[2020-08-17 22:09:22,361] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Starting infinite command loop...
[2020-08-17 22:09:22,363] INFO     [2020-08-17 22:09:22.363038] Opening serial connection "/dev/ttyUSB0" using options: {'baudrate': 115200, 'bytesize': 8, 'parity': 'N', 'stopbits': 1, 'xonxoff': 1, 'rtscts': 0}
[2020-08-17 22:09:22,617] DEBUG    [2020-08-17 22:09:22.617368] Read 1 Byte(s)
[2020-08-17 22:09:22,869] DEBUG    [2020-08-17 22:09:22.869498] Read 0 Byte(s)
[2020-08-17 22:09:22,919] DEBUG    [2020-08-17 22:09:22.919501] Read 512 Byte(s)
[2020-08-17 22:09:23,170] DEBUG    [2020-08-17 22:09:23.170490] Read 149 Byte(s)
[2020-08-17 22:09:23,177] DEBUG    Received telegram:
/ISK5\2M550E-1012

1-3:0.2.8(50)
0-0:1.0.0(200817220920S)
0-0:96.1.1(4530303433303037313939333439303138)
1-0:1.8.1(001881.045*kWh)
1-0:1.8.2(001180.113*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.339*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00007)
0-0:96.7.9(00004)
1-0:99.97.0(2)(0-0:96.7.19)(180813133540S)(0000000368*s)(190606025827S)(0000004252*s)
1-0:32.32.0(00007)
1-0:32.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(227.6*V)
1-0:31.7.0(001*A)
1-0:21.7.0(00.338*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303339303031383332303639353138)
0-1:24.2.1(200817220500S)(00350.897*m3)
!4CE4
[2020-08-17 22:09:23,364] ERROR    dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
psycopg2.IntegrityError: duplicate key value violates unique constraint "dsmr_datalogger_dsmrreading_pkey"
DETAIL:  Key (id)=(16601538) already exists.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 90, in run_once
    self.run(data=self.data, **options)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 30, in run
    dsmr_datalogger.services.datalogger.telegram_to_reading(data=telegram)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services/datalogger.py", line 94, in telegram_to_reading
    return _map_telegram_to_model(parsed_telegram=parsed_telegram, data=data)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services/datalogger.py", line 148, in _map_telegram_to_model
    new_instance = DsmrReading.objects.create(**reading_kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 447, in create
    obj.save(force_insert=True, using=self.db)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 751, in save
    force_update=force_update, update_fields=update_fields)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 789, in save_base
    force_update, using, update_fields,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 892, in _save_table
    results = self._do_insert(cls._base_manager, using, fields, returning_fields, raw)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 932, in _do_insert
    using=using, raw=raw,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 1249, in _insert
    return query.get_compiler(using=using).execute_sql(returning_fields)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1395, in execute_sql
    cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
[... cut ...]
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
django.db.utils.IntegrityError: duplicate key value violates unique constraint "dsmr_datalogger_dsmrreading_pkey"
DETAIL:  Key (id)=(16601538) already exists.`

dennissiemensma, 2020-08-17:
Het lijkt erop dat er dubbele records in de dataloger staan. Je kunt proberen of dit het oplost:

```
sudo su - dsmr
./manage.py dsmr_sqlsequencereset
```

Je ziet geen output bij uitvoeren. Daarna opnieuw in de log kijken of het iets heeft opgelost.

Schaduwval, 2020-08-17:
Er lijken weer metingen door te komen via de webinterface. Echter het duurt +- 50 seconden voordat er resfresh is, dus zie het nu niet realtime. Dit lijkt mij niet goed? er hoort niet zoveel tijd tussen te zitten?

Merk verder dat alles heel traag gaat.

`==> /var/log/supervisor/dsmr_datalogger.log <==
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging

==> /var/log/supervisor/dsmr_datalogger-stderr---supervisor-QumZtj.log <==
0-0:96.13.0()
1-0:32.7.0(226.4*V)
1-0:31.7.0(001*A)
1-0:21.7.0(00.245*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303339303031383332303639353138)
0-1:24.2.1(200817223000S)(00350.903*m3)
!B4F2
[2020-08-17 22:32:04,337] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
[2020-08-17 22:32:09,343] DEBUG    [2020-08-17 22:32:09.343773] Read 512 Byte(s)
[2020-08-17 22:32:09,344] DEBUG    [2020-08-17 22:32:09.344528] Read 512 Byte(s)
[2020-08-17 22:32:09,351] DEBUG    Received telegram:
/ISK5\2M550E-1012

1-3:0.2.8(50)
0-0:1.0.0(200817223156S)
0-0:96.1.1(4530303433303037313939333439303138)
1-0:1.8.1(001881.045*kWh)
1-0:1.8.2(001180.219*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.246*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00007)
0-0:96.7.9(00004)
1-0:99.97.0(2)(0-0:96.7.19)(180813133540S)(0000000368*s)(190606025827S)(0000004252*s)
1-0:32.32.0(00007)
1-0:32.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(226.2*V)
1-0:31.7.0(001*A)
1-0:21.7.0(00.245*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303339303031383332303639353138)
0-1:24.2.1(200817223000S)(00350.903*m3)
!7C3D
[2020-08-17 22:32:09,484] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
[2020-08-17 22:32:14,490] DEBUG    [2020-08-17 22:32:14.490547] Read 512 Byte(s)
[2020-08-17 22:32:14,491] DEBUG    [2020-08-17 22:32:14.491294] Read 512 Byte(s)
[2020-08-17 22:32:14,492] DEBUG    [2020-08-17 22:32:14.492019] Read 512 Byte(s)
[2020-08-17 22:32:14,499] DEBUG    Received telegram:
/ISK5\2M550E-1012

1-3:0.2.8(50)
0-0:1.0.0(200817223158S)
0-0:96.1.1(4530303433303037313939333439303138)
1-0:1.8.1(001881.045*kWh)
1-0:1.8.2(001180.219*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.244*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00007)
0-0:96.7.9(00004)
1-0:99.97.0(2)(0-0:96.7.19)(180813133540S)(0000000368*s)(190606025827S)(0000004252*s)
1-0:32.32.0(00007)
1-0:32.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(226.1*V)
1-0:31.7.0(001*A)
1-0:21.7.0(00.243*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303339303031383332303639353138)
0-1:24.2.1(200817223000S)(00350.903*m3)
!539F
[2020-08-17 22:32:14,609] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
[2020-08-17 22:32:19,621] ERROR    dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 90, in run_once
    self.run(data=self.data, **options)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 27, in run
    telegram = next(self.telegram_generator)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 45, in read_telegram
    len(buffer)
RuntimeError: It took too long to detect a telegram. Check connection params. Bytes currently in buffer: 0`

dennissiemensma, 2020-08-17:
Ik denk dat je nu eerst weer kunt proberen om naar v4.1.1 te gaan.

```
sudo su - dsmr
git checkout v4
./deploy.sh
```

Maakt dat het weer wat sneller?

Schaduwval, 2020-08-17:
> Ik denk dat je nu eerst weer kunt proberen om naar v4.1.1 te gaan.
> 
> ```
> sudo su - dsmr
> git checkout v4
> ./deploy.sh
> ```
> 
> Maakt dat het weer wat sneller?

Nee helaas. Verder vroeg ik mij af hoelang het duurt voordat je een refresh van de gemeten meting, voorheen was dit toch direct? Ook lijkt de weer grafriek verdwenen te zien na het terug zetten van de backup.

dennissiemensma, 2020-08-17:
De upgrade naar v4.1 zet de datalogger sleep sowieso op 5 seconden om de datagrootte te verminderen. Dat kan al iets verklaren, maar niet 50 seconden. 
Het kan wel zijn dat de tijd van je slimme meter (of Pi) niet synchroon loopt, maar dat is puur een verschil qua tijden en niet de snelheid van verversen.

Welke grafiek bedoel je overigens? Ik moet echt heel specifiek weten wat je mist of anders is.

dennissiemensma, 2020-08-17:
Wat je nog kunt doen is kijken wat de grootte van je database is en of er indexes missen.

### Grootte:
```
sudo su - postgres
psql dsmrreader

SELECT nspname || '.' || relname AS "relation",
    pg_size_pretty(pg_total_relation_size(C.oid)) AS "total_size"
  FROM pg_class C
  LEFT JOIN pg_namespace N ON (N.oid = C.relnamespace)
  WHERE nspname NOT IN ('pg_catalog', 'information_schema')
    AND C.relkind <> 'i'
    AND nspname !~ '^pg_toast'
  ORDER BY pg_total_relation_size(C.oid) DESC
  LIMIT 20;
```

----

Indexes:
```
sudo su - postgres
psql dsmrreader

\dS dsmr_datalogger_dsmrreading;
```

Deze moeten er zijn:
```
...

Indexes:
    "dsmr_datalogger_dsmrreading_pkey" PRIMARY KEY, btree (id)
    "dsmr_datalogger_dsmrreading_9d090a40" btree (processed)
    "dsmr_datalogger_dsmrreading_timestamp_1f7020d1_uniq" btree ("timestamp")

```

----

```
\dS dsmr_consumption_electricityconsumption;
```

Deze moeten er zijn:
```
...

Indexes:
    "dsmr_consumption_electricityconsumption_pkey" PRIMARY KEY, btree (id)
    "dsmr_consumption_electricityconsumption_read_at_key" UNIQUE CONSTRAINT, btree (read_at)
    "dsmr_consumption_electrici_phase_power_current_l1_3d9fd70d" btree (phase_power_current_l1)
    "dsmr_consumption_electrici_phase_voltage_l1_72e3c8f8" btree (phase_voltage_l1)
    "dsmr_consumption_electrici_phase_voltage_l2_917726c7" btree (phase_voltage_l2)
    "dsmr_consumption_electrici_phase_voltage_l3_1a9d7c37" btree (phase_voltage_l3)
    "dsmr_consumption_electricityc_currently_delivered_5bf9be2d_uniq" btree (currently_delivered)
    "dsmr_consumption_electricityco_currently_returned_5f91fed7_uniq" btree (currently_returned)
    "dsmr_consumption_electricityconsumption_2241f8ee" btree (phase_currently_delivered_l2)
    "dsmr_consumption_electricityconsumption_5714c500" btree (phase_currently_delivered_l1)
    "dsmr_consumption_electricityconsumption_e90842a6" btree (phase_currently_delivered_l3)
```

Schaduwval, 2020-08-17:
> De upgrade naar v4.1 zet de datalogger sleep sowieso op 5 seconden om de datagrootte te verminderen. Dat kan al iets verklaren, maar niet 50 seconden.
> Het kan wel zijn dat de tijd van je slimme [...]

Ik heb hier bij voorgaande versies geen last van gehad. Hoe kan ik zorgen dat de tijd weer  synchroon loopt? 

Met de weergrafiek bedoel ik grafiek onderaan de pagina waar je ziet wat de temperatuur is op de ingestelde locatie.

Schaduwval, 2020-08-17:
> Wat je nog kunt doen is kijken wat de grootte van je database is en of er indexes missen.
>
> ### Grootte:
> ```
> sudo su - postgres
> psql dsmrreader
>
>
> SELECT nspname || '.' || relname AS "relation", [...]

Deze indexes staan erbij mij ook.

Schaduwval, 2020-08-18:
Kan het er nog mee te maken hebben dat de backups van v3 zijn ipv v4?

dennissiemensma, 2020-08-18:
Bedankt voor het checken. 

- Hoe groot is je database nu? Mogelijk heeft dat ook nog invloed.

- Het kan een mogelijke oorzaak zijn dat de tijd niet synchroon loopt, maar daar zou je systeem zelf al voor moeten zorgen. Als voorbeeld: Bij mij zijn mijn metingen volgens het Dashboard altijd 30 tot 40 seconden uit, wegens tijdsverschil met de meter. Maar dat is puur strikt, want in de praktijk is die meting vaak maar enkele seconden oud. Dus het hangt er vanaf of je bedoelt dat de laatste meting volgens het Dashboard altijd vertraagd is, maar wel regelmatig, of dat je bedoelt dat het telkens 50 seconden duurt voordat er een nieuwe meting verwerkt is.

- Het terugzetten van oude v3 backups gaat altijd goed. Bij deploy worden eventuele achterstallige migraties alsnog uitgevoerd. Ook als die van v3 of ouder is.

dennissiemensma, 2020-08-18:
Als ik namelijk naar je eerdere v4.2 logs kijk, zie ik telkens 5 seconden vertraging, exact de tijd van de sleep (wat ik dus verwacht):

```
...

!B4F2
[2020-08-17 22:32:04,337] DEBUG dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
[2020-08-17 22:32:09,343] DEBUG [2020-08-17 22:32:09.343773] Read 512 Byte(s)

...

[2020-08-17 22:32:09,484] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
[2020-08-17 22:32:14,490] DEBUG    [2020-08-17 22:32:14.490547] Read 512 Byte(s)
[2020-08-17 22:32:14,491] DEBUG    [2020-08-17 22:32:14.491294] Read 512 Byte(s)
[2020-08-17 22:32:14,492] DEBUG    [2020-08-17 22:32:14.492019] Read 512 Byte(s)
[2020-08-17 22:32:14,499] DEBUG    Received telegram:

...

```

dennissiemensma, 2020-08-18:
Wat betreft de weergrafiek, check even of de temperatuurmetingen wel aan staan in de admin.

dennissiemensma, 2020-08-18:
En tot slot, wat geeft de Status-pagina nu aan als overzicht?

Schaduwval, 2020-08-18:
Hoi Dennis, 

Allereerst bedankt voor je hulp met deze issue. 

Ik kreeg gister allemaal foutmeldingen als bijvoorbeeld naar de pagina configuratie wilde gaan. Er is wat niet helemaal juist gaat. 

Dus ik ben van een nieuwe installatie erop te zetten zonder dat ik de backups terug zet. Dan kan ik controleren hoe alles dan werkt. En zou het aan de P1 kabel kunnen liggen of is dat uitgesloten? 

Wat is jou advies? 

Wat mij nu wel bedenk is het volgende: 

Toen ik de backups terug aan het zetten was via de uitleg kreeg ik wel een paar keer de melding dat er informatie dubbel was. Weet niet of je dit iets zegt?

dennissiemensma, 2020-08-18:
Bedankt voor je aanvulling. Dat is vrij cruciale informatie, want dat kan betekenen dat de huidige installatie conflicten of dubbele data heeft waar het niet mee om kan gaan.

Je zult bij een herstel altijd een lege database moeten aanmaken, dan de backup terugzetten en dan pas de applicatie weer moeten aanzetten (of installeren). Ik zal dat duidelijker vermelden in de docs.

----

Mijn advies is om (mits je nog een goede backup hebt) de installatie te stoppen, de database te verwijderen, dan opnieuw aan te maken, backup terugzetten en dan alles weer aanzetten.

```
sudo supervisorctl stop all

sudo -u postgres dropdb dsmrreader
sudo -u postgres createdb -O dsmrreader dsmrreader

<BACKUP Terugzetten>

sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/dsmr-reader/manage.py migrate
sudo supervisorctl start all

sudo su - dsmr
./deploy.sh
```

Schaduwval, 2020-08-19:
> Bedankt voor je aanvulling. Dat is vrij cruciale informatie, want dat kan betekenen dat de huidige installatie conflicten of dubbele data heeft waar het niet mee om kan gaan.
>
> Je zult bij een [...]

Ik ben de bakups aan het terug zetten en loop tegen het volgende aan: 

> 

```
pi@raspberrypi:~ $ zcat /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-backup-Monday.sql.gz | sudo -u postgres psql dsmrreader
SET
SET
SET
SET
SET
 set_config 
------------
 
(1 row)

SET
SET
SET
SET
SET
SET
ERROR:  relation "auth_group" already exists
ALTER TABLE
ERROR:  relation "auth_group_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "auth_group_permissions" already exists
ALTER TABLE
ERROR:  relation "auth_group_permissions_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "auth_permission" already exists
ALTER TABLE
ERROR:  relation "auth_permission_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "auth_user" already exists
ALTER TABLE
ERROR:  relation "auth_user_groups" already exists
ALTER TABLE
ERROR:  relation "auth_user_groups_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "auth_user_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "auth_user_user_permissions" already exists
ALTER TABLE
ERROR:  relation "auth_user_user_permissions_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "django_admin_log" already exists
ALTER TABLE
ERROR:  relation "django_admin_log_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "django_content_type" already exists
ALTER TABLE
ERROR:  relation "django_content_type_id_seq" already exists
ALTER TABLE
ALTER SEQUENCE
ERROR:  relation "django_migrations" already exists
ALTER TABLE
ERROR:  relation "django_migrations_id_seq" already exists
ALTER TABLE
[... 683 lines cut ...]
ERROR:  relation "dsmr_backend_scheduledprocess_module_55cb894b_like" already exists
ERROR:  relation "dsmr_backend_scheduledprocess_planned_234388f1" already exists
ERROR:  relation "dsmr_consumption_electrici_currently_delivered_0f28e53a" already exists
ERROR:  relation "dsmr_consumption_electrici_currently_returned_b7a910c8" already exists
ERROR:  relation "dsmr_consumption_electrici_phase_currently_delivered__0315221e" already exists
ERROR:  relation "dsmr_consumption_electrici_phase_currently_delivered__21a515a7" already exists
ERROR:  relation "dsmr_consumption_electrici_phase_currently_delivered__9998ce55" already exists
ERROR:  relation "dsmr_consumption_electrici_phase_power_current_l1_3d9fd70d" already exists
ERROR:  relation "dsmr_consumption_electrici_phase_voltage_l1_72e3c8f8" already exists
ERROR:  relation "dsmr_consumption_electrici_phase_voltage_l2_917726c7" already exists
ERROR:  relation "dsmr_consumption_electrici_phase_voltage_l3_1a9d7c37" already exists
ERROR:  relation "dsmr_datalogger_dsmrreading_processed_add8a852" already exists
ERROR:  relation "dsmr_datalogger_dsmrreading_timestamp_1f7020d1" already exists
ERROR:  relation "dsmr_stats_daystatistics_total_cost_c898d987" already exists
ERROR:  relation "dsmr_stats_note_day_83d05681" already exists
ERROR:  constraint "auth_group_permissio_permission_id_84c5c92e_fk_auth_perm" for relation "auth_group_permissions" already exists
ERROR:  constraint "auth_group_permissions_group_id_b120cbf9_fk_auth_group_id" for relation "auth_group_permissions" already exists
ERROR:  constraint "auth_permission_content_type_id_2f476e4b_fk_django_co" for relation "auth_permission" already exists
ERROR:  constraint "auth_user_groups_group_id_97559544_fk_auth_group_id" for relation "auth_user_groups" already exists
ERROR:  constraint "auth_user_groups_user_id_6a12ed8b_fk_auth_user_id" for relation "auth_user_groups" already exists
ERROR:  constraint "auth_user_user_permi_permission_id_1fbb5f2c_fk_auth_perm" for relation "auth_user_user_permissions" already exists
ERROR:  constraint "auth_user_user_permissions_user_id_a95ead1b_fk_auth_user_id" for relation "auth_user_user_permissions" already exists
ERROR:  constraint "django_admin_log_content_type_id_c4bce8eb_fk_django_co" for relation "django_admin_log" already exists
ERROR:  constraint "django_admin_log_user_id_c564eba6_fk_auth_user_id" for relation "django_admin_log" already exists
pi@raspberrypi:~ $

dennissiemensma, 2020-08-19:
Ik mis de output van je voorgaande commands. Het is mij nu onduidelijk welke stappen je vooraf hebt uitgevoerd.

Heb je de andere commands ook uitgevoerd? En zonder fouten?

Als dit op een lege database gebeurt, is er mogelijk iets met de backup zelf, maar dat zou ik raar vinden. Ik neem aan dat dit de backup is van anderhalve week terug op v3?

Schaduwval, 2020-08-19:
> Ik mis de output van je voorgaande commands. Het is mij nu onduidelijk welke stappen je vooraf hebt uitgevoerd.
>
> Heb je de andere commands ook uitgevoerd? En zonder fouten?
>
> Als dit op een lege [...]

Ja ik heb de onderstaande commando uitgevoerd zonder foutmeldingen (neem aan dat je deze bedoelde?)

sudo supervisorctl stop all

sudo -u postgres dropdb dsmrreader
sudo -u postgres createdb -O dsmrreader dsmrreader

> Als dit op een lege database gebeurt, is er mogelijk iets met de backup zelf, maar dat zou ik raar vinden. Ik neem aan dat dit de backup is van anderhalve week terug op v3

Ja dit is de backup van 11-08-2020 op v3.

dennissiemensma, 2020-08-19:
Als je het niet erg vind je data te delen, dan mag je de backup eventueel ergens neerzetten en linken in een e-mail naar github @ dennissiemensma punt nl. Dan kijk ik even of ik hem hier wel zonder fouten geimporteerd krijg. 

En anders kan ik eventueel proberen om alle dag- en uurstatistieken eruit te halen zodat die je in een schone installatie kan gebruiken (dus zonder eerdere metingen etc).

dennissiemensma, 2020-08-19:
Ik kan het helaas niet reproduceren. Je Monday backup lijkt gewoon goed. 

Je kunt proberen om ook een andere DB-naam er naast aan te maken en daarop te importeren als test.  Ik heb de data wel uitgepakt overigens, maar dat zou verder niet uit moeten maken.

```
postgres@pc:~$ createdb -O dsmrreader test_import
postgres@pc:~$ psql -f ext-backup.sql test_import
SET
SET
SET
SET
SET
 set_config 
------------
 
(1 row)

SET
SET
SET
SET
SET
SET
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
[... 614 lines cut ...]
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
```

----

Data lijkt groot. Het importeren van de uitgepakte data duurde zelfs op mn (i5) PC voor mn gevoel al 5+ minuten.
```
                    relation                     | total_size 
-------------------------------------------------+------------
 public.dsmr_datalogger_dsmrreading              | 2990 MB
 public.dsmr_consumption_electricityconsumption  | 134 MB
 public.dsmr_consumption_gasconsumption          | 8152 kB
 public.dsmr_stats_hourstatistics                | 808 kB
 public.dsmr_weather_temperaturereading          | 704 kB
 public.dsmr_stats_daystatistics                 | 104 kB
```

----

Als je nog geen automatische opschoning aan hebt staan en/of een lage datalogger sleep, kan ik je aanraden dat straks sowieso aan te zetten / te verhogen naar 5 seconden.

dennissiemensma, 2020-08-19:
Voor de rest zal ik een export voor je maken zonder metingen en elektra:

```
test_import=# delete from dsmr_datalogger_dsmrreading ;
DELETE 16672159

test_import=# delete from dsmr_consumption_gasconsumption ;
DELETE 78783

test_import=# delete from dsmr_consumption_electricityconsumption ;
DELETE 403422
```

Zie mail.

Die kun je dan in `/var/lib/postgresql/bare_export.sql` zetten en dan:

```
sudo su - postgres

createdb -O dsmrreader test_import
psql -f bare_export.sql test_import
```

Als dat werkt kun je hetzelfde doen met:
```
sudo supervisorctl stop all
sudo su - postgres

dropdb dsmrreader
createdb -O dsmrreader dsmrreader

psql -f bare_export.sql dsmrreader

logout
sudo su - dsmr
./deploy.sh
sudo supervisorctl start all
```

In theorie dan.

Schaduwval, 2020-08-20:
> * Het kan een mogelijke oorzaak zijn dat de tijd niet synchroon loopt, maar daar zou je systeem zelf al voor moeten zorgen. Als voorbeeld: Bij mij zijn mijn metingen volgens het Dashboard altijd 30 [...]

> Ik kan het helaas niet reproduceren. Je Monday backup lijkt gewoon goed.
>
> Je kunt proberen om ook een andere DB-naam er naast aan te maken en daarop te importeren als test. Ik heb de data wel [...]

De automatische opschoning had ik nog niet aan staan. Welke optie raad je aan om de data te verminderen? Hoe groot zou de data 'normaal' moeten zijn? De dataloggersleep stond al op 5 seconden.

Schaduwval, 2020-08-20:
> Voor de rest zal ik een export voor je maken zonder metingen en elektra:
>
> ```
> test_import=# delete from dsmr_datalogger_dsmrreading ;
> DELETE 16672159
>
> test_import=# delete from [...]

Ik heb het bovenstaande uitgevoerd en het importen van bare_export.sql dsmrreader ging zonder problemen.

Schaduwval, 2020-08-20:
> Ik kan het helaas niet reproduceren. Je Monday backup lijkt gewoon goed.

Heb je een idee waarom het terugzetten van de backups niet lukt, of is het gewoon domme pech?

Schaduwval, 2020-08-20:
> * Het kan een mogelijke oorzaak zijn dat de tijd niet synchroon loopt, maar daar zou je systeem zelf al voor moeten zorgen. Als voorbeeld: Bij mij zijn mijn metingen volgens het Dashboard altijd 30 [...]

Om hier een duidelijker antwoord op te geven: 

Ik bedoel dat de laatste meting volgens het dashboard nu 22 seconden geleden, daarna begint het dashboard weer van 0 te tellen. Maar als ik in de /var/log/supervisor/dsmr_datalogger.log kijk zie ik dat het om de 5 seconden er een meting is zoal aangeven in de instellingen van de dataloggerconfiguratie.

Het viel mij op omdat vroeger nooit was?

Dus het hangt er vanaf of je bedoelt dat de laatste meting volgens het Dashboard altijd vertraagd is, maar wel regelmatig,

dennissiemensma, 2020-08-20:
Het is lastig om te zeggen waarom dingen anders zijn of trager lijken. Het is niet expliciet veranderd voor zover ik weet. Hoe dan ook denk ik dat de hoeveelheid metingen niet meehelpt.

Een goede balans tussen sleep en opschoning hangt af van je eigen wensen. Als je geen waarde hecht aan het oneindig bewaren van alle metingen, en puur de uur/dagtotalen wilt, dan kun je de opschoning op 1 week of 1 maand zetten.

En waarom de volledige database terugzetten niet werkt blijft ook gissen, dus tenzij je waarde hecht aan alle voorgaande metingen (los van de uur/dagtotalen) dan denk ik dat je wellicht het beste kunt proberen verder te gaan met de export die ik je gestuurd hebt. Als het goed is werkt die dan wel en ook snel.

dennissiemensma, 2020-08-24:
@Schaduwval is het je inmiddels gelukt met de afgeslankte export?

Schaduwval, 2020-08-28:
> @Schaduwval is het je inmiddels gelukt met de afgeslankte export?

Ik was afwezig, vandaar mijn late reactie.

Ja het terugzetten van de afgeslankte export is gelukt, DSMR-reader werkt weer goed en verder geen problemen meer ondervonden na het terug zetten van de export.

@dennissiemensma Fijn dat je deze data hebt kunnen veiligstellen. En nogmaals bedankt voor je hulp!

Zie hieronder:

`pi@raspberrypi:~ $ sudo supervisorctl status`
`dsmr_backend  RUNNING   pid 1438, uptime 8 days, 19:56:53`
`dsmr_datalogger RUNNING   pid 1568, uptime 8 days, 19:51:53`
`dsmr_webinterface RUNNING   pid 930, uptime 8 days, 20:19:04`

dennissiemensma, 2020-08-29:
Fijn om te horen, bedankt voor de reactie!

closed 2020-08-29 by dennissiemensma