dsmrreader/dsmr-reader #1083: Virutalenv maken zonder --system-site-packages

dennissiemensma, 2020-08-17:
- Deels terugdraaien van #1013.
- `--system-site-packages` niet meer gebruiken.
- `python3-psycopg2` niet meer gebruiken maar proberen te vervangen door `psycopg2-binary`.

Wieter, 2020-11-01:
Voor FreeBSD is Systemd geen alternatief. 

Mag ik vragen welk probleem er hier opgelost moet worden? 
Misschien dat ik iets gemist heb, maar `--system-site-packages` was niet nodig.
Mijn installatie op FreeBSD ging ongeveer als volgt (helaas heb ik het niet exact bijgehouden):

(psycopg was al geinstalleerd voor iets anders, wordt volgens mij niet gebruikt voor dsmrreader): `pkg install py37-psycopg2`
virtualenv aanmaken: ` python3 -m venv .venv`
vanuit bash activeren: `. .venv/bin/activate`
en vervolgens vanuit daar verder met `pip install` alle requirements installeren. 

note: ik maak geen gebruik van gunicorn, maar ipv daarvan [uwsgi](https://uwsgi-docs.readthedocs.io/en/latest/tutorials/Django_and_nginx.html). 
Dit kan ook op linux gebruikt worden, en start (in geval van FBSD) al als service. Applicaties zijn instelbaar dmv. een .ini file (vergelijkbaar als met supervisor), met ook de optie om [mules](https://uwsgi-docs.readthedocs.io/en/latest/Mules.html?highlight=mules) in te stellen; worker processes. 
Aan gezien dsmrreader helaas nog geen support hiervoor heeft(lees: ik heb het nog niet out-of-the-box werkend gekregen), draai ik inderdaad nog apart een backend processor dmv `py37-supervisor`. Maar volgens mij is daar maar een kleine code-implementation voor nodig, aangezien vanuit `manage.py` de callback `dsmr_backend` al bestaat. Ik zal zelf binnenkort hier ook weer eens naar kijken.

dennissiemensma, 2020-11-02:
De voorgeschiedenis staat in #1013. Het gebruiken van systemd is geen must, meer ter onderzoek of dat eventueel Supervisor zou kunnen uitfaseren. Als dat tzt dan niets toe lijkt te voegen laat ik dat voor wat het is. 

Wat betreft je eigen situatie, je bent helemaal vrij om het te draaien op een manier die voor jou werkt. Uiteindelijk heb je alleen een WSGI-server nodig en iets wat de datalogger en backend processen draait. Standaard kan dat via Gunicorn en Supervisor, maar probeer gerust iets heel anders.

dennissiemensma, 2020-11-08:
Update: Van het weekend ook nog gekeken naar `pipenv`, want in de eerste instantie prima leek te werken en zelf de virtualenv beheert. 
Helaas kreeg ik het niet lekker werkend voor tests en ik zag dat er in de Python-community ook wat verdeeldheid heerste over deze (vrij jonge) tool. Mogelijk dat ik nog naar een andere tool kijk. Ik lees her en der wat over `poetry`, dus wellicht moet ik daar nog naar kijken.

dennissiemensma, 2020-11-14:
Issue is niet meer relevant. Een volgende major release gaat zeer waarschijnlijk gebruikmaken van `poetry` die zelf de virtualenvs beheert.

closed 2020-11-14 by dennissiemensma