dsmrreader/dsmr-reader #1084: Feature request: piekvermogen per dag/maand

VBP8501, 2020-08-17:
De VREG (Belgisch opperhoofd van energie distributie) heeft beslist dat er vanaf 2022 netvergoeding betaalt moet worden op de gemiddelde maandvermogenpiek van de laatste 12 maanden.

De maandpiek is het maximaal kwartiervermogen.

En het kwartiervermogen is het gemiddelde vermogen gedurende 1 kwartier. 

Wat voor mij nog onduidelijk is, is of het hier gaat over het totale vermogen (over de 3 fasen) of per fase.

Het zou interessant zijn om de kwartiervermogens of maandpieken ergens in grafiek of tabelvorm te kunnen raadplegen. Vandaar deze feature request.

dennissiemensma, 2020-08-18:
Bedankt voor je verzoek. Om goed te beoordelen welke gegevens exact nodig zijn, zul je denk ik wat meer erover moeten aanleveren. 
Wellicht dat er een grafiek voor nodig is of wellicht dat ik er simpelweg een query voor jullie voor kan maken waarmee je makkelijk het gemiddelde voor een jaar kunt zien. Daarnaast kan het dus zijn dat jullie in Belgie, voor een nauwkeurige nacalculatie daarvan, in DSMR-reader de (automatische) opschoning van metingen moeten uitschakelen. Dat heeft natuurlijk weer gevolgen voor de opslag en performance op den duur.

Dus wellicht kun je wat meer details hierover aanleveren. Anderzijds, als er landelijk nog geen details zijn, kun je ook overwegen om dit even te laten voor wat het is en er over een jaar of wat op terug te komen. En ondertussen kijken of je alle metingen vanaf 2021 wilt bewaren.

VBP8501, 2020-08-18:
Dit gaat in België vrij bepalend worden voor de prijs van de netvergoeding te drukken. Maar daarnaast is het wellicht ook interessant voor mensen die overwegen om een thuisbatterij te plaatsen en/of volledig offgrid te gaan. Maximaal vermogen is heel relevant in die omstandigheden. Uiteraard kunnen we dat ook op 1 of andere manier uit de huidige database trekken met een query, maar als dat in realtime bekeken kan worden (en via domotica/smart devices/iot op ingespeeld kan worden om pieken te verkleinen), maakt dat de dsmr-reader zoveel waardevoller.

Wat ik in gedacht had was:
1) Onder "dashboard":  extra regel voor max kwartierpiek zowel bij de dagsamenvatting als de maandsamenvatting
2) Onder "archive": extra grafiek met het piek kwartiervermogen en in de tabel een extra regel voor max kwartierpiek (indien dag/maand geselecteerd) of gemiddeld maandelijks max kwartierpiek (indien jaar geselecteerd)
3) Onder "compare", idem van "archive" maar dan enkel ivm de tabelvorm

En dan uiteraard soortgelijke aanpassingen bij de API en de MQTT.

Ik heb zelf al eens gekeken naar mijn eigen maandpieken. Een query die daarvoor werkt is deze:

```
with cal as
(
        SELECT ts AS t_begin, ts+ '15 minutes'::interval AS t_end
        FROM generate_series('2020-07-01 0:00'::timestamp, '2020-08-01 0:00', '15 minutes'::interval) ts
)
,avg_15min_gridpower as 
(
	select 
		cal.t_begin, 
		cal.t_end, 
		greatest(avg(dsmr.electricity_currently_delivered), avg(electricity_currently_returned)) as avg_15min_gridpower
	from cal
	left join public.dsmr_datalogger_dsmrreading dsmr on dsmr.timestamp >= cal.t_begin and dsmr.timestamp < cal.t_end
	group by cal.t_begin, cal.t_end
	order by cal.t_begin
)
,max_daily_gridpower as 
(
	select 
		date(t_begin) as date, 
		max(avg_15min_gridpower) as max_daily_gridpower
	from avg_15min_gridpower
	group by date(t_begin)
)
,max_monthly_gridpower as 
(
	select 
		date_trunc('month',date) as month, 
		max(max_daily_gridpower) as max_monthly_gridpower
	from max_daily_gridpower
	group by date_trunc('month',date)
),avg_yearly_gridpower as 
(
	select 
		date_trunc('year',month) as year, 
		avg(max_monthly_gridpower) as avg_yearly_gridpower
	from max_monthly_gridpower
	group by date_trunc('year',month)
)
select * from max_monthly_gridpower
```

Maar dit on demand berekenen is natuurlijk wel tijdrovend. Het zou beter zijn om de kwartierwaarden ergens weg te schrijven. Als de dsmr_consumption_electricityconsumption records gemiddelden bevatten van het opgenomen of verbruikt vermogen dan kan de query ook herschreven worden om deze tabel te gebruiken ipv dsmr_datalogger_dsmrreading. 

Bonus:
- Onder "live charts": extra grafiek met de gemiddelde kwartierwaarden van de afgelopen 24u
- Onder "statistics": extra grafiek met hoogste kwartierpieken per tijdslot. Gelijkaardig aan de grafiek "AVERAGE ELECTRICITY CONSUMED EACH HOUR (IN %) WITHIN THE PAST 4 WEEKS"

dennissiemensma, 2020-08-19:
Dank voor de uitgebreide toelichting. 

Intepreteer ik het goed dat voor juli 2020 bij mij het hoogste **gemiddelde** verbruik binnen alle kwartieren `2,69 kW` is geweest?

```
         month          | max_monthly_gridpower  
------------------------+------------------------
 2020-07-01 00:00:00+01 |     2.6930000000000000
```

En als ik voor een jaar kijk:

```
         month          | max_monthly_gridpower  
------------------------+------------------------
 2019-08-01 00:00:00+01 |     2.7340000000000000
 2019-09-01 00:00:00+01 |     2.3980000000000000
 2019-10-01 00:00:00+01 |     3.4080000000000000
 2019-11-01 00:00:00+00 |     3.1360000000000000
 2019-12-01 00:00:00+00 |     2.3810000000000000
 2020-01-01 00:00:00+00 |     3.1350000000000000
 2020-02-01 00:00:00+00 |     4.1980000000000000
 2020-03-01 00:00:00+00 |     3.2080000000000000
 2020-04-01 00:00:00+01 |     5.2400000000000000
 2020-05-01 00:00:00+01 |     4.0560000000000000
 2020-06-01 00:00:00+01 |     2.9030000000000000
 2020-07-01 00:00:00+01 |     2.6930000000000000
```

Dan zou ik (wanneer Belgisch) dus een vergoeding moeten betalen over het gemiddelde van bovenstaande? *Los van of het over een kalenderjaar of contractjaar gaat.*

En voor mijn beeld, niet nodig voor de feature, maar politiek gezien: Wat probeert men hiermee te belonen of af te straffen?

VBP8501, 2020-08-19:
> Dank voor de uitgebreide toelichting.
>
> Intepreteer ik het goed dat voor juli 2020 bij mij het hoogste **gemiddelde** verbruik binnen alle kwartieren `2,69 kW` is geweest?
>
> ``` [...]

Dat is de correcte interpretatie ja. Je hoogste kwartiervermogen tijdens juli was 2,69 kW

> En als ik voor een jaar kijk:
>
> ```
>          month          | max_monthly_gridpower
> ------------------------+------------------------
>  2019-08-01 00:00:00+01 |     2.7340000000000000
>  2019-09-01 [...]

Dat lijkt mij een normaal verloop. Bij mij is het praktisch altijd 4kW - 500W. 4kW als het maximaal vermogen van de zonnepanelen en 500W wat zo ongeveer mijn standby verbruik is tijdens de dag. De VREG spreekt van een gemiddelde van 2,5kW voor een huishouden in België. Dat zal wel een gemiddeld gezin zonder zonnepanelen, een elektrische wagen of een groot inductievuur zijn. 

> Dan zou ik (wanneer Belgisch) dus een vergoeding moeten betalen over het gemiddelde van bovenstaande? _Los van of het over een kalenderjaar of contractjaar gaat._

Dat klopt. Op dit moment betaal je in België een prijs per verbruikseenheid kWh, zowel voor de energie zelf (in zekere zin logisch) als voor de distributie van die energie (niet zo logisch). De VREG zegt nu dat de distributiekost geen verband houdt met de hoeveelheid energie (kWh) dat er doorgaat maar wel met het piekvermogen (de kW). Vanaf 2022 wordt de prijs die een consument zal betalen voor de distributie, berekend op basis van het maximaal vermogen en niet langer op basis van de kWh. 

Eigenaars van zonnepanelen moeten nu al in België een extra bijdrage betalen om hun panelen aan het net te mogen aansluiten. Dat is ook een functie van het AC vermogen van de omvormer. Ongeveer 100€/kW. In mijn geval betaal ik dus 400€ per jaar voor mijn omvormer. 

Meestal hebben mensen met zonnepanelen een netto nulverbruik, 0 kWh. Dus betalen ze geen energiekost noch een distributiekost (op uitzondering van het prosumententarief). Met de aanpassing van de VREG zal dat dus veranderen en zijn eigenaars van zonnepanelen dubbel gezien..

> En voor mijn beeld, niet nodig voor de feature, maar politiek gezien: Wat probeert men hiermee te belonen of af te straffen?

Ha jah... 
FYI: https://www.vreg.be/nl/toekomst-nettarieven-capaciteitstarief
Ik heb het hierboven al wat aangehaald. De officiële reden is dat ze mensen willen sensibiliseren dat het maximaal vermogen zo laag mogelijk moet zijn. Dus geen auto gaan opladen terwijl je frieten bakt. Of een thuisbatterij interessant maken. 
Ik heb natuurlijk mijn eigen mening en veronderstellingen bij de nieuwe regeling. Het maakt bv niet uit hoevaak je een piek veroorzaakt (continue 5kW lijkt mij ergers dan 1x 5kW) of wanneer die juist optreedt (wat maakt het uit als je een piekafname doet wanneer er overschot is aan energie?). Volgens mij zien ze hun winsten dalen omdat er zoveel mensen zonnepanelen hebben en ze moeten hun inkomsten ergens vandaan halen... Soit, dat is mijn gedacht ... 

Feit is wel dat niemand kaas heeft gegeten van kW's. De meesten begrijpen wel het principe van kWh en dat ze dit moeten betalen. Ik voorspel dat er veel verontwaardigde mensen gaan zijn als ze willen proberen hun energiefactuur te doen dalen. "Komaan, nu heb ik heel het jaar goed opgelet dat ik geen frieten bak en tegelijk de vaat doe en toch moet ik meer betalen dan vorig jaar". Terwijl het eigenlijk hun zonnepanelen zijn die meer kW's op het net zetten dan eender welk ander toestel dat ze in huis hebben kan verbruiken. 

Door de kW's zichtbaar te maken (op de live grafiek van dsmr-reader is dit al zo, dus kudos!) worden die mensen geholpen.

dennissiemensma, 2020-08-20:
Bedankt voor alle toelichting. Ik zal dit later nog eens rustig doornemen en kijken wat er mogelijk is. Ik vermoed dat er wel eea te regelen valt en ook wat toevoegt aan DSMR-reader, al weet ik nog niet exact in welke vorm (grafieken, tabellen, exports).

VBP8501, 2020-08-26:
Correctie van wat ik hierboven heb vermeld. Het capaciteitstarief is enkel op afname, dus niet op injectie. De query wordt dan:

```
with cal as
(
        SELECT ts AS t_begin, ts+ '15 minutes'::interval AS t_end
        FROM generate_series('2020-07-01 0:00'::timestamp, '2020-08-01 0:00', '15 minutes'::interval) ts
)
,avg_15min_gridpower as 
(
	select 
		cal.t_begin, 
		cal.t_end, 
		avg(dsmr.electricity_currently_delivered) as avg_15min_gridpower
	from cal
	left join public.dsmr_datalogger_dsmrreading dsmr on dsmr.timestamp >= cal.t_begin and dsmr.timestamp < cal.t_end
	group by cal.t_begin, cal.t_end
	order by cal.t_begin
)
,max_daily_gridpower as 
(
	select 
		date(t_begin) as date, 
		max(avg_15min_gridpower) as max_daily_gridpower
	from avg_15min_gridpower
	group by date(t_begin)
)
,max_monthly_gridpower as 
(
	select 
		date_trunc('month',date) as month, 
		max(max_daily_gridpower) as max_monthly_gridpower
	from max_daily_gridpower
	group by date_trunc('month',date)
),avg_yearly_gridpower as 
(
	select 
		date_trunc('year',month) as year, 
		avg(max_monthly_gridpower) as avg_yearly_gridpower
	from max_monthly_gridpower
	group by date_trunc('year',month)
)
select * from max_monthly_gridpower
```

dennissiemensma, 2020-08-26:
Dank voor de update! Ik kijk er meer in detail naar wanneer ik de feature inhoudelijk integreer.

brainkiller, 2020-08-29:
> Correctie van wat ik hierboven heb vermeld. Het capaciteitstarief is enkel op afname, dus niet op injectie.

Mag ik vragen waar je deze info gevonden hebt ? Ik kan geen verschil vinden tussen afnamen en injectie tijdens al mijn google searches.
Ik zou ook voorstellen om het configureerbaar te maken, ik zie de VREG nog hun beslissing herzien in de toekomst.

VBP8501, 2020-08-29:
> Mag ik vragen waar je deze info gevonden hebt ? Ik kan geen verschil vinden tussen afnamen en injectie tijdens al mijn google searches.
> Ik zou ook voorstellen om het configureerbaar te maken, ik zie [...]

Inderdaad, in de publieke communicatie van de vreg zag ik ook niets staan over of het nu ging over afname of injectie. Maar aangezien er het prosumententarief eigenlijk al een capaciteitstarief is op injectie was ik wat achterdochtig en heb ik de vraag aan de vreg gesteld. Dit waren mijn vragen:

> 1) ik het 3x220V aansluiting. Geldt het piekvermogen voor 1 fase of voor de drie fases samen?
> 
> 2) wordt de kwartier piek bepaalt op enkel verbruik of ook op injectie? 
> 
> 3) Indien het piekvermogen bepaald wordt op beide, als er tijdens een kwartier zowel een hoog piekverbruik als een hoge piekinjectie is, niet tegelijkertijd, wordt dan de maximalepiek genomen of neutraliseren beide zich in dat kwartier?
> 
> 4) zijn het vaste kwartieren over de dag? Bv 0u00 tot 0u15 enzovoort of zijn het vlottende kwartieren ifv pieken?

En dit was het antwoord:

> Het capaciteitstarief zal aangerekend worden op basis van de ‘gemiddelde maandpiek (kW)’. Deze maandpiek staat los van het aansluitingsvermogen. Het maakt dus niet uit of u een eenfasige of driefasige aansluiting heeft. 
> 
>  
> 
> Deze gemiddelde maandpiek zal berekend worden als het gemiddelde van uw 12 laatste ‘maandpieken’. De maandpiek is het hoogste kwartiervermogen (ofwel ‘piekvermogen’) dat u in een maand hebt gebruikt. Het wordt geregistreerd door de digitale meter. Het maakt hierbij niet uit hoe vaak (tijdens 1 kwartier of tijdens meerdere kwartieren) u dit piekvermogen heeft gebruikt. Als er nog geen meetgegevens van 12 maanden beschikbaar zijn, bv. na de vervanging van een klassieke meter door een digitale meter, zal de gemiddelde maandpiek berekend worden op basis van alle beschikbare maandpieken.
> 
>  
> 
> Wanneer een maandpiek kleiner is dan 2,5 kW, zal voor die maand een minimumwaarde van 2,5 kW worden verondersteld. Hierdoor zal élke netgebruiker een minimale jaarlijkse bijdrage betalen in de netkosten. Deze bijdrage zal gelijk zijn aan 2,5 kW vermenigvuldigd met het tarief voor de gemiddelde maandpiek.
> 
> De andere kosten opgenomen in de distributienettarieven, met name de kosten voor openbaredienstverplichtingen, de toeslagen en de overige transmissiekosten, zullen blijven  aangerekend worden op basis van de afgenomen kWh. 
> 
>  
> 
> Het capaciteitstarief is een afnametarief: het wordt aangerekend o.b.v. uw maandelijkse afnamepieken; injectiepieken worden niet in rekening genomen.
> 
>  
> 
> Het zijn vaste kwartieren. Dus tussen 18:00 en 18:15, tussen 0:00 en 0:15,…
> 
>  
> 
> De kwartierwaarden (kWh) die de digitale meter registreert zijn per ‘vast’ kwartier.
> 
> Ook voor de volledige marktwerking (allocatie van energiehoeveelheden, evenwichtsverantwoordelijkheid naar Elia toe, e.d.) wordt met deze vaste kwartieren gewerkt.

Op mijn eerste vraag is niet echt een antwoord gekomen. Maar ik ga ervan uit dat de som van de drie fasen van toepassing is.

gigatexel, 2021-01-26:
Op basis van de Vlaamse digitale meter, uitlezing via DSMR in Home Assistant, is dit de query die je de piek van de afgelopen maand geeft.
Bij nieuwe maand dient de oude steeds opgeslagen te worden omdat deze impact heeft op de verrekening van die specifieke maand.
Opgelet! De kwartierpiek geldt niet voor injectie!

`    queries:
      - name: Maandpiek
        column: 'Maandpiek'
        unit_of_measurement: "kW"
        query: "
        select Round((a.verschil+b.verschil)*4,2) as Maandpiek, a.time as time FROM 
        (SELECT
        FROM_UNIXTIME((UNIX_TIMESTAMP(created) DIV 900 * 900)+3600, '%d-%c-%Y %H:%i:%s') AS time,
        entity_id AS metric,
        (state - lag(state) over (order by created)) as verschil
        FROM states
        WHERE
        MONTH(created) = MONTH(CURRENT_DATE()) AND YEAR(created) = YEAR(CURRENT_DATE()) AND entity_id = 'sensor.energy_consumption_tarif_1' AND state != 'unknown'
        GROUP BY 1 ORDER BY created DESC) as a
        inner JOIN
        (SELECT
        FROM_UNIXTIME((UNIX_TIMESTAMP(created) DIV 900 * 900)+3600, '%d-%c-%Y %H:%i:%s') AS time,
        entity_id AS metric,
        (state - lag(state) over (order by created)) as verschil
        FROM states
        WHERE
        MONTH(created) = MONTH(CURRENT_DATE()) AND YEAR(created) = YEAR(CURRENT_DATE()) AND entity_id = 'sensor.energy_consumption_tarif_2' AND state != 'unknown'
        GROUP BY 1 ORDER BY created DESC) as b
        on a.time = b.time
        ORDER BY Maandpiek DESC 
        LIMIT 1"`

dennissiemensma, 2022-02-04:
@VBP8501 @gigatexel zijn hier nog ontwikkelingen in geweest bij de invoering dit jaar, tov wat jullie tzt geprojecteerd hadden?

Deze feature is helaas afgelopen jaar flink vertraagd door andere issues in DSMR-reader, maar het staat nog wel op de planning. Mits het enigszins te behappen is uiteraard.

gigatexel, 2022-02-05:
> @VBP8501 @gigatexel zijn hier nog ontwikkelingen in geweest bij de invoering dit jaar, tov wat jullie tzt geprojecteerd hadden?
>
> Deze feature is helaas afgelopen jaar flink vertraagd door andere [...]

Als alles volgens plan gaat - wat in gepolitiseerde dossiers geen zekerheid is - gaat het capaciteitstarief van start op 01/07/2022

Ik heb ondertussen bovenstaande query nog wat geoptimaliseerd en ook specifieke indexen gemaakt in MariaDB omdat het soms langer dan 10 seconden duurder om te draaien.

VBP8501, 2022-02-05:
Inderdaad.. De huidige hoge elektriciteitsprijzen spreken niet in het voordeel van het capaciteitstarief. Ik acht de kans op aanpassingen wel vrij klein, de terugdraaiende teller is uiteindelijk ook afgeschaft, ongeacht wat de politiek ervan dacht. Er zijn op dit moment in België enkele studies die aantonen dat het capaciteitstarief aanleiding geeft tot een hogere factuur bij kleinverbruikers (wat logisch is) en een lagere bij grootverbruikers. En dat komt natuurlijk niet zo goed over.

Ik heb de query zelf ook verder geoptimaliseerd door periodiek het kwartiervermogen weg te schrijven. In dsmr zou je dat kunnen opnemen in de dagsamenvatting. Een analyse doen op de dag zelf (welk kwartier exact de piek plaatsvond) is wel te doen qua snelheid met een query op de minuutgemiddelden. Maar voor de meeste use cases te dekken zullen dag, maand en jaar resultaten wel voldoende zijn

gigatexel, 2022-02-05:
Kan je jouw query ook delen?

VBP8501, 2022-02-05:
> Kan je jouw query ook delen?

Jazeker, wel pas deze avond

VBP8501, 2022-02-05:
> Kan je jouw query ook delen?

Ik heb uiteindelijk twee views gemaakt die ik via grafana benut:

```
create view own.kwartiervermogens as (
with cal as (
	SELECT ts AS t_begin, ts+ '15 minutes'::interval AS t_end
	FROM generate_series('2019-11-01 0:00'::timestamp, now()-interval '15 minutes', '15 minutes'::interval) ts
)

	select 
		cal.t_begin, 
		cal.t_end, 
		avg(dsmr.currently_delivered) as avg_15min_gridpower
	from cal
	left join public.dsmr_consumption_electricityconsumption dsmr on dsmr.read_at >= cal.t_begin and dsmr.read_at < cal.t_end
	group by cal.t_begin, cal.t_end
	order by cal.t_begin
)
```

```
create view own.kwartiervermogen_monthly as (
with avg_15min_gridpower as (select * from own.kwartiervermogens)
,max_daily_gridpower as 
(
	select 
		date(t_begin) as date, 
		max(avg_15min_gridpower) as max_daily_gridpower
	from avg_15min_gridpower
	group by date(t_begin)
)
,max_monthly_gridpower as 
(
	select 
		date_trunc('month',date) as month, 
		max(max_daily_gridpower) as max_monthly_gridpower
	from max_daily_gridpower
	group by date_trunc('month',date)
),avg_yearly_gridpower as 
(
	select 
		date_trunc('year',month) as year, 
		avg(max_monthly_gridpower) as avg_yearly_gridpower
	from max_monthly_gridpower
	group by date_trunc('year',month)
)
select * from max_monthly_gridpower)
```

("own" is een eigen schema naast public)

De berekening waar ik naar verwijs is eigenlijk een download van mijn data in pvoutput.org. Omdat dsmrreader de gegevens van mijn pv omvormers niet benut, moet ik die nog op een andere manier samenbrengen om bv analyses te kunnen doen naar een thuisbatterij. Pvoutput heeft maar datapunten iedere 5minuten dus de berekeningen voor kwartierpieken (gelijkaardig aan bovenstaande views) gaan heel vlot.

dennissiemensma, 2022-02-06:
Bedankt voor jullie aanvulling! Betekent dit dat een lagere frequentie van telegrammen in DSMR-reader ook een grote invloed heeft op de nauwkeurigheid van bovenstaande? Dus in het geval dat iemand de meter slechts elke 5 a 10 seconden uitleest, ipv elke seconde.
In principe zal teruglevering en grootverbruik (wasmachine, oven, etc) niet om de 5 a 10 seconden wisselen van piekbelasting, maar ze wisselen wel en dat lijkt me nog wel een uitdaging dan.

Aan de andere kant zullen de meeste gebruikers wel een DSMR-5 meter hebben die elke seconden uitleest, dus misschien is het wat minder een issue als ik de standaard datalogger-sleep wat verlaag. Inmiddels schoont DSMR-reader later wel de metingen op en als ik ervoor zorg dat ik voor die tijd de kwartiervermogens heb berekend en apart opsla (wat sowieso al het plan was), dan is dat misschien afdoende.

VBP8501, 2022-02-06:
Het zijn altijd vaste kwartieren. Dus altijd van 0 tot 15min, 15 tot 30 enzovoort. Dus qua nauwkeurigheid is het voldoende om de meter 1 keer per kwartier uit te lezen. Om dan te komen tot kwartiervermogen doe je delta verbruik in kWh maal 4 om kW gemiddeld over dat kwartier te hebben. (Op dat vlak kan ik mijn queries ook nog wat verbeteren. Straks is proberen als de kinderen in bed liggen)

VBP8501, 2022-02-06:
Ik denk dat het voor dsmr reader voldoende is om kwartiervermogens uit te rekenen als het kwartier voorbij is op basis van het verbruik (kWh) en dus niet door het uitmiddelen van de vermogens aan de frequentie van de meter.

Ik stuur mij verbruikers (voor zover mogelijk..) aan via minuutwaarden. En de ledstrip die het instantaansluiting verbruik weergeeft via de mqtt berichten van dsmr

dennissiemensma, 2022-02-06:
Ah prima. Als het ook via kWh kan, dan maakt dat het een stuk makkelijker berekenen! Ik kon namelijk al niet helemaal hierboven je gebruik van `dsmr_consumption_electricityconsumption.currently_delivered` plaatsen, gezien dat alleen het Wattage is.

VBP8501, 2022-02-06:
Hierbij de betere versie op basis van dsmr_consumption_electricityconsumption.delivered_

```
CREATE OR REPLACE VIEW own.kwartiervermogens
AS WITH cal AS (
         SELECT ts.ts AS t_begin,
            ts.ts + '00:15:00'::interval AS t_end
           FROM generate_series('2019-11-01 00:00:00'::timestamp without time zone::timestamp with time zone, now() - '00:15:00'::interval, '00:15:00'::interval) ts(ts)
        )
 SELECT cal.t_begin,
    cal.t_end,
    ((dsmr_to.delivered_1+dsmr_to.delivered_2) - (dsmr_from.delivered_1+dsmr_from.delivered_2))*4 AS avg_15min_gridpower
   FROM cal
     LEFT JOIN dsmr_consumption_electricityconsumption dsmr_from ON dsmr_from.read_at = cal.t_begin 
     LEFT JOIN dsmr_consumption_electricityconsumption dsmr_to ON dsmr_to.read_at = cal.t_end 
  ORDER BY cal.t_begin;
```

Werkt dubbel zo snel als de voorgaande :-)

dennissiemensma, 2022-02-07:
Bedankt voor je aanvulling. Hoe moet ik trouwens die x 4 zien? Ik ben op dat gebied niet heel bekend met elektra en de berekeningen.

Overigens zal ik het wel met het ORM moeten doen, dus ik kan geen harde query overnemen. Maar als het een kwestie is van de eerste en laatste standen van elk vast kwartier nemen en daarmee rekenen, moet dat goed komen.

VBP8501, 2022-02-07:
Aah ik kan je misschien niet helpen met de programmatie zelf maar wat elektriciteit uitleggen, dat kan ik wel :). Het komt er eigenlijk om neer dat er 4 kwartieren zijn in een uur en de eenheid van energie kWh is, waarin dus 'uur' zit. Maar ik leg het even van 0 uit voor de duidelijkheid. 

Hetgeen we zoeken is het gemiddelde vermogen, uitgedrukt in Watt gedurende ieder kwartier. Op zich zouden we dus zoveel mogelijk veel meetpunten van "het ogenblikkelijk vermogen" kunnen nemen gedurende dat kwartier en dan delen door het aantal om een benadering van het gemiddelde vermogen te bekomen. Hoe meer punten je meet, hoe beter het resultaat. Maar het zal altijd een benadering zijn, iets wat je zelf ook al aanhaalde.

Als je een constant vermogen zou afnemen van 10 Watt, bijvoorbeeld die stomme digicorders in standby van je digitale TV, dan verbruik je 10 Wattuur op 1 uur tijd. Zet je een boiler op van 2 kW gedurende een uur dan heb je 2 kWuur ofwel 2kWh verbruikt.

Omgekeerd, op je meter staat je energieverbruik, uitgedrukt in kWh. Als dat gedurende 1 uur, 1 kWh eenheid zou toenemen, dan heb je gemiddeld gezien djus 1 kW verbruikt. Ongeacht of dat vermogen nu een half uur 2 kW was en een half uur 0 kW, het eindresultaat is een gemiddelde van 1 kW over dat uur.

Hetzelfde kun je dus doen voor dat kwartier. In 1 uur gaan 4 kwartieren. Dus als je energieverbruik gedurende een kwartier is gestegen met 1 kWh en je zou hetzelfde vermogen laten draaien over het volledige uur, dan zou je 4 kWh verbruikt hebben over dat uur. Oftwel een gemiddelde vermogen van 4 kW. 

Dus in een kwartier, 1 kWh verbruiken wil zeggen 1 x 4 = 4 kW gemiddeld vermogen.

Om het even met de analogie van snelheid (~ vermogen kW) en afstand (~ energieverbruik kWh) uit te leggen. Wanneer je op een kwartier 30 km hebt afgelegd, hoe snel reed je dan gemiddeld? Juist ja maal 4 dus 120 km/u.
De vermogens en energiën kunnen ook in andere eenheden worden uitgedrukt wat het rekenen eigenlijk gemakkelijker maakt. Zou is Watt = Joule / seconde en kWh = kW * uur oftwel equalent aan [een factor x] Joule. Als je dat doorhebt wordt het rekenen met vermogen en energie eigenlijk veel logischer. Maarja, een joule is nu eenmaal 1, 3600.000ste van een kWh :)

dennissiemensma, 2022-02-07:
Bedankt voor je uitgebreide toelichting! In dat geval snap ik waar mijn verwarring vandaan kwam, namelijk dat het verbruik wordt gemeten in een kwartier, maar dat het vermogen weer over een uur berekend wordt.

dennissiemensma, 2022-02-07:
Ja exact, je latere edit weergeeft dat ook al. Dank!

dennissiemensma, 2022-02-07:
Iets heel anders, DSMR-reader v5 is nu ook beschikbaar, dus voel je vrij om (een keer) te upgraden. Neem daar wel wat tijd voor als je de handmatige installatie hebt gedaan, het is niet 5 minuten werk **en** incompatible met v4 (of je moet heel snel zijn).

dennissiemensma, 2022-02-07:
Dit issue zal ik proberen op te nemen in v5.1. Wellicht dat ik dan eerst het berekenen en opslaan daarvan maak (en dan de data via de admininteface ontsluit), want ik ben er nog niet helemaal uit hoe dit qua grafieken of tabellen samen te voegen is met de bestaande data/weergaves in DSMR-reader

VBP8501, 2022-02-07:
> Iets heel anders, DSMR-reader v5 is nu ook beschikbaar, dus voel je vrij om (een keer) te upgraden. Neem daar wel wat tijd voor als je de handmatige installatie hebt gedaan, het is niet 5 minuten werk **en** incompatible met v4 (of je moet heel snel zijn).

Thx, nog even wachten tot de docker image gebuild is

qbushome, 2022-02-14:
Zou het mogelijk zijn om de huidige piek (van het lopende kwartier) ook live ter beschikking te kunnen stellen?
Zou kan je in Home Assistant, Domoticz, ... een beslissing nemen. Is mijn huidige piek te hoog, start dan NIET toestel A. of, Is mijn huidige piek te hoog, schakel toestel B UIT.

In HA gebruik ik nu het volgende (en het werkt prima), maar wanneer DSMR dit zou kunnen aanleveren is dit beschikbaar voor iedereen, zonder extra code, ongeacht welk domotica-systeem je gebruikt.

`sensor:
  - platform: template
    sensors:
      fluvius_verbruik_energy_kwartier_gemiddelde:
        value_template: "{{ states('sensor.fluvius_verbruik_energy_kwartier') | float(default=0) * 4 }}"
        unit_of_measurement: "kWh"
        device_class: energy
        icon_template: mdi:calculator
        friendly_name: "Verbruik huidig kwartier"

utility_meter:
  fluvius_verbruik_energy_kwartier:
    source: sensor.fluvius_meterstand_gebruikt                # source: DSMR (uom is in kWh already)
    cycle: quarter-hourly`

Je zou dan in DSMR gelijk elke kwartierwaarde op 14.59s van elk kwartier kunnen opslagen.

verstrepeng, 2022-02-20:
Komt deze feature in een toekomstige release ?

dennissiemensma, 2022-02-20:
https://github.com/dsmrreader/dsmr-reader/issues/1084#issuecomment-1031911570

dennissiemensma, 2022-02-21:
Nog even wat context gezocht, o.a.: https://www.engie.be/nl/blog/allerlei/capaciteitstarief-Vlaanderen/

> Tot eind juni 2022 hangen de distributienetkosten uitsluitend af van het aantal kWh dat de gebruiker effectief afneemt van het net. Vanaf juli 2022 telt het verbruik nog maar voor 20% mee in het distributietarief.

> En de resterende 80% dan? Die hangen af van de verbruikspieken die je veroorzaakt. Als je bijvoorbeeld tegelijk je strijkijzer, [airco](https://www.engie.be/nl/blog/verwarming/airco-keuze-voor-en-nadelen-verbruik-kostprijs)­ en [wasmachine](https://www.engie.be/nl/blog/bespaartips/tips-wasmachine-minder-elektriciteit-water-verbruiken) inschakelt en je elektrische auto oplaadt, dan piekt je verbruik. Hoe meer verbruikspieken je in een maand veroorzaakt en hoe meer je verbruikt op piekmomenten (’s ochtends tussen 6 en 8 uur en ’s avonds tussen 18 en 20 uur), hoe hoger de distributienetkosten die je moet betalen.

> De verbruikspiek wordt gemeten over 15 minuten. Het gemiddelde over die periode bepaalt de piek. Dus als je een kom soep 1 minuut in de microgolfoven met 1000 watt opwarmt, bedraagt de piek niet 1000 watt, maar 1/15e van 1000 watt, dus 66 watt.

> Ben je bang dat één hoge piek je elektriciteitsfactuur de hoogte zal doen inschieten? Geen paniek! Voor de berekening van je capaciteitstarief telt immers alleen het gemiddelde van 12 maandelijkse pieken die in de loop van een jaar worden gemeten. Per maand is er dus maar één piek (de hoogste, gemeten over 15 minuten) die in aanmerking komt. Eén enkele heel hoge piek telt dus maar mee voor 1/12e …

> Goed om te weten: door de invoering van het capaciteitstarief komt er een einde aan het huidige onderscheid tussen dag- en nachttarief. Toch voor wat de distributiekosten betreft. Je energieleverancier mag je nog altijd een verschillend tarief aanrekenen voor de stroom die je ’s nachts en overdag gebruikt, maar vanaf 2022 valt dit onderscheid weg voor het distributienettarief.

----

Als ik dat zo lees lijkt me dat best pittig. Even 1x in de maand niet opletten en dan ligt dat meteen vast als 1/12e. Pfoe. Al snap ik hun overweging ook wel, qua pieken op het net, het klinkt wel erg strikt.

VBP8501, 2022-02-21:
Tsja het is maar hoe je het bekijkt. Als die pieken allemaal tesamen vallen zal het wel stevig zijn voor een distributienet. Maar een piek die ergens 'snachts voorkomt is totaal niet belangrijk. 

Maarja, begin maar is.. Als ik hier eten maak en daarbij 2 delen van mijn elektrisch kookvuur nodig en de oven dan zit ik al rap aan 4kW. Tel daar nog eens een TV bij om de kinderen bezig te houden, al de rest van het standby vermogen, enzovoort.. Dat je een elektrische auto en warmtepomp op dat moment gestuurd afzet is nog wel te doen. Elektische keukenboilerkes ook nog. Maar toch, onder die 4kW lijkt mij toch moeilijk. En 't is niet dat ik de ene maand wel kook en de andere maand niet...

En ja zoals je zegt, als je 1x per maand niet oplet dan hang je eraan. Aan de andere kant, je moet gedurende die maand niet beter proberen doen dan dat je al hebt gedaan.. Ik heb een led strip onder de keukenkasten tegen de plint die aanspringt als het kwartiergemiddelde dreigt de maandpiek te overschrijden. En dan rood als het de foute kant op gaat en vica versa voor groen. Maarja, ik reset de maandpiek standaard op 3.5kW want anders zijn we heel nerveus aan het koken :D

dennissiemensma, 2022-02-28:
Ik bump deze naar v5.2 gezien de korte deadline die Dropbox mij geeft in #1210

fverpoest, 2022-04-24:
> Ik heb een led strip onder de keukenkasten tegen de plint die aanspringt als het kwartiergemiddelde dreigt de maandpiek te overschrijden. En dan rood als het de foute kant op gaat en vica versa voor [...]

@VBP8501 Zie je het zitten om me even te contacteren hoe je dit precies heb opgebouwd? fverpoest@outlook.com

dennissiemensma, 2022-05-17:
Het duurde even wegens beschikbare tijd en vooral complexiteit, vooral om te zorgen dat het werkt voor meerdere situaties. Maar heb inmiddels een eerste opzet staan: https://github.com/dsmrreader/dsmr-reader/commit/7c4b7154e61eca349a5bcbe01f1f9a28bc485f88#diff-79b655a9839c8d926e6778808d4f0ffe7a31f9d6038143b972023d3dd9f5c56dR43

Het zal initieel alleen de data/berekeningen zijn, beschikbaar via adminportaal, omdat het makkelijker is om achteraf het alsnog ergens in te plotten of te onsluiten via MQTT of een API of wat dan ook. Belangrijkste is ook dat het vooral alvast in werking wordt gezet.

Ik moet er nog wel tests voor maken, maar ik vermoed dat dit mechanisme relatief simpel en goed te testen is, omdat het in de basis relatief weinig nodig heeft.

FredericMa, 2022-05-18:
@dennissiemensma Ik was benieuwd en heb snel eens naar de code gekeken.
Het lijkt me dat op deze twee lijnen de som moet worden gemaakt van tarief 1 en tarief 2 om het kwartierverbruik te berekenen?
https://github.com/dsmrreader/dsmr-reader/blob/7c4b7154e61eca349a5bcbe01f1f9a28bc485f88/dsmr_consumption/services.py#L110-L111

dennissiemensma, 2022-05-18:
@FredericMa bedankt voor de scherpte! Ik heb het alvast aangepast. Bespaart me mogelijk wat uitzoekwerk zodra ik met de tests ga beginnen, had het zelf niet zo 1-2-3 gespot. 234edcec3fca96e92577c15a92afc8638db9cba0

dennissiemensma, 2022-05-19:
Eerste tests zitten er nu ook in: https://github.com/dsmrreader/dsmr-reader/compare/wip/v5.2.0...feature/1084/kwh-peak#diff-f9298e9ca6b20429ad55b08e37f87c23aff403903f48fc2fca67c3feb3eaa006

Moet nog wel wat edge-cases, vooral qua data-gaten toevoegen.

dennissiemensma, 2022-05-19:
Qua visuele opzet is het nog vrij simpel, maar goed genoeg om eerst te kijken of het uberhaupt goed werkt. Plotten enzo kan later relatief makkelijk, maar deze release moet wel het liefste voor 1 juni klaar.

[image]

----

[image]

----

Het filteren op een datum-range + daarna/daarbij kunnen sorteren zorgt er iig voor dat gebruikers makkelijk een dag of maand kunnen checken, wanneer de hoogste/laagste was en wanneer dat was.

[image]

dennissiemensma, 2022-05-19:
@VBP8501 zie de bovenste screenshot van https://github.com/dsmrreader/dsmr-reader/issues/1084#issuecomment-1132129035

**Wat moet de berekening zijn als de gemeten periode niet exact 15 minuten is?** Het komt niet op een seconde aan, maar ik kan me voorstellen dat in sommige situaties gebruikers bijvoorbeeld 10 seconden na start/voor einde de eerste/laatste telegrammen krijgen (is reeel met niet-v5-meters of met datalogger delay/sleep).

Dus een ideale gemeten periode van 15 minuten is 900 seconden. Maar als die periode bijvoorbeeld 800 seconden is, moet ik dan alsnog gewoon x4 doen voor het uur en de onnauwkeurigheid voor lief nemen?
Of moet ik de berekening relativeren op het uur? Dus het kwartiergemiddelde delen door die (bijvoorbeeld) 800 seconden, x 3600 voor het uur?

Het kan overigens ook gewoon een kwestie van afwachten zijn of het uberhaupt een probleem is. Doordat ik de lengte van de gemeten periode log/bereken, is het opzich ook weer duidelijk voor de gebruiker als daar een verschil in zit.

dennissiemensma, 2022-05-19:
Ik heb de weergave nog iets aangepast. Verder zit de huidige opzet alvast inde v5.2 branch, zodat ik hem zelf ook al kan draaien.

Het mechanisme voor terugwerkende kracht werkt goed, want tot aan het punt van verwijderde data (bij mij staat retentie op een week) is goed te zien vanaf welk punt er weer genoeg metingen zijn:

[image]

dennissiemensma, 2022-05-19:
Verder heb ik nu ook mn recent verbruik bekeken en ik heb alvast mn mobiele airco aangezet voor het slapen, dus dat is ook goed te zien.

# Laatste X uur
Tot aan 21:45 een constant verbruik van 100 a 300 Watt, daarna schiet het omhoog en in de 3 a 4 kwartieren daarna is het goed te zien en zit het rond de 800 a 900 watt en dat komt het in gemiddelde ook goed naar voren.

[image]

## Bijbehorende grafiek

[image]

----

# Hoogste kwartiergemiddelde deze maand
Hoogste pieken vooral rond etentijd (ik kook inductie) maar soms ook de close-in boiler vermoed ik overdag.

[image]

dennissiemensma, 2022-05-19:
Tot slot is goed te zien dat ik nog een v4 meter heb. Ik heb telkens meetperiodes van 14:40 lang, door de 10 seconde interval aan telegrammen.

dennissiemensma, 2022-05-23:
Follow-up via #1635. Eerst maar eens een tijdje draaien om te zien of het goed werkt. Verder ook geen idee hoeveel gebruikers dit daadwerkelijk nodig hebben, gezien ik verwacht dat het gros zich in NL bevindt.

dennissiemensma, 2022-05-23:
Als alles meezit dan morgenavond v5.2 releasen.

closed 2022-05-23 by dennissiemensma

VBP8501, 2022-05-23:
> @VBP8501 zie de bovenste screenshot van [#1084 (comment)](https://github.com/dsmrreader/dsmr-reader/issues/1084#issuecomment-1132129035)
> 
> **Wat moet de berekening zijn als de gemeten periode niet exact 15 minuten is?** Het komt niet op een seconde aan, maar ik kan me voorstellen dat in sommige situaties gebruikers bijvoorbeeld 10 seconden na start/voor einde de eerste/laatste telegrammen krijgen (is reeel met niet-v5-meters of met datalogger delay/sleep).
> 
> Dus een ideale gemeten periode van 15 minuten is 900 seconden. Maar als die periode bijvoorbeeld 800 seconden is, moet ik dan alsnog gewoon x4 doen voor het uur en de onnauwkeurigheid voor lief nemen? Of moet ik de berekening relativeren op het uur? Dus het kwartiergemiddelde delen door die (bijvoorbeeld) 800 seconden, x 3600 voor het uur?
> 
> Het kan overigens ook gewoon een kwestie van afwachten zijn of het uberhaupt een probleem is. Doordat ik de lengte van de gemeten periode log/bereken, is het opzich ook weer duidelijk voor de gebruiker als daar een verschil in zit.

idealiter ga je inderdaad de effectieve seconden gebruiken om de vermeningvuldigingsfactor te bepalen. Als het dus exact 15 min of 900s is dan doe je maal 4 (want 3600/900=4). Bij een afwijkende waarde moet je inderdaad 3600/echte_seconden doen. Dus in jouw voorbeeld 3600/800=4,5

VBP8501, 2022-05-23:
> Follow-up via #1635. Eerst maar eens een tijdje draaien om te zien of het goed werkt. Verder ook geen idee hoeveel gebruikers dit daadwerkelijk nodig hebben, gezien ik verwacht dat het gros zich in NL bevindt.

Alvast 2 gebruikers bevinden zich in België :)
En aan de opmerkingen van anderen in deze thread te zien, zijn het er precies wel meer ^^

verstrepeng, 2022-05-24:
Hier nog een Belg ;-)
Ik gebruik de HA add-on, dus wachten tot deze geupdate is naar 5.2 ?

dennissiemensma, 2022-05-24:
Follow-up dus via #1635. De versie die in v5.2 zit is een begin en ik verwacht echt nog wel wat nodige aanpassingen, maar dan wordt er "iets" berekend. 

En bij toekomstige wijzigingen is het vrij makkelijk om het te laten herberekenen door DSMR-reader (zolang de metingen er zijn), gezien deze data over piekvermogens naast de bestaande data draait en dus verder ook niets anders raakt. Het is meer verrijking van bestaande data.

Een ander punt is bijvoorbeeld de rotatie van deze data. Opzich valt de grootte wel mee, want bij een gemiddelde maand van 30 dagen maal 96 kwartieren in een dag, zijn < 3000 records per maand. Op lange termijn kost dat wel wat opslag, maar het gaat (gelukkig) uiteraard lang niet zo snel als elke seconde een meting opslaan.

qbushome, 2022-05-27:
Zeker nog 1 Belg extra hier.
Dennis, bedankt dat je dit wil uitwerken/toevoegen.

Bereken je ook de piek van het huidige kwartier?
Waarom is dat nuttig? Wel, je hebt een grote verbruiker die je kan in/uitschakelen (dus geen wasmachine maar bvb een elektrische boiler).
Stel: die verbruik 3kw en je wil dat je pieken nooit boven de 3kwh uitkomen. Wanneer je (zoals in mijn geval in Home Assistant) dan zou weten dat de huidige kwartierpiek bvb 2,5kwh is (na ongeveer 10 minuten), dan zou je die boiler kunnen uitschakelen. Zo blijft er nog over voor andere verbruikers zonder dat je die 3kwh passeert.
Gewoon de boiter 10 minuten aan, en 5 minuten uit is risicovoller, want je weet niet of je in dat specifieke kwartier een andere grote verbruiker hebt (bvb 5 minuten een haardroger).

Maar dan moet die huidige (voorlopige) piek wel ook beschikbaar zijn via MQTT uiteraard.

Is dat haalbaar denk je?
- huidige piek berekenen
- beschikbaar maken via mqtt

Bedankt!

dennissiemensma, 2022-05-27:
Bedankt voor je suggestie! Ik denk dat het handiger is om dat zelf in HA te berekenen, zodat je zelf de controle hebt over de intervallen/updates/triggers. Je kunt het op meerdere manieren en punten berekenen, want wellicht wil je altijd live de piek van de laatste 5, 10 en 15 minuten berekenen, zodat je op basis daarvan beslissingen kan nemen.

Daarnaast lijkt HA me daar ook een wat logischere plek voor, helemaal omdat DSMR-reader slechts "een" manier is om meterstanden in HA te krijgen, maar zeker niet de meest efficiente. Idem voor gerelateerde data.

DSMR-reader richt zich vooral op de analyse van gegevens achteraf, waarbij het nooit een vereiste is (of is geweest) dat het realtime verwerking moet zijn. Hoewel DSMR-reader dat in het gunstigste geval wel doet, kan ik dat nooit garanderen. De MQTT-integratie is meer ter deling van de beschikbare gegevens, maar niet een einddoel.
En ik denk dat je bij zo'n belangrijke monitor voor je use-case hierboven je eigenlijk zo dicht mogelijk bij de data moet zitten en ook zelf vrij moet zijn in wat je in de gaten wilt houden en op welke momenten.

VBP8501, 2022-05-27:
Ik gebruik openhab en doe de berekening van het kwartiervermogen daarin. Een bijkomend voordeel, naast wat Dennis al aanhaalt, is dat je ook rekening kunt houden met de huidige maximale piek van de lopende maand. Als je standaard schakelt bij om onder 3kW gemiddeld uit te komen en je gaat er toch nog eens over, bijvoorbeeld je hebt al eens 4kW kwartiervermogen bereikt, dan is het logischer om verder te werken met de nieuwe piek, 4kW, om te schakelen.

dennissiemensma, 2022-06-23:
@VBP8501 zijn de (lijstweergaves) in DSMR-reader v5.2 representatief voor wat je zelf al registreerde? Afgezien van de afwijking doordat DSMR-reader (nu nog) 15 minuten gebruikt als periode om te rekenen ipv de exact gemeten periode.

FredericMa, 2022-06-24:
Ik heb tot nu toe ook eigen metingen gemaakt in HA en de waarden in HA matchen met de geregistreerde waarden in DSMR-reader.

dennissiemensma, 2022-06-24:
Dank voor het checken!

VBP8501, 2022-06-27:
> @VBP8501 zijn de (lijstweergaves) in DSMR-reader v5.2 representatief voor wat je zelf al registreerde? Afgezien van de afwijking doordat DSMR-reader (nu nog) 15 minuten gebruikt als periode om te rekenen ipv de exact gemeten periode.

Sorry, ik heb al 2 weken performantie problemen met dsmr-reader. (cpu van mijn rapi gaat naar 100% en de docker container van dsmr crasht daardoor). Sinds de stroom in de straat is uit gezet omwille van een brand. Geen idee of de omstandigheden iets te maken hebben met de problemen maar daardoor kan ik niet verifiëren of de berekening OK is. Nadat ik had geupgrade naar 5.2 heb ik wel eens vlug gekeken en de getallen waren in grote orde wel correct. Van zodra ik mijn systeem terug stabiel krijg dan post ik hier het resultaat

VBP8501, 2022-07-17:
@dennissiemensma : performantieproblemen waren te wijten aan een andere container die op dezelfde pi draaide. Heel vreemd maar toen ik die op een andere pi opstartte werkte alles in dsmr reader weer normaal. 

De berekende waarden van dsmr reader zijn correct. Ik heb de waarden van de afgelopen dag vergeleken met wat ik zou berekenen en zie een minimaal verschil (gemiddeld 0,5W). 

Ik heb wel een paar opmerkingen over de eenheden:

[image]

De waarde die je berekend is in kW (dus niet in kWh of kW/h). In de screenshot zouden de rode vakjes "kW" moeten worden en de blauwe "kWh/15m".

dennissiemensma, 2022-07-17:
Bedankt voor je update! Ik voeg het toe aan #1635

Milithor, 2022-11-24:
Nog een belg hier, die even een bedankje wil uitbrengen aan degene die de moeite doen om dit erin te krijgen!
Ook die MQTT ontsluiting naar HomeAssistant lijkt me zeer interessant!

[1 more reply]