dsmrreader/dsmr-reader #1085: X-Frame-Options uitzetten/aanpassen

syphernl, 2020-08-19:
In DSMR-Reader 3.x was er een optie in `settings.py`:
```X_FRAME_OPTIONS = 'ALLOWALL'``` zodat hij in een frame geladen kon worden (bijv in HASS).

In 4.x is de configuratie aangepast naar `.env`, echter [ondersteund](https://dsmr-reader.readthedocs.io/en/latest/env_settings.html) deze geen configuratie om de `X-Frame-Options` aan te passen. Is het mogelijk om dit toe te voegen of op een andere manier deze instelling te wijzigen?

dennissiemensma, 2020-08-19:
Je mag dit op dezelfde plek in settings.py toevoegen. Zie ook #1080. Vergeet niet te reloaden/herstarten.

syphernl, 2020-08-19:
Bedankt voor je snelle reactie! Ik ging ervan uit dat de `.env` de `settings.py` volledig zou vervangen maar dit werkt inderdaad 👍🏻

closed 2020-08-19 by syphernl

dennissiemensma, 2020-08-19:
Ik zal de docs bijwerken met de volgende release zodat dit wat duidelijker vermeld staat.

De .env is vanuit gebruikersperspectief een vervanger, maar technisch is het een alias of schil om de settings heen.

dennissiemensma, 2020-10-03:
Vanaf `v4.5` is dit straks in te stellen (overschrijven) via de env als `DJANGO_USE_X_FRAME_OPTIONS`.