dsmrreader/dsmr-reader #1086: Vervanger voor Status endpoint?

syphernl, 2020-08-20:
Er is in mijn setup ooit iets mis gegaan waardoor de Unprocessed Telegrams opliepen. Om dat te monitoren heb ik binnen HASS een sensor gemaakt die `/api/v2/application/status` uitleest. 

Met 3.x werkte dat prima, sinds mijn upgrade naar 4.x is deze sensor unavailable geworden en blijkt dat dit endpoint in 4.x [verwijderd](https://github.com/dennissiemensma/dsmr-reader/issues/1024) is. Is hier een alternatief voor handen?

dennissiemensma, 2020-08-20:
Bedankt voor je reminder. Ik heb er nog geen alternatief voor omdat een deel van de processen nog omgebouwd moet worden en het voormalige endpoint dat deels in de weg zat. Ik zal dit weer opnieuw op de backlog zetten.

helmo, 2020-09-01:
Mijn voorlopige oplossing is nu een Nagios http check:

```
/usr/lib/nagios/plugins/check_http -H dsmr.local --regex=alert-danger --invert-regex -u http://dsrm.local/status
```

dennissiemensma, 2020-09-10:
Alvast een kleine opzet gemaakt. Ik denk dat ik het allemaal wat ga versimpelen, ook op de status pagina.

Straks is er een endpoint die vertelt hoeveel issues er zijn, met beknopte details. Dat kan er zo uit komen te zien:

```
{
    "problems": 6,
    "details": [
        {
            "source": "dsmr_backend.apps",
            "description": "Process behind schedule: Create daily backup",
            "since": "2020-09-10T06:00:00Z"
        },
        {
            "source": "dsmr_backend.apps",
            "description": "Process behind schedule: Weather update",
            "since": "2020-09-09T20:00:00Z"
        },
        {
            "source": "dsmr_backend.apps",
            "description": "Process behind schedule: Generate day and hour statistics",
            "since": "2020-09-09T22:00:00Z"
        },
        {
            "source": "dsmr_backend.apps",
            "description": "Process behind schedule: Generate consumption data",
            "since": "2020-09-09T19:05:54.824064Z"
        },
        {
            "source": "dsmr_backend.apps",
            "description": "Process behind schedule: Retention data rotation",
            "since": "2020-08-26T06:38:56.510712Z"
        },
        {
            "source": "dsmr_datalogger.apps",
            "description": "No recent readings received",
            "since": "2020-09-09T19:05:25Z"
        }
    ]
}
```

GieltjE, 2020-09-16:
Voor "simpele" tools als nagios is het practisch als er een http endpoint is die een non 200 code (500, 502) teruggeeft als er iets niet in orde is, merk nu na een week pas dat ik ergens de kernel gestript heb van de drivers voor de usb<>serial convertor.

dennissiemensma, 2020-09-16:
Bedankt voor je aanvulling. Ik zal het meenemen.

dennissiemensma, 2020-09-16:
@GieltjE welke status code wordt daar overigens doorgaans voor gebruikt? Ik zoek er eentje waar zowel Nagios als HTTP clients mee overweg kunnen. 

In principe heeft Nagios daar namelijk zelf ook al tools voor om de request zelf te parsen (en de structuur blijft immers toch gelijk in alle situaties): 
https://exchange.nagios.org/directory/Plugins/Websites,-Forms-and-Transactions/check_json--2D-check-JSON-attributes-via-http(s)/details

dennissiemensma, 2020-09-16:
Vermoedelijk heb je alleen zoiets nodig straks:

```
./check_json.pl --url http://IP/api/v2/application/monitoring --attribute '{problems}' --warning :1 --critical :5 --perfvars '{problems}' 
```

helmo, 2020-09-17:
Die versie van check_json mist alleen nog wel authenticatie, maar een van de forks daarvan lijkt dat wel te hebben.

Hieronder als referentie het check script wat ik voor v3 in gebruik had. 

```python
#! /usr/bin/env python
import sys
import requests         # debian://python-requests
import json

api_url = "http://dsmr.example.com/api/v2/application/status"
api_key = "*******"

unprocessed_warning_level = 300
seconds_ago_warning_level = 300

# Read data from the API
response = requests.get(
    api_url,
    headers={'X-AUTHKEY': api_key},
)

# NAGIOS return codes :
# https://nagios-plugins.org/doc/guidelines.html#AEN78
OK       = 0
WARNING  = 1
CRITICAL = 2
UNKNOWN  = 3

if response.status_code != 200:
    print('Error: {}'.format(response.text))
    sys.exit(UNKNOWN)
else:
    data = json.loads(response.text)
    #print('Data: {}'.format(data))
    if (data["readings"]["unprocessed"]["count"] > unprocessed_warning_level):
        print('WARNING: Unprocessed telegrams {}'.format(data["readings"]["unprocessed"]["count"]));
        sys.exit(WARNING)
    elif (data["readings"]["seconds_since"] > seconds_ago_warning_level):
        print('WARNING: Last telegram was received too long ago. Seconds: {}'.format(data["readings"]["seconds_since"]));
        sys.exit(WARNING)
    else:
        print("OK")
        sys.exit(OK)
```

dennissiemensma, 2020-09-18:
Dank voor je toevoeging! In dat geval lijkt het me helemaal prima om gewoon de HTTP-standaard (2xx) te volgen qua status codes, zodat elke client ermee overweg kan.

dennissiemensma, 2020-09-22:
Inmiddels is het gelukt om een goede vervanger te maken voor vanaf v4.4. 

Er komt een nieuw endpoint `/api/v2/application/monitoring` met als voorbeeld-responses:

## OK
```
{
    "problems": 0,
    "details": []
}
```

## Problemen
```
{
    "problems": 8,
    "details": [
        {
            "source": "dsmr_backend.apps",
            "description": "Process behind schedule: Backup per email",
            "since": "2020-09-22T21:39:19.953043+02:00"
        },
        {
            "source": "dsmr_backend.apps",
            "description": "Process behind schedule: Retention data rotation",
            "since": "2020-09-22T09:39:24.928241+02:00"
        },
        {
            "source": "dsmr_backend.apps",
            "description": "Process behind schedule: Create daily backup",
            "since": "2020-09-22T08:00:00+02:00"
        },
        {
            "source": "dsmr_backend.apps",
            "description": "Process behind schedule: Generate day and hour statistics",
            "since": "2020-09-22T00:00:00+02:00"
        },
        {
            "source": "dsmr_backend.apps",
            "description": "Process behind schedule: Weather update",
            "since": "2020-09-21T23:00:00+02:00"
        },
        {
            "source": "dsmr_backend.apps",
            "description": "Process behind schedule: Generate consumption data",
            "since": "2020-09-21T22:52:20.576332+02:00"
        },
        {
            "source": "dsmr_datalogger.apps",
            "description": "No recent readings received",
            "since": "2020-09-21T22:51:12+02:00"
        },
        {
            "source": "dsmr_pvoutput.apps",
            "description": "Waiting for the next PVOutput export to be executed",
            "since": "2019-05-09T06:00:00+02:00"
        }
    ]
}
```

----

Verder is de Status-pagina helemaal omgegooid zodat het een representatie is van hetzelfde endpoint. Dit maakt het veel makkelijker om ze in sync te houden:

[image]

Daarmee staat er veel minder informatie voor de gebruikers. Men hoeft niet meer de complete pagina af te scannen of er iets stuk is, dat is direct te zien:

[image]

----

Minder belangrijk voor de Status, maar wel voor de layout: Het menu is ook flink opgeschoond en samengevoegd tot een algemeen "About" item.

dennissiemensma, 2020-09-22:
Gaat mee in v4.4

closed 2020-09-22 by dennissiemensma

dennissiemensma, 2020-09-25:
Zojuist uitgebracht in [v4.4](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.4)

helmo, 2020-09-26:
Top, hier is ook de bijgewerkte variant van het check script.

```python
#! /usr/bin/env python
import sys
import requests         # debian://python-requests
import json

# EDIT HERE, get the API key from /admin/dsmr_api/apisettings/1/change/
api_url = "http://dsmr.example.com/api/v2/application/monitoring"
api_key = ""

# Read data from the API
response = requests.get(
    api_url,
    headers={'X-AUTHKEY': api_key},
)

# NAGIOS return codes :
# https://nagios-plugins.org/doc/guidelines.html#AEN78
OK       = 0
WARNING  = 1
CRITICAL = 2
UNKNOWN  = 3

if response.status_code != 200:
    print('Error: {}'.format(response.text))
    sys.exit(UNKNOWN)
else:
    data = json.loads(response.text)
    #print('Data: {}'.format(data))
    if (data["problems"] > 0):
        print('WARNING: Detected problems on status page: {}'.format(data["problems"]));
        sys.exit(WARNING)
    else:
        print("OK")
        sys.exit(OK)
```

GieltjE, 2020-09-28:
Script werkt perfect, moest alleen even python door python3 vervangen

dennissiemensma, 2020-10-18:
Ter info, in de aankomende release (waarschijnlijk morgen) is de Status-pagina helemaal weg en staan de checks op de "About" pagina.

Dit betekent dat naar gebruikers toe de Status niet zo belangrijk meer is, maar meer of er problemen zijn die spelen. Dus omgedraaid. Met de v4.4 release had ik hier al een eerste aanzet voor gemaakt en dat is nu voltooid.

### About
[image]

----

### Badge
Mochten er issues zijn dan zijn die daar te vinden. Daarnaast heb ik nu een badge toegevoegd in het menu, zodat wat sneller duidelijker is als er iets aan de hand is.

[image]

----

### Samengevoegd
Hiermee is de refactoring en samenvoeging van de oude Status-pagina en de monitoring compleet. Het is nu een en dezelfde logica waar ik veel makkelijker checks aan kan toevoegen dan eerst. Nieuwe checks zijn bijvoorbeeld:
- meer dan 20 miljoen opgeslagen metingen
- database is groter dan 1 GB 

De API-toevoeging van v4.4 maakt dus ook nog steeds gebruik van deze zelfde structuur.

### Verder
De exacte offsets zijn uiteraard nog te tunen en wellicht dat ik nog ooit warnings van maak, dus dat er naast errors ook warnings komen. Maar ik kijk eerst wel even hoevaak iemand er tegenaan loopt in de huidige vorm en of het wel of niet 'irritant' is in deze vorm.

Mixpower, 2020-10-18:
> Ter info, in de aankomende release (waarschijnlijk morgen) is de Status-pagina helemaal weg en staan de checks op de "About" pagina.
>
> Dit betekent dat naar gebruikers toe de Status niet zo [...]

Top, thanks for the heads up ziet er strak uit zo!

commit b5ab8f12 (2020-09-21): Pylama fix #1086