dsmrreader/dsmr-reader #1089: Paar kleine foutjes in de handleiding?

bduijnhouwer, 2020-08-25:
Goedemiddag,

Ik heb (wellicht) een paar kleine foutjes (of onduidelijkheden) gevonden in de handleiding.

Het gaat om de handleiding: https://dsmr-reader.readthedocs.io/nl/v4/installation/explained.html

Wanneer ik in stap 8:  ./manage.py migrate  uitvoer, krijg ik een foutmelding dat dependency 'psycopg2' mist.
"pip3 install psycopg2" lost dit op voor mij.

Wanneer ik in stap 8: ./manage.py dsmr_superuser krijg ik : ""CommandError: DSMR_USER or DSMR_PASSWORD (or both) are empty"
-- Ik snap nu dat ik eerst een username + password in dien te vullen in het .env bestand, maar dit was (voor een leek) niet direct duidelijk.

Stap 9:
Wanneer ik in stap 9: "sudo service nginx reload" doe, krijg ik: "nginx.service is not active, cannot reload."
Oplossing: sudo service nginx start

-- Dit komt waarschijnlijk omdat sudo apt install nginx een foutmelding geeft bij een nieuwe Raspberry Pi installatie.
Hij wordt wel geinstalleerd, maar kan niet starten omdat er standaard een configfout in zit vermoed ik.

Bastiaan

dennissiemensma, 2020-08-25:
Bedankt voor het doorgeven! 

-Welke versie van Raspberry Pi OS gebruik je?

## psycopg2
Als het goed is heb je bij stap 2 `python3-psycopg2` geinstalleerd. Het hangt er even vanaf of dat niet goed is gegaan of niet (meer) werkt op het OS.

## dsmr_superuser
Ik zal de foutmelding wat duidelijker maken.

## nginx
Wellicht is dit veranderd in het laatste OS. Ik kan het eventueel omzetten naar bijvoorbeeld `systemctl`, de opvolger van `service`, plus het enablen van de service.

bduijnhouwer, 2020-08-25:
Ik heb een "Raspberry Pi 4 Model B Rev 1.1"
Ik zou inderdaad bij stap 2 python3-psycopg2 geinstalleerd moeten hebben, maar ik zie net dat deze niet is geinstalleerd.
Kan dit zijn omdat ik e.e.a. in een virtualenv heb geinstalleerd?

Wanneer ik nu apt get install python3-psycopg2 doe wordt deze wel geinstalleerd, maar dus zonder deze 'toevoeging' werkt e.e.a. wel bij mij. (Zelf denk ik dus vanwege de virtualenv).

Als ik tijd heb zal ik morgen even een nieuwe installatie op een 'vers' sd kaartje proberen. En dan zal ik wat meer aantekeningen maken.

dennissiemensma, 2020-08-25:
Bedankt voor je aanvulling. Ik zoek vooral de versie van het OS dat je gebruikt:
```
lsb_release -a
```

De normale installatie zou `python3-psycopg2` systeembreed moeten installeren en vervolgens bij het aanmaken van de virtualenv door de `--system-site-packages` meegekopieerd moeten worden.

De manier die je nu hebt gebruikt kan ook, dat is hoe DSMR-reader het in eerdere versies deed.

dennissiemensma, 2020-08-26:
Ik heb de superuser-foutmelding aangepast en zal de nginx directives wijzigen naar de modernere versie (die ook op mn PI lijkt te werken):
```
    sudo nginx -t
    sudo systemctl restart nginx.service
```

dennissiemensma, 2020-08-26:
Mocht je nog meer dingen vinden, laat maar hier of in een apart issue weten!

closed 2020-08-26 by dennissiemensma

dennissiemensma, 2020-08-28:
Uitgebracht in [v4.3](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.3.0) (docs).