dsmrreader/dsmr-reader #1090: Telegram parse error in 4.2

leadenchopsticks, 2020-08-25:
Wat gebruik je?
- DSMR-reader versie:   4.2
- Type RaspberryPi or server:   Raspberry Pi 2 Model B Rev 1.1
- Standaardinstallatie of Docker:   standaard
- DSMR-protocol slimme meter v4/v5:   v4/v5

---

Heb vandaag geüpdate naar 4.2 (vanaf 3) zonder problemen. Echter het parsen van de data gaat ineens niet meer goed:

```
[2020-08-25 22:09:06,354] ERROR    Failed to perform CRC validation because the telegram is incomplete. The checksum and/or content values are missing.
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services/datalogger.py", line 86, in telegram_to_reading
    parsed_telegram = parser.parse(data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 39, in parse
    self.validate_checksum(telegram_data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 69, in validate_checksum
    'Failed to perform CRC validation because the telegram is '
dsmr_parser.exceptions.ParseError: Failed to perform CRC validation because the telegram is incomplete. The checksum and/or content values are missing.

```
Als ik met cu kijk zie ik wel gewoon complete data:

```
1-3:0.2.8(50)
0-0:1.0.0(200825221033S)
0-0:96.1.1(4530303534303037343932313330363139)
1-0:1.8.1(000599.716*kWh)
1-0:1.8.2(000536.424*kWh)
1-0:2.8.1(000000.153*kWh)
1-0:2.8.2(000001.270*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.674*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00016)
0-0:96.7.9(00008)
1-0:99.97.0(6)(0-0:96.7.19)(200617142221S)(0000002699*s)(200629140534S)(0000000279*s)(200709143737S)(0000008325*s)(200718221801S)(0000000472*s)(200720180504S)(0000025401*s)(200806094440S)(0000000486*s)
1-0:32.32.0(00001)
1-0:52.32.0(00001)
1-0:72.32.0(00001)
1-0:32.36.0(00001)
1-0:52.36.0(00001)
1-0:72.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(232.2*V)
1-0:52.7.0(232.3*V)
1-0:72.7.0(232.2*V)
1-0:31.7.0(000*A)
1-0:51.7.0(001*A)
1-0:71.7.0(002*A)
1-0:21.7.0(00.044*kW)
1-0:41.7.0(00.172*kW)
1-0:61.7.0(00.456*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303332353635353339383339373137)
0-1:24.2.1(200825221008S)(00664.210*m3)
!F789
/ISK5\2M550T-1013
```

Vreemde is dat er af en toe wel een telegram in de db terecht komt maar dat is maar 1 keer per uur ofzo.

dennissiemensma, 2020-08-25:
Bedankt voor je melding. Zou je het loglevel eens op DEBUG willen zetten en dan de datalogger herstarten?

https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#logging-levels

Je zou dan in de datalogger heel uitgebreid moeten zien wat die doet. Ter referentie van output: https://github.com/dennissiemensma/dsmr-reader/issues/1081#issuecomment-675084906

leadenchopsticks, 2020-08-26:
Hier is een complete cycle in debug mode:

```
[2020-08-26 08:11:09,268] DEBUG    dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 0.5s
[2020-08-26 08:11:09,520] DEBUG    [2020-08-26 08:11:09.520455] Read 5 Byte(s)
[2020-08-26 08:11:09,772] DEBUG    [2020-08-26 08:11:09.771874] Read 0 Byte(s)
[2020-08-26 08:11:10,023] DEBUG    [2020-08-26 08:11:10.023367] Read 0 Byte(s)
[2020-08-26 08:11:10,125] DEBUG    [2020-08-26 08:11:10.125456] Read 512 Byte(s)
[2020-08-26 08:11:10,189] DEBUG    [2020-08-26 08:11:10.188991] Read 512 Byte(s)
[2020-08-26 08:11:10,202] DEBUG    Received telegram:
/ISK5\2M550T-1013

1-3:0.2.8(50)
0-0:1.0.0(200826081107S)
0-0:96.1.1(4530303534303037343932313330363139)
1-0:1.8.1(000605.555*kWh)
1-0:1.8.2(000537.806*kWh)
1-0:2.8.1(000000.153*kWh)
1-0:2.8.2(000001.270*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.496*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00016)
0-0:96.7.9(00008)
1-0:99.97.0(6)(0-0:96.7.19)(200617142221S)(0000002699*s)(200629140534S)(0000000279*s)(200709143737S)(0000008325*s)(200718221801S)(0000000472*s)(200720180504S)(0000025401*s)(200806094440S)(0000000486*s)
1-0:32.32.0(00001)
1-0:52.32.0(00001)
1-0:72.32.0(00001)
1-0:32.36.0(00001)
1-0:52.36.0(00001)
1-0:72.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(233.9*V)
1-0:52.7.0(234.4*V)
1-0:72.7.0(234.1*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(002*A)
1-0:21.7.0(00.030*kW)
1-0:41.7.0(00.077*kW)
1-0:61.7.0(00.390*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303332353635353339383339373137)
0-1:24.2.1(200826081006S)(00664.867*m3)
!C
[2020-08-26 08:11:10,216] WARNING  Rejected telegram (Failed to perform CRC validation because the telegram is incomplete. The checksum and/or content values are missing.) (base64 encoded): b'L0lTSzVcMk01NTBULTEwMTMNCg0KMS0zOjAuMi44KDUwKQ0KMC0wOjEuMC4wKDIwMDgyNjA4MTEwN1MpDQowLTA6OTYuMS4xKDQ1MzAzMDM1MzQzMDMwMzczNDM5MzIzMTMzMzAzNjMxMzkpDQoxLTA6MS44LjEoMDAwNjA1LjU1NSprV2gpDQoxLTA6MS44LjIoMDAwNTM3LjgwNiprV2gpDQoxLTA6Mi44LjEoMDAwMDAwLjE1MyprV2gpDQoxLTA6Mi44LjIoMDAwMDAxLjI3MCprV2gpDQowLTA6OTYuMTQuMCgwMDAyKQ0KMS0wOjEuNy4wKDAwLjQ5NiprVykNCjEtMDoyLjcuMCgwMC4wMDAqa1cpDQowLTA6OTYuNy4yMSgwMDAxNikNCjAtMDo5Ni43LjkoMDAwMDgpDQoxLTA6OTkuOTcuMCg2KSgwLTA6OTYuNy4xOSkoMjAwNjE3MTQyMjIxUykoMDAwMDAwMjY5OSpzKSgyMDA2MjkxNDA1MzRTKSgwMDAwMDAwMjc5KnMpKDIwMDcwOTE0MzczN1MpKDAwMDAwMDgzMjUqcykoMjAwNzE4MjIxODAxUykoMDAwMDAwMDQ3MipzKSgyMDA3MjAxODA1MDRTKSgwMDAwMDI1NDAxKnMpKDIwMDgwNjA5NDQ0MFMpKDAwMDAwMDA0ODYqcykNCjEtMDozMi4zMi4wKDAwMDAxKQ0KMS0wOjUyLjMyLjAoMDAwMDEpDQoxLTA6NzIuMzIuMCgwMDAwMSkNCjEtMDozMi4zNi4wKDAwMDAxKQ0KMS0wOjUyLjM2LjAoMDAwMDEpDQoxLTA6NzIuMzYuMCgwMDAwMSkNCjAtMDo5Ni4xMy4wKCkNCjEtMDozMi43LjAoMjMzLjkqVikNCjEtMDo1Mi43LjAoMjM0LjQqVikNCjEtMDo3Mi43LjAoMjM0LjEqVikNCjEtMDozMS43LjAoMDAwKkEpDQoxLTA6NTEuNy4wKDAwMCpBKQ0KMS0wOjcxLjcuMCgwMDIqQSkNCjEtMDoyMS43LjAoMDAuMDMwKmtXKQ0KMS0wOjQxLjcuMCgwMC4wNzcqa1cpDQoxLTA6NjEuNy4wKDAwLjM5MCprVykNCjEtMDoyMi43LjAoMDAuMDAwKmtXKQ0KMS0wOjQyLjcuMCgwMC4wMDAqa1cpDQoxLTA6NjIuNy4wKDAwLjAwMCprVykNCjAtMToyNC4xLjAoMDAzKQ0KMC0xOjk2LjEuMCg0NzMwMzAzMzMyMzUzNjM1MzUzMzM5MzgzMzM5MzczMTM3KQ0KMC0xOjI0LjIuMSgyMDA4MjYwODEwMDZTKSgwMDY2NC44NjcqbTMpDQohQw=='
[2020-08-26 08:11:10,217] ERROR    Failed to perform CRC validation because the telegram is incomplete. The checksum and/or content values are missing.
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/services/datalogger.py", line 86, in telegram_to_reading
    parsed_telegram = parser.parse(data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 39, in parse
    self.validate_checksum(telegram_data)
  File "/home/dsmr/dsmr-reader/dsmr_parser/parsers.py", line 69, in validate_checksum
    'Failed to perform CRC validation because the telegram is '
dsmr_parser.exceptions.ParseError: Failed to perform CRC validation because the telegram is incomplete. The checksum and/or content values are missing.
```

dennissiemensma, 2020-08-26:
Dank voor de update, ik heb nu wel een vermoeden. Zou je in dsmr_datalogger/scripts/dsmr_datalogger_api_client.py op regel 23 de waarde van MAX_BYTES_READ van 512 naar 600 willen verhogen en dan de datalogger willen herstarten?

leadenchopsticks, 2020-08-26:
Yup dat gaat een stuk beter.. kennelijk spuugt mijn meter meer uit dan verwacht :-)

dennissiemensma, 2020-08-26:
Fijn dat dat het direct oploste! Je hebt eerlijk gezegd gewoon pech gehad, want de grootte van het bericht maakt niet uit (DSMR-reader bijft doorlezen), maar wel het moment van afkappen.

De datalogger stopt met lezen zodra die het einde van een telegram ziet (de `!`). Na dat teken komen nog 4 tekens voor de checksum die faalde. Alleen is die checksum (de 4 tekens) **optioneel** omdat oude meters die niet altijd hebben. 
En jij hebt net de pech dat jouw berichten **exact** `1027` tekens zijn en zonder de checksum `1023` tekens. Dus na 2x `512` Bytes te lezen ziet de datalogger het einde en kapt het af. Met een halve checksum, waardoor je dus jouw fouten kreeg dat de parser er niets van snapte.

Dit komt overigens door de variabele regel die begint met `1-0:99.97.0` en bij iedereen anders is. Daarom heb ik er zelf nooit last van gehad bij ontwikkelen.

Lang verhaal kort: Dus je hebt een hele mooie bug/edge-case gevonden die ik meteen zal fixen voor de volgende release!

dennissiemensma, 2020-08-26:
Vergeet overigens niet om je loglevel van `DEBUG` af te halen. Dat scheelt je onnodige writes op je SD-kaartje.

Voor nu kun je de aangepaste waarde in de datalogger laten voor wat het is. Bij de volgende update van DSMR-reader is het afdoende om eerst `git reset --hard HEAD` uit te voeren en daarna de `./deploy.sh` voor de update as usual.

dennissiemensma, 2020-08-26:
`MAX_BYTES_PER_READ` voor volgende release (v4.3) verhoogd naar  `2048` Bytes.

closed 2020-08-26 by dennissiemensma

dennissiemensma, 2020-08-28:
Uitgebracht in [v4.3](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.3.0).

Je kunt updaten met:

```
sudo su - dsmr
git reset --hard HEAD
./deploy.sh
```

leadenchopsticks, 2020-08-29:
Thanks, gelukt.