dsmrreader/dsmr-reader #1092: Server error

Simanias, 2020-08-27:
- DSMR-reader versie:   4.2
- Type RaspberryPi or server:   ESXi 6.5
- Standaardinstallatie of Docker:   standaard
- DSMR-protocol slimme meter v4/v5:   v5
---

Na upgrade van DSMR-reader naar versie 4.2, krijg ik opera elke keer de volgende foutmelding als ik op configuratie klik:

```
Server Error
Sorry, something unexpected happened.
Exception:
Error: Incorrect padding
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 179, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/generic/base.py", line 73, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/auth/mixins.py", line 50, in dispatch
    if not request.user.is_authenticated:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/functional.py", line 240, in inner
    self._setup()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/functional.py", line 376, in _setup
    self._wrapped = self._setupfunc()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/auth/middleware.py", line 23, in <lambda>
    request.user = SimpleLazyObject(lambda: get_user(request))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/auth/middleware.py", line 11, in get_user
    request._cached_user = auth.get_user(request)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/auth/__init__.py", line 174, in get_user
    user_id = _get_user_session_key(request)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/auth/__init__.py", line 58, in _get_user_session_key
    return get_user_model()._meta.pk.to_python(request.session[SESSION_KEY])

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/sessions/backends/base.py", line 65, in __getitem__
    return self._session[key]

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/sessions/backends/base.py", line 220, in _get_session
    self._session_cache = self.load()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/sessions/backends/db.py", line 44, in load
    return self.decode(s.session_data) if s else {}

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/sessions/backends/base.py", line 122, in decode
    return self._legacy_decode(session_data)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/sessions/backends/base.py", line 126, in _legacy_decode
    encoded_data = base64.b64decode(session_data.encode('ascii'))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/base64.py", line 87, in b64decode
    return binascii.a2b_base64(s)
```

Dit is niet het geval bij firefox of safari.

dennissiemensma, 2020-08-27:
Bedankt voor je melding. Ik heb dit niet eerder gezien. 

Aan de stacktrace van de foutmelding te zien zit het in Django, het onderliggende framework dat door DSMR-reader gebruikt wordt. Ik zal kijken of ik er meer over kan vinden, maar het lijkt me niet iets wat heel eenvoudig opgelost kan worden.

- Heb je al geprobeerd om in Opera de cookies voor DSMR-reader weg te gooien en opnieuw in te loggen? Of ben je juist nog niet ingelogd?

Simanias, 2020-08-27:
Ik heb de cookies verwijderd en opnieuw ingelogd en nu werkt het wel gewoon weer.
Bedankt voor de snelle response.

dennissiemensma, 2020-08-27:
Jij ook bedankt voor het proberen en de snelle terugkoppeling!

closed 2020-08-27 by dennissiemensma