dsmrreader/dsmr-reader #1093: status page often alerts about 'Generate consumption data'

helmo, 2020-08-30:
Wat gebruik je?
- DSMR-reader versie:   4.x
- Type RaspberryPi of server:  VM
- Standaardinstallatie of Docker:   standaard
- DSMR-protocol slimme meter v4/v5:   v4

---

Het viel me op na de upgrade naar 4.x dat op de status page elke paar seconden een item rood is.
Het gaat om "Generate consumption data".

Ik was opzoek naar een tijdelijke vervanger voor #1086 en hoopte met het matchen op een css class 'alert-danger' wat te hebben.

`/usr/lib/nagios/plugins/check_http -H dsmr.local  --regex="alert-danger" --invert-regex -u http://dsrm.local/status`

Ik heb het idee dat de 'Next run' tijd nog niet is bijgewerkt, waardoor die in het verleden ligt... 

Rood:
```
Last run: a second ago
Next run: now

Last run: 2 seconds ago
Next run: a second ago

Last run: 3 seconds ago
Next run: 2 seconds ago

Last run: 4 seconds ago
Next run: 3 seconds ago
```

Groen:
```
Last run: now
Next run: now
```

dennissiemensma, 2020-08-30:
Bedankt voor je melding. Ik kan me voorstellen dat het wat verwarrend is. De tijden bij `Next run` zijn eigenlijk `Next run planned`. Ik zal dat bij een toekomstige release wat duidelijker maken. En wellicht de kleuren voor die processen neutraler maken.

dennissiemensma, 2020-08-31:
Ik heb de tekst en kleur aangepast voor de volgende release.

closed 2020-08-31 by dennissiemensma

dennissiemensma, 2020-09-25:
Zojuist uitgebracht in [v4.4](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.4)

dennissiemensma, 2020-09-25:
Dat zeggende hebben, de Status-pagina is op de schop gegaan deze week. Dus bovenstaande gaat uberhaupt niet meer werkend bedenk ik me net.

Mixpower, 2020-09-27:
> Dat zeggende hebben, de Status-pagina is op de schop gegaan deze week. Dus bovenstaande gaat uberhaupt niet meer werkend bedenk ik me net.

Wat is er gebeurt met de Status pagina? Er word hier nu helemaal niets meer weergegeven, en ook het checken van de versie is verdwenen?

dennissiemensma, 2020-09-27:
Dit is gewijzigd in #1093, zie daar voor meer informatie. De versie-check zal ik toevoegen aan "About" zodat die weer beschikbaar is.

Mixpower, 2020-09-27:
Oké, maar ik denk dat de Status pagina meer geschikt is voor de versie-check.

dennissiemensma, 2020-10-18:
@Mixpower ter info, voor de aankomende v4.5 release wederom wijzingen: https://github.com/dennissiemensma/dsmr-reader/issues/1086#issuecomment-711391141