dsmrreader/dsmr-reader #1094: Kan energiecontract niet aanpassen

gnissahr, 2020-08-31:
DSMR-reader versie: 4.3
Type RaspberryPi of server: Proxmox LXC container, Debian 10
Standaardinstallatie of Docker: standaard
DSMR-protocol slimme meter v4/v5: v5

Na het updaten naar versie 4.3 geprobeerd om de dagprijs voor electra toe te voegen. Maar kan niets aanpassen. Ook het aanmaken van een nieuw contract lukt niet.

Hieronder de foutmelding:

Server Error
Sorry, something unexpected happened.
Exception:
ValueError: Cannot use None as a query value
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 179, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/admin/options.py", line 614, in wrapper
    return self.admin_site.admin_view(view)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/decorators.py", line 130, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/decorators/cache.py", line 44, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/admin/sites.py", line 233, in inner
    return view(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/admin/options.py", line 1653, in add_view
    return self.changeform_view(request, None, form_url, extra_context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/decorators.py", line 43, in _wrapper
    return bound_method(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/decorators.py", line 130, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/admin/options.py", line 1534, in changeform_view
    return self._changeform_view(request, object_id, form_url, extra_context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/admin/options.py", line 1573, in _changeform_view
    form_validated = form.is_valid()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/forms/forms.py", line 177, in is_valid
    return self.is_bound and not self.errors

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/forms/forms.py", line 172, in errors
    self.full_clean()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/forms/forms.py", line 375, in full_clean
    self._clean_form()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/forms/forms.py", line 402, in _clean_form
    cleaned_data = self.clean()

  File "/home/dsmr/dsmr-reader/dsmr_consumption/forms.py", line 27, in clean
    end__gte=current_end

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 942, in filter
    return self._filter_or_exclude(False, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 962, in _filter_or_exclude
    clone._filter_or_exclude_inplace(negate, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 969, in _filter_or_exclude_inplace
    self._query.add_q(Q(*args, **kwargs))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/query.py", line 1358, in add_q
    clause, _ = self._add_q(q_object, self.used_aliases)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/query.py", line 1380, in _add_q
    split_subq=split_subq, check_filterable=check_filterable,

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/query.py", line 1244, in build_filter
    check_filterable=check_filterable,

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/query.py", line 1380, in _add_q
    split_subq=split_subq, check_filterable=check_filterable,

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/query.py", line 1244, in build_filter
    check_filterable=check_filterable,

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/query.py", line 1380, in _add_q
    split_subq=split_subq, check_filterable=check_filterable,

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/query.py", line 1319, in build_filter
    condition = self.build_lookup(lookups, col, value)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/query.py", line 1170, in build_lookup
    raise ValueError("Cannot use None as a query value")

dennissiemensma, 2020-08-31:
Bedankt voor je melding. 

- Op welke versie zat je eerst?
- Welke gegevens vul je in?

Ik kan het zelf namelijk niet reproduceren met zowel een aanpassing als een nieuw contract:

[image]

- Er is in v4.2 wel extra validatie toegevoegd op overlappende contracten (`Fixed` Prevent overlapping dates in energy contracts [#1012]). Al zou je daar een normale foutmelding van moeten krijgen en geen crash.

gnissahr, 2020-08-31:
Ik ben begonnen op 3.11. ik heb toen 1 enkel contract ingevuld. Ik heb nu nog steeds 1 contract. Maar bij alles wat ik doe krijg ik "Cannot use None as a query value".

De reden dat ik er nu tegenaan loop is dat ik het dagtarief in wil vullen.
Naar aanleiding van je screenshot heb ik ook een einddatum voor het contract ingevuld (heb ik niet, maar goed).

Dan doet ie het wel.... Dus de "none" slaat blijkbaar op het ontbreken van een einddatum.

Mooi stukje software trouwens!

dennissiemensma, 2020-08-31:
Dank voor de debuggen! Ik zal het dan meenemen in een volgende release.

Voor nu kun je dan het contract 10 jaar (of verder) in de toekomst zetten als workaround.

dennissiemensma, 2020-08-31:
Opgelost in de volgende release. Nogmaals dank voor het uitzoeken.

closed 2020-08-31 by dennissiemensma

dennissiemensma, 2020-09-25:
Zojuist uitgebracht in [v4.4](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.4)