dsmrreader/dsmr-reader #1096: Current Power Usage wordt niet ge-update in HAA (mqtt)

Nenodema, 2020-08-31:
Sinds versie 4.0 wordt, zoals het lijkt, de laatste 0 waarde van De Current Power Usage niet meer via MQTT naar HAA gestuurd. In het voorbeeld hieronder kun je ziet er 7 minuten geleden 737kW werd verbruikt. Als ik vervolgens in DSMR kijk dan zie ik de Current Power Usage netjes op 0 staan en zie ik dat de zonnepanelen terugleveren.

Wat zou dit kunnen zijn? 

[image]

niekniek89, 2020-08-31:
Hallo,

Ik zit op V4.2, en heb dit probleem niet. Zodra de zonnepanelen wat gaan doen, schiet het stroomverbruik netjes naar 0 (zie screenshot).

[image]

Nenodema, 2020-08-31:
> Hallo,
> 
> Ik zit op V4.2, en heb dit probleem niet. Zodra de zonnepanelen wat gaan doen, schiet het stroomverbruik netjes naar 0 (zie screenshot).
> 
> [image]
> 
> op het moment van typen, zijn de zonnepanelen "weer" bezig (bewolking),

Dank voor je reactie! Heb je nog iets gewijzigd in HAA met de upgrade van dsmr 3 naar 4?

niekniek89, 2020-08-31:
> Dank voor je reactie! Heb je nog iets gewijzigd in HAA met de upgrade van dsmr 3 naar 4?

Nee, in HA heb ik niets veranderd/aangepast. Nu bedenk ik me dat ik wel het MQTT verkeer in DSMR gestopt en gestart heb (zie bijlage):

[image]

ik heb het vinkje weggehaald, opgeslagen en het vinkje vervolgens weer terug gezet. 
draait inmiddels nu 10 dagen zonder problemen en een reboot van de container (draai hem in docker).

Nenodema, 2020-08-31:
> Nee, in HA heb ik niets veranderd/aangepast. Nu bedenk ik me dat ik wel het MQTT verkeer in DSMR gestopt en gestart heb (zie bijlage):
>
> [image]
>
> ik heb het vinkje weggehaald, opgeslagen en het [...]

Het uit en inschakelen van MQTT heeft volgens mij geholpen! Ik wacht nog ff een dagje af en dan sluit ik hem. Thanks!

dennissiemensma, 2020-08-31:
Bedankt voor het melden en debuggen! Mocht je het alsnog niet werkend krijgen morgen, laat maar weten.

Nenodema, 2021-04-28:
Hi all,

Sinds een paar weken, ik denk vanaf versie 4.15, lijkt het issue weer op te duiken. MQTT uit en aanzetten lijkt dit keer ook niet te helpen.

[image]

Iemand?

dennissiemensma, 2021-04-28:
Je kunt het beste eerst kijken of er MQTT-berichten in de wacht staan in DSMR-reader, via de admin: `/admin/dsmr_mqtt/message/`. Bij goed gedrag zou daar niets in moeten (blijven) staan.

Als daar niets uit te herleiden is kun je de `dsmr_backend` logfiles bekijken of daar een fout in staat: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/logfiles.html

Verder kun je nog DEBUG-logging aanzetten: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
En dan in `dsmr_backend` kijken of je iets van MQTT-debuginfo voorbij ziet komen.

En als dat niets geeft, kun je eventueel andere MQTT-databronnen binnen DSMR-reader aanzetten en met een MQTT-explorer oid kijken of je dan iets ziet langskomen.

dennissiemensma, 2021-04-28:
Je zou trouwens ook nog kunnen kijken of een anders QoS-level in DSMR-reader, bij de MQTT-broker-instellingen, iets voor je verbetert. 
Het helpt overigens alleen als de verbinding naar de MQTT-broker wegvalt. Meer info in #1375, met de kanttekening dat ik zelf nog niet expliciet QoS 2 heb getest icm wegvallende broker, maar wellicht helpt het ook in jouw geval.

Nenodema, 2021-04-28:
Dank voor je uitgebreide reactie Dennis. Ik zal vandaag en morgen de QoS-level instellingen ook wijzigen en testen. 

Zou het ook nog aan de "Datalogger process sleep:" instelling kunnen liggen? Deze stond op 5 seconden en het lijkt erop dat de Raspberry PI niet snel genoeg is om alles verwerken. Als ik in de `/admin/dsmr_mqtt/message/` kijk dan staat deze eigenlijk constant vol.

dennissiemensma, 2021-04-28:
De datalogger stuurt de MQTT-berichten niet, dat doet de backend. Als de sleep voor de backend echter te hoog staat, dan kan dat ook nog een oorzaak zijn, al verwacht ik niet dat die dan op willekeurige momenten niets verstuurt (als het meestal wel goed gaat).

DSMR-reader bewaart momenteel max 200 MQTT-berichten, dus als je in die queue telkens 200 berichten ziet, dan zit daar ook nog (MQTT) dataverlies in. Maar dat is een ander dingetje, die limiet is nog wel vrij makkelijk op te hogen als dat ook nog meespeelt.

Nenodema, 2021-04-28:
Oké helder, dank voor je uitleg. Wat zijn de aanbevolen waarde voor datalogger en backend sleep?

QoS heb ik nu op 2 staan.

Nenodema, 2021-04-28:
In onderstaand screenshot kun je exact zien wat er precies misgaat. De 0.00 van de Currently Delivered wordt niet bijgewerkt, terwijl de Currently Returned wel bijgewerkt wordt.

[image]

dennissiemensma, 2021-04-28:
> Oké helder, dank voor je uitleg. Wat zijn de aanbevolen waarde voor datalogger en backend sleep?
> 
> QoS heb ik nu op 2 staan.

De datalogger hangt af wat je zelf wil qua datadichtheid, voor de totalen per uur/dag maakt het geen verschil. Opzich is een datalogger sleep van 5 a 10 seconden prima, dat scheelt je ook een factor 5 a 10 qua metingen in de opslag.

Voor de backend hangt het af of je zware load hebt. Ik zou die op 1 seconde zetten als dat verder geen problemen lijkt te geven qua load.

Nenodema, 2021-04-28:
> De datalogger hangt af wat je zelf wil qua datadichtheid, voor de totalen per uur/dag maakt het geen verschil. Opzich is een datalogger sleep van 5 a 10 seconden prima, dat scheelt je ook een factor [...]

Top, ga ik de backend sleep op 1 seconde zetten.

dennissiemensma, 2021-04-28:
> In onderstaand screenshot kun je exact zien wat er precies misgaat. De 0.00 van de Currently Delivered wordt niet bijgewerkt, terwijl de Currently Returned wel bijgewerkt wordt.
> 
> In de logging kom ik geen vreemde zaken tegen.

Ik zoek de gegevens die direct uit DSMR-reader verstuurd worden. De screenshots geven alleen het begin en eind van de gegevens-flow aan en daar zitten teveel schakels tussen om een oorzaak te vinden. 
De metingen van DSMR-reader zeggen niets over wat er verstuurd wordt en de interface in HA zegt niet perse wat over wat verzonden is.

Als je DEBUG-logging inschakelt zou je (dacht ik) moeten kunnen zien wat DSMR-reader precies verstuurt via MQTT, inclusief de topics en waardes. Die kun je dan het beste naast de metingen leggen om te kijken waar er verschil in zit (ik verwacht hooguit enkele seconden vertraging).

dennissiemensma, 2021-04-28:
Het lijkt er trouwens op dat een hoop gelogd wordt, maar toch niet de data inhoudelijk. Ik zal dat voor een volgende release uitbreiden. 

Los daarvan kun je nog kijken of de queue in `/admin/dsmr_mqtt/message/` niet meer zo vol staat.

Nenodema, 2021-04-28:
> Ik zoek de gegevens die direct uit DSMR-reader verstuurd worden. De screenshots geven alleen het begin en eind van de gegevens-flow aan en daar zitten teveel schakels tussen om een oorzaak te [...]

Snap ik. Hou je vast ;-)

```
/ISK5\2X5d0E-1232

1-3:0.2.8(50)
0-0:1.0.0(210428201358S)
0-0:96.1.1(4530303433323454223439363439343137)
1-0:1.8.1(003436.314*kWh)
1-0:1.8.2(003613.582*kWh)
1-0:2.8.1(001222.222*kWh)
1-0:2.8.2(002633.598*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.326*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00009)
0-0:96.7.9(00003)
1-0:99.97.0(1)(0-0:96.7.19)(181020954105S)(0000005160*s)
1-0:32.32.0(00018)
1-0:32.36.0(00002)
0-0:96.13.0()
1-0:32.7.0(236.3*V)
1-0:31.7.0(002*A)
1-0:21.7.0(00.286*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(473030333233894223343303234313137)
0-1:24.2.1(210428201001S)(03568.745*m3)
!F6B7
2021-04-28 20:12:57,994 | DEBUG    messages.queue_message:38 | MQTT: Queued message for dsmr/reading/timestamp
2021-04-28 20:12:57,999 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/reading/electricity_delivered_1 due to cache
2021-04-28 20:12:58,003 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/reading/electricity_returned_1 due to cache
2021-04-28 20:12:58,007 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/reading/electricity_delivered_2 due to cache
2021-04-28 20:12:58,012 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/reading/electricity_returned_2 due to cache
2021-04-28 20:12:58,033 | DEBUG    messages.queue_message:38 | MQTT: Queued message for dsmr/reading/electricity_currently_delivered
2021-04-28 20:12:58,038 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/reading/electricity_currently_returned due to cache
2021-04-28 20:12:58,059 | DEBUG    messages.queue_message:38 | MQTT: Queued message for dsmr/reading/phase_currently_delivered_l1
2021-04-28 20:12:58,070 | DEBUG    messages.queue_message:36 | MQTT: Ignoring dsmr/reading/phase_currently_delivered_l2 due to queue
2021-04-28 20:12:58,080 | DEBUG    messages.queue_message:36 | MQTT: Ignoring dsmr/reading/phase_currently_delivered_l3 due to queue
2021-04-28 20:12:58,084 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/reading/extra_device_timestamp due to cache
2021-04-28 20:12:58,088 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/reading/extra_device_delivered due to cache
2021-04-28 20:12:58,093 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/reading/phase_currently_returned_l1 due to cache
2021-04-28 20:12:58,103 | DEBUG    messages.queue_message:36 | MQTT: Ignoring dsmr/reading/phase_currently_returned_l2 due to queue
2021-04-28 20:12:58,112 | DEBUG    messages.queue_message:36 | MQTT: Ignoring dsmr/reading/phase_currently_returned_l3 due to queue
2021-04-28 20:12:58,138 | DEBUG    messages.queue_message:38 | MQTT: Queued message for dsmr/reading/phase_voltage_l1
2021-04-28 20:12:58,148 | DEBUG    messages.queue_message:36 | MQTT: Ignoring dsmr/reading/phase_voltage_l2 due to queue
2021-04-28 20:12:58,158 | DEBUG    messages.queue_message:36 | MQTT: Ignoring dsmr/reading/phase_voltage_l3 due to queue
2021-04-28 20:12:58,168 | DEBUG    messages.queue_message:36 | MQTT: Ignoring dsmr/reading/phase_power_current_l1 due to queue
2021-04-28 20:12:58,177 | DEBUG    messages.queue_message:36 | MQTT: Ignoring dsmr/reading/phase_power_current_l2 due to queue
2021-04-28 20:12:58,187 | DEBUG    messages.queue_message:36 | MQTT: Ignoring dsmr/reading/phase_power_current_l3 due to queue
2021-04-28 20:12:58,211 | DEBUG    messages.queue_message:38 | MQTT: Queued message for dsmr/reading/id
2021-04-28 20:12:58,233 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/meter-stats/dsmr_version due to cache
2021-04-28 20:12:58,239 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/meter-stats/electricity_tariff due to cache
2021-04-28 20:12:58,244 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/meter-stats/power_failure_count due to cache
2021-04-28 20:12:58,250 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/meter-stats/long_power_failure_count due to cache
2021-04-28 20:12:58,255 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/meter-stats/voltage_sag_count_l1 due to cache
2021-04-28 20:12:58,267 | DEBUG    messages.queue_message:36 | MQTT: Ignoring dsmr/meter-stats/voltage_sag_count_l2 due to queue
2021-04-28 20:12:58,279 | DEBUG    messages.queue_message:36 | MQTT: Ignoring dsmr/meter-stats/voltage_sag_count_l3 due to queue
2021-04-28 20:12:58,284 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/meter-stats/voltage_swell_count_l1 due to cache
2021-04-28 20:12:58,296 | DEBUG    messages.queue_message:36 | MQTT: Ignoring dsmr/meter-stats/voltage_swell_count_l2 due to queue
2021-04-28 20:12:58,306 | DEBUG    messages.queue_message:36 | MQTT: Ignoring dsmr/meter-stats/voltage_swell_count_l3 due to queue
2021-04-28 20:12:58,310 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/meter-stats/rejected_telegrams due to cache
2021-04-28 20:12:58,322 | DEBUG    mixins.run_loop:73 | dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
2021-04-28 20:13:03,333 | INFO     dsmr_datalogger_api_client.read_telegram:26 | [2021-04-28 20:13:03.333376] Opening connection "/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AI1TO64P-if00-port0" using options: {'baudrate': 115200, 'bytesize': 8, 'parity': 'N', 'stopbits': 1, 'xonxoff': 1, 'rtscts': 0}
[... cut ...]
2021-04-28 20:13:16,611 | INFO     dsmr_datalogger_api_client.read_telegram:26 | [2021-04-28 20:13:16.611276] Opening connection "/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AI1TO64P-if00-port0" using options: {'baudrate': 115200, 'bytesize': 8, 'parity': 'N', 'stopbits': 1, 'xonxoff': 1, 'rtscts': 0}
2021-04-28 20:13:16,950 | DEBUG    dsmr_datalogger_api_client.read_telegram:51 | [2021-04-28 20:13:16.950003] Read 633 Byte(s)
2021-04-28 20:13:16,962 | DEBUG    datalogger.telegram_to_reading:74 | Received telegram:
/ISK5\2X5d0E-1232
```

dennissiemensma, 2021-04-28:
```
MQTT: Ignoring <veld> due to cache
```

Al die berichten geven aan dat de queue vol zit en dan worden de berichten verwijderd om overload te voorkomen. In dat geval kun je proberen om de queue te vergroten. 
Je kunt dit toevoegen, onderaan `dsmrreader/settings.py`:

```
...

DSMRREADER_MQTT_MAX_MESSAGES_IN_QUEUE = 1000
```

En **herstart** de applicatie dan.

----

Maar het is ook belangrijk dat de queue wel verwerkt wordt en ik zie daar geen logging van voorbijkomen. Ik verwacht iets als:
```
2021-04-28 20:20:57,798 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m1), 'b'dsmr/telegram/id'', ... (8 bytes)
2021-04-28 20:20:57,799 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m2), 'b'dsmr/reading/phase_currently_returned_l3'', ... (5 bytes)
2021-04-28 20:20:57,800 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m3), 'b'dsmr/reading/phase_currently_returned_l2'', ... (5 bytes)
2021-04-28 20:20:57,801 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m4), 'b'dsmr/reading/phase_currently_returned_l1'', ... (5 bytes)
2021-04-28 20:20:57,802 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m5), 'b'dsmr/telegram/extra_device_delivered'', ... (8 bytes)
2021-04-28 20:20:57,803 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m6), 'b'dsmr/telegram/extra_device_timestamp'', ... (25 bytes)
2021-04-28 20:20:57,804 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m7), 'b'dsmr/telegram/phase_currently_delivered_l3'', ... (5 bytes)
2021-04-28 20:20:57,805 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m8), 'b'dsmr/telegram/phase_currently_delivered_l2'', ... (5 bytes)
2021-04-28 20:20:57,806 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m9), 'b'dsmr/telegram/phase_currently_delivered_l1'', ... (5 bytes)
2021-04-28 20:20:57,807 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m10), 'b'dsmr/telegram/electricity_currently_returned'', ... (5 bytes)
2021-04-28 20:20:57,809 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m11), 'b'dsmr/telegram/electricity_currently_delivered'', ... (5 bytes)
2021-04-28 20:20:57,811 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m12), 'b'dsmr/telegram/electricity_returned_2'', ... (5 bytes)
2021-04-28 20:20:57,812 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m13), 'b'dsmr/telegram/electricity_delivered_2'', ... (9 bytes)
2021-04-28 20:20:57,813 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m14), 'b'dsmr/telegram/electricity_returned_1'', ... (5 bytes)
2021-04-28 20:20:57,814 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m15), 'b'dsmr/telegram/electricity_delivered_1'', ... (9 bytes)
2021-04-28 20:20:57,815 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Sending PUBLISH (d0, q2, r1, m16), 'b'dsmr/telegram/timestamp'', ... (25 bytes)
2021-04-28 20:20:57,817 | DEBUG    broker.on_log:141 | MQTT: Paho client: on_log(userdata, level, buf): None 16 Received CONNACK (0, 0)
```

Nenodema, 2021-04-28:
> ```
> MQTT: Ignoring <veld> due to cache
> ```
>
> Al die berichten geven aan dat de queue vol zit en dan worden de berichten verwijderd om overload te voorkomen. In dat geval kun je proberen om de queue [...]

Oké, ik gebruik zelf het Docker image van xirixiz. Kan ik deze setting opnemen als env variable?

Nenodema, 2021-04-28:
En nog wat logging uit /var/log/supervisor/

```
2021-04-28 20:37:49,330 | DEBUG    broker.on_log:140 | MQTT: Paho client: on_log(userdata, level, buf) None 16 Received PINGRESP
2021-04-28 20:37:49,336 | DEBUG    mixins.run_loop:73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2021-04-28 20:37:50,349 | DEBUG    schedule.execute_scheduled_processes:28 | SP: 1 backend service(s) ready to run
2021-04-28 20:37:50,350 | DEBUG    schedule.execute_scheduled_processes:31 | SP: Running "Generate consumption data" (dsmr_consumption.services.run)
2021-04-28 20:37:50,389 | DEBUG    schedule.reschedule:72 | SP: Rescheduled "Generate consumption data" to 2021-04-28 20:37:51.376935+02:00 (ETA 0:00:00.987978)
2021-04-28 20:37:50,416 | DEBUG    persistent_clients.run:30 | CLIENTS: Running 1 active client(s)
2021-04-28 20:37:50,442 | DEBUG    mixins.run_loop:73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2021-04-28 20:37:51,454 | DEBUG    schedule.execute_scheduled_processes:28 | SP: 1 backend service(s) ready to run
2021-04-28 20:37:51,455 | DEBUG    schedule.execute_scheduled_processes:31 | SP: Running "Generate consumption data" (dsmr_consumption.services.run)
2021-04-28 20:37:51,493 | DEBUG    schedule.reschedule:72 | SP: Rescheduled "Generate consumption data" to 2021-04-28 20:37:52.481121+02:00 (ETA 0:00:00.987755)
2021-04-28 20:37:51,521 | DEBUG    persistent_clients.run:30 | CLIENTS: Running 1 active client(s)
2021-04-28 20:37:51,541 | DEBUG    mixins.run_loop:73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2021-04-28 20:37:52,551 | DEBUG    schedule.execute_scheduled_processes:28 | SP: 1 backend service(s) ready to run
2021-04-28 20:37:52,552 | DEBUG    schedule.execute_scheduled_processes:31 | SP: Running "Generate consumption data" (dsmr_consumption.services.run)
2021-04-28 20:37:52,795 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/electricity1 due to cache
2021-04-28 20:37:52,809 | DEBUG    messages.queue_message:38 | MQTT: Queued message for dsmr/day-consumption/electricity2
2021-04-28 20:37:52,813 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/electricity1_returned due to cache
2021-04-28 20:37:52,818 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/electricity2_returned due to cache
2021-04-28 20:37:52,832 | DEBUG    messages.queue_message:38 | MQTT: Queued message for dsmr/day-consumption/electricity_merged
2021-04-28 20:37:52,836 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/electricity_returned_merged due to cache
2021-04-28 20:37:52,840 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/electricity1_cost due to cache
2021-04-28 20:37:52,844 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/electricity2_cost due to cache
2021-04-28 20:37:52,848 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/electricity_cost_merged due to cache
2021-04-28 20:37:52,852 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/total_cost due to cache
2021-04-28 20:37:52,856 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/gas due to cache
2021-04-28 20:37:52,861 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/gas_cost due to cache
2021-04-28 20:37:52,865 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/energy_supplier_price_electricity_delivered_1 due to cache
2021-04-28 20:37:52,869 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/energy_supplier_price_electricity_delivered_2 due to cache
2021-04-28 20:37:52,873 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/energy_supplier_price_electricity_returned_1 due to cache
2021-04-28 20:37:52,877 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/energy_supplier_price_electricity_returned_2 due to cache
2021-04-28 20:37:52,881 | DEBUG    messages.queue_message:30 | MQTT: Ignoring dsmr/day-consumption/energy_supplier_price_gas due to cache
2021-04-28 20:37:52,914 | DEBUG    services.compact:76 | Compact: Processed reading: 5308438 @ 2021-04-28 20:38:52+02:00
2021-04-28 20:37:52,922 | DEBUG    schedule.reschedule:72 | SP: Rescheduled "Generate consumption data" to 2021-04-28 20:37:53.915644+02:00 (ETA 0:00:00.993480)
2021-04-28 20:37:52,937 | DEBUG    persistent_clients.run:30 | CLIENTS: Running 1 active client(s)
2021-04-28 20:37:52,948 | INFO     broker.run:71 | MQTT: Processing 20 message(s)
2021-04-28 20:37:52,949 | DEBUG    broker.on_log:140 | MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m550), 'b'dsmr/day-consumption/electricity_merged'', ... (5 bytes)
2021-04-28 20:37:52,958 | DEBUG    broker.on_log:140 | MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m551), 'b'dsmr/day-consumption/electricity2'', ... (5 bytes)
2021-04-28 20:37:52,965 | DEBUG    broker.on_log:140 | MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m552), 'b'dsmr/meter-stats/voltage_swell_count_l3'' (NULL payload)
2021-04-28 20:37:52,971 | DEBUG    broker.on_log:140 | MQTT: Paho client: on_log(userdata, level, buf) None 16 Sending PUBLISH (d0, q0, r1, m553), 'b'dsmr/meter-stats/voltage_swell_count_l2'' (NULL payload)
[... cut ...]
2021-04-28 20:38:11,865 | DEBUG    schedule.execute_scheduled_processes:31 | SP: Running "Generate consumption data" (dsmr_consumption.services.run)
2021-04-28 20:38:11,904 | DEBUG    schedule.reschedule:72 | SP: Rescheduled "Generate consumption data" to 2021-04-28 20:38:12.892050+02:00 (ETA 0:00:00.987875)
2021-04-28 20:38:11,927 | DEBUG    persistent_clients.run:30 | CLIENTS: Running 1 active client(s)
2021-04-28 20:38:11,947 | DEBUG    mixins.run_loop:73 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
```

dennissiemensma, 2021-04-28:
Vergeet wat ik zei, ik haal twee dingen door elkaar heen, ze lijken wel op elkaar:

```
MQTT: Ignoring %s due to cache
MQTT: Ignoring %s due to queue
```
https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_mqtt/services/messages.py#L30

De melding die je krijgt betekent dat exact hetzelfde bericht al recent verstuurd is. En de MQTT-berichten worden met `retain` verstuurd, waardoor de MQTT-broker de laatste waarde zou moeten bewaren.

Dat zeggende hebben, denk ik dat de cache wellicht niet goed geinvalideerd wordt. Dat zou wel een oorzaak kunnen zijn.

dennissiemensma, 2021-04-28:
Bij nader inzien zou dat wel goed moeten gaan. En de topics in de meldingen over de cache lijken ook vooral velden die niet vaak (als in na elke meting) perse wijzigen. Hmmmm.

dennissiemensma, 2021-04-28:
Desondanks denk ik dat ik wel een env var wil maken die de cache uit zet. Want caching zorgt eenmaal voor vage issues soms en ik durf nu niet te zeggen of het onderdeel is van het probleem of dat het zo _lijkt_.

Nenodema, 2021-04-28:
Check, ik snap wat je bedoeld. Het vreemde is dat het issue ongeveer een jaar is weggeweest. Zou het nog aan de mqtt server kunnen liggen? Deze draait op dezelfde machine als HomeAssistant en wordt regelmatig (als addon) geupdate.

dennissiemensma, 2021-04-28:
Ik vind het lastig om te zeggen eerlijk gezegd. Zoals ik eerder aangaf zit er een behoorlijke reeks aan schakels tussen het moment dat een meting binnenkomt en wat je uiteindelijk in HA ziet.

Wellicht dat een volgende stap is om via een MQTT-explorer oid in de gaten te houden hoevaak een waarde daar bijgewerkt wordt? In combinatie met de DEBUG-log kun je dan ook een `tail` + `grep` doen op het betreffende topic dat je in de gaten houdt. Misschien kun je daar nog een conclusie uit trekken.

dennissiemensma, 2021-04-29:
@Nenodema ben je nog wat wijzer geworden?

In de volgende release komt er wat meer logging en ook wat instellingen voor de queue/cache.En daar bovenop komt er een grote verbetering voor het gebruik van QoS 2, waarbij DSMR-reader gaat wachten totdat het MQTT-bericht daadwerkelijk verstuurd en ontvangen is. 

Dit alles zou het veel betrouwbaarder en makkelijker te debuggen moeten maken.

Nenodema, 2021-04-30:
Hi Dennis,

Gisteravond heb ik via een MQTT explorer util de data flows bekeken. Ik zie dat alle data netjes aankomt bij de MQTT server maar de data (het laatste datapoint) wordt niet altijd direct door Home Assistant opgepakt wordt en soms helemaal niet. 

Ik zie dat bijv. in MQTT bij phase_currently_returned de laatste waarde 0.00 is maar in home assistant wordt de waarde 0.15 getoond. Deze wordt dan pas weer bij een volgende aanlevering van een datapunt gewijzigd, al lijkt dit ook niet consistent te zijn. 

Zegt jou dit iets?

Daarnaast dacht ik ook dat het misschien aan de Postgres database kon liggen (performance issue oid), hier schrijft de Home Assistant recorder de data weg. Deze heb ik gepurged maar dit lijkt ook niet te helpen.

Ik kijk uit naar de nieuwe queue/cache features :-) Misschien dat dit iets gaat oplossen. Al lijkt alles er wel op te duiden dat het niet aan DSMR ligt!

Thanks voor je hulp!

dennissiemensma, 2021-04-30:
Bedankt voor je update. Ik denk dat het dat even wachten is op de volgende release, waar dus ook wat diepere debuglogging in zit en het wat makkelijker zoeken wordt waarom DSMR-reader iets wel of niet stuurt.

dennissiemensma, 2021-05-10:
Het is nu onderdeel van [v4.16](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.16.0) dus probeer het vooral!

Nenodema, 2021-05-10:
Dank Dennis! Zodra xirixiz de een nieuwe container klaar heeft staan ga ik het testen. Ben benieuwd!

dennissiemensma, 2021-05-11:
Er is nu ook aanvulling via [v4.16.1](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.16.1) en een bugfix voor de release van gisteren.

closed 2021-05-11 by dennissiemensma

Nenodema, 2021-05-13:
@dennissiemensma 

Het lijkt erop dat de setting "DSMRREADER_MQTT_MAX_CACHE_TIMEOUT=0" het probleem heeft opgelost. Thanks voor het fixen!

dennissiemensma, 2021-05-13:
Dank voor je update, fijn om te horen!