dsmrreader/dsmr-reader #1100: changing my password not visible

MRL-33, 2020-09-01:
When i want to change my password it's not visible on PC-screen. (Chrome)
[image]
Watch the vertical scrollbar.

Solution/workaround: zoom out till de vertical scrollbar disappear. Only then it's visible to change my password.

dennissiemensma, 2020-09-01:
Bedankt voor je melding. Ik denk niet dat ik het voor je kan oplossen, want deze (nieuwe) pagina komt uit het de meest recente versie van Django, het open-source framework wat ik gebruik.

Dat zeggende hebben kun je nog wel wat proberen, want volgens mij hoort het er niet zo uit te zien en mogelijk zit de cache of iets in de weg:
- Static files wissen en opnieuw kopieren:
```
sudo su - dsmr
./manage.py collectstatic --clear --no-input
./reload.sh
```
- Cache en cookies van je browser weggooien (hoeft uiteraard alleen voor DSMR-reader)

dennissiemensma, 2020-09-03:
@MRL-33 helpt bovenstaande? In #1103 lijkt het een soortgelijk issue op te lossen.

closed 2020-09-09 by dennissiemensma