dsmrreader/dsmr-reader #1101: 2 energie contracten

Simanias, 2020-09-03:
Sinds begin deze maand heb ik een aparte leverancier voor elektra(oftewel eentje voor gas en eentje voor elektra).
Als ik een apart contract toevoeg, krijg ik de foutmelding dat er al eentje bestaat in de bewuste tijdsspanne:

`At least one other contract collides with this date range: `

Is het mogelijk om dit aan te passen, zodat je wel verschillende leveranciers kunt toevoegen, aangezien de ingangsdatum en ook de einddatum van de twee contracten niet hetzelfde zijn.

dennissiemensma, 2020-09-03:
Bedankt voor je melding. Het is momenteel niet mogelijk om overlappende contract in te stellen, omdat DSMR-reader anders niet weet welke van de twee die moet gebruiken.

Wat ik wel kan proberen is kijken of ik het mogelijk kan maken dat contracten elkaar aanvullen wanneer velden niet ingevuld zijn. Ik zal het onderzoeken.

dennissiemensma, 2020-09-09:
Ik heb hiervoor iets kunnen maken.

Vanaf v4.4 kun je overlappende contracten invullen, mits die geen overlappende prijzen hebben. Dus als je in contract A elektra invult, kun je in contract B alleen nog gasprijzen invullen. 
Daardoor is als gevolg ook aan de voorkant het totaaloverzicht van contracttotalen bijgewerkt. Voortaan wordt daar alleen verbruik en kosten getoond als er prijzen ingesteld zijn. Als je dus ook een gasmeter hebt, maar geen gasprijs invult, wordt het gasverbruik daar niet meer getoond. Dit is nog wel gewoon zichtbaar in het Archief.

closed 2020-09-09 by dennissiemensma

dennissiemensma, 2020-09-25:
Zojuist uitgebracht in [v4.4](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v4.4)

Simanias, 2020-09-29:
Super, ik ga het meteen bekijken